thesofproject/sof #7338: [BUG] bug observed in ASRC module dai device finding

btian1, 2023-03-28:
**Describe the bug**
in ASRC.c, asrc_dai_find, initial purpose is find dai device, however, copier type also assigned as: SOF_COMP_DAI
So, line 641:
	/* Point dai_dev to found DAI */
	cd->dai_dev = dev;

is copier dev, not dai dev, check copier line: 683
			if (create_dai(dev, cd, &dev->ipc_config, copier, ipc_pipe->pipeline)) {

Due to CI does not cover ASRC, so just observe.

**To Reproduce**
observed, not sure whether windows have a way to test it.

**Reproduction Rate**
100%

**Expected behavior**
asrc_dai_find should find dai dev instead of copier dev.

**Impact**
block ASRC module.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

btian1, 2023-03-28:
@singalsu @lyakh @softwarecki , Please have a check!

btian1, 2023-03-28:
https://github.com/thesofproject/sof/pull/7229/commits/8b928b630c3bef753d55786b2592e4e1c4c5f953

lgirdwood, 2023-03-29:
> > I think here the topology should define the OUT rate and aplay the IN rate. This way teh ASRC does not need to look for the DAI.
> 
> The DAI is searched to access the timestamp driver info for the DAI. It tracks the real data flow rate vs. system wall-clock.

ok, btw does it make sense to allow this to be specified from topology (in the case where we have a demux) ?

singalsu, 2023-03-30:
> ok, btw does it make sense to allow this to be specified from topology (in the case where we have a demux) ?

A demux would cause the search to fail, though the ASRC should be next to DAI copier. In what kind of form would the topology pass the DAI reference? The component IDs in FW seem to be rather different than in topology. They are assigned in runtime by kernel.

lgirdwood, 2023-04-04:
@singalsu @btian1 so today we search and find teh first DAI as timestamp. This is fine except where we have more than one candidate DAI and in that case we should be able to look up by name. i.e.
1) topology sets the timestamp DAI id via a tuple in the ASRC. This is looked up at runtime for timestamp API.

and then later 

2) topology sets an array of DAI id to be used by ASRC and creates a kcontrol enum to select the ID for each use case.

singalsu, 2023-09-28:
@andrula-song Can this be closed now?

andrula-song, 2023-09-28:
yes,   I think we can close it now.

andrula-song, 2023-09-28:
I think it can be closed since we can get timestamp by dai_ts_xx ops of module_endpoint_ops now.

singalsu, 2023-09-28:
> I think it can be closed since we can get timestamp by dai_ts_xx ops of module_endpoint_ops now.

Great! Thank you for addressing this issue!

closed 2023-09-28 by singalsu