thesofproject/sof #7340: [BUG] chain-DMA errors when pipewire probes PCM devices at boot

kv2019i, 2023-03-28:
**Describe the bug**
With Ubuntu 22.10, a system with HDA-codec and DMIC, and SOF built with IPC4, following end up to kernel log:

```
[ 2908.920444] sof-audio-pci-intel-tgl 0000:00:1f.3: no llp found, fall back to default HDA path
[ 2908.924732] sof-audio-pci-intel-tgl 0000:00:1f.3: no llp found, fall back to default HDA path
[ 2908.928428] sof-audio-pci-intel-tgl 0000:00:1f.3: no llp found, fall back to default HDA path
[ 2908.958974] sof-audio-pci-intel-tgl 0000:00:1f.3: no llp found, fall back to default HDA path
[ 2908.972481] sof-audio-pci-intel-tgl 0000:00:1f.3: no llp found, fall back to default HDA path
[ 2908.976602] sof-audio-pci-intel-tgl 0000:00:1f.3: FW reported error: 6 - Unknown error while processing the request
[ 2908.976719] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc error for msg 0xe040000|0x0
[ 2908.976741] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_free on 0000:00:1f.3: -22
[ 2908.977207] sof-audio-pci-intel-tgl 0000:00:1f.3: FW reported error: 6 - Unknown error while processing the request
[ 2908.977268] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc error for msg 0xe040101|0x0
[ 2908.977299] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_free on 0000:00:1f.3: -22
[ 2908.977735] sof-audio-pci-intel-tgl 0000:00:1f.3: FW reported error: 6 - Unknown error while processing the request
[ 2908.977840] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc error for msg 0xe040202|0x0
[ 2908.977861] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_free on 0000:00:1f.3: -22
```

Functionality seems ok, but this is annoying to have these error level logs in dmesg on every boot.

**To Reproduce**
Install SOF 2.5-rc1 to any recent Ubuntu or Fedora machine that uses Pipewire as audio daemon.

**Reproduction Rate**
100%

**Expected behavior**
No errors in kernel dmesg in boot.

**Impact**
Annoyance (this is bound to result in bug reports)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 6.2 (or sof-dev of 2023-03-21)
    * SOF: SOF 2.5-rc1
2) Name of the topology file
    * Topology: sof-hda-generic
3) Name of the platform(s) on which the bug is observed.
    * Platform: tgl

kv2019i, 2023-03-28:
FYI @ujfalusi

makarukp, 2023-03-29:
@kv2019i Do we start scenario with msg 0xe040000|0x0 -> removing chain dma ?

lgirdwood, 2023-05-10:
@jsarha is this fixed, can we close ?

ujfalusi, 2023-05-26:
"no llp found, fall back to default HDA path" should not be printed on TGL, the delay reporting is MTL only feature, afaik.

jsarha, 2023-05-30:
Also HDMI audio appears to work as well as before. Thanks @ujfalusi ?

closed 2023-05-30 by kv2019i