thesofproject/sof #7356: [FEATURE]Convert  current generic Cadence Codec API to nongeneric(sepcific) supporting reduced codecs set

makarukp, 2023-03-31:
**Is your feature request related to a problem? Please describe.**
Currently we expose Cadence Codec API adapter as generic for all codecs provided by Cadence but it is not always true.

During attempt to implement Cadence Opus enc/dec form latest Cadence relase problem with adaptation was detected. It is not possible to adapt xa_opus_enc/xa_opus_dec with the rest API behind that to current generic API because it has totally different function signatures. After analysis there is no clear idea of how to extend it.
It seems that Cadence does not support generic API for all codecs and it could vary in each Cadence release.

**Describe the solution you'd like**
Convert current generic Cadence API adapter to nongeneric appliable for specific set of codecs supported by converge fw.
Introduce new cadence adapter specific for Opus enc/dec. 

**Describe alternatives you've considered**
Reconsider how tu support changable Cadence Codec API.

**Additional context**
Example:
Opus enc/dec functtion signature does not match generic xa_codec_func_t.

paulstelian97, 2023-03-31:
The opus codec is a special case that is dealt with separately in the proprietary internal firmware I have access to as well, instead of the generic interface that all other codecs we have access to satisfy. Every codec other than Opus does obey the interface in the current Cadence Codec API.

makarukp, 2023-04-04:
> The opus codec is a special case that is dealt with separately in the proprietary internal firmware I have access to as well, instead of the generic interface that all other codecs we have access to satisfy. Every codec other than Opus does obey the interface in the current Cadence Codec API.

@paulstelian97 What do you mean by writing "dealt with separately in the proprietary internal firmware I have access to as well"? Why is it a special case ?
What is the scope of quantifier "every" in "every codec other than Opus"?

paulstelian97, 2023-04-04:
> What is the scope of quantifier "every" in "every codec other than Opus"?

The AAC, BSAC, DAB, DRM, MP2, MP3 and SBC codecs all fit this interface. The OPUS one is the only one that is integrated that doesn't. We currently only have AAC and MP3 integrated in SOF, but do intend to integrate the rest to reach feature parity.

I don't have an idea why Opus is different from all of the others, but it is. We did not internally find a common interface that can cover Opus vs _literally everything else we got_. I suppose we'd have to either add a separate case, or lose a lot of functionality by generalizing to cover both. Or perhaps somehow add a second adaptor layer.

makarukp, 2023-04-05:
What about other APIs ? Those (AAC, BSAC, DAB, DRM, MP2, MP3 and SBC) are only short fragment of dozens of cases on wide Cadence codecs list page. Is Opus the only exception on that list?

paulstelian97, 2023-04-05:
I have only considered the ones that we have integrated here. The ones that NXP never considered might have other signatures as well, and it is quite possible that there is nothing common in there. Anyway I suspect the Opus one can be translated into this implementation via a simple adaptor layer.