thesofproject/sof #7408: [BUG] XRUNs happen when testing suspend/resume with capture on MTL-NOCODEC

keqiaozhang, 2023-04-06:
**Describe the bug**
We observed that capture process status exception with overruns during the suspend/resume test.  The reproduction is about ~30% and it appeared twice in daily CI test.

**console log**
```
2023-04-06 00:06:50 UTC Sub-Test: [REMOTE_INFO] ===== Round(5/5) =====
2023-04-06 00:06:50 UTC Sub-Test: [REMOTE_COMMAND] Run the command: rtcwake -m mem -s 5
rtcwake: assuming RTC uses UTC ...
rtcwake: wakeup from "mem" using /dev/rtc0 at Thu Apr  6 00:06:56 2023
2023-04-06 00:06:56 UTC Sub-Test: [REMOTE_COMMAND] sleep for 5
arecord: pcm_read:2221: read error: Input/output error
2023-04-06 00:07:01 UTC Sub-Test: [REMOTE_INFO] Check for the kernel log status
declare -- cmd="journalctl_cmd --since=@1680739609"
2023-04-06 00:07:02 UTC Sub-Test: [REMOTE_INFO] Check for wakeup_count and suspend_stats
declare -- cmd="journalctl_cmd --since=@1680739609"
2023-04-06 00:07:02 UTC Sub-Test: [REMOTE_INFO] pkill -TERM -f mtrace-reader.py
2023-04-06 00:07:02 UTC Sub-Test: [REMOTE_INFO] Test Result: PASS!
declare -- cmd="journalctl_cmd --since=@1680739559"
process: 17076, status: not found
2023-04-06 00:07:03 UTC [REMOTE_ERROR] process status is abnormal
```

**mtrace**
```
[00:00:00.684,600] <inf> dai_intel_ssp: dai_ssp_get_properties: dai_ssp_get_properties dai_index 0
[00:00:00.684,608] <inf> dai_intel_ssp: dai_ssp_get_properties: dai_ssp_get_properties fifo 163856
[00:00:00.684,616] <inf> dai_intel_ssp: dai_ssp_get_properties: dai_ssp_get_properties handshake 2
[00:00:00.684,626] <inf> dai_intel_ssp: dai_ssp_get_properties: dai_ssp_get_properties init delay 0
[00:00:00.684,638] <inf> dai_intel_ssp: dai_ssp_pre_start: dai_ssp_pre_start
[00:00:00.684,713] <inf> dai_intel_ssp: dai_ssp_start: dai_ssp_start
[00:00:00.684,725] <inf> dai_intel_ssp: dai_ssp_update_bits: dai_ssp_update_bits base 28000, reg 34, mask 100, value 100
[00:00:00.684,791] <wrn> dai_comp: comp:0 0x40000 dai_copy(): nothing to copy
[00:00:00.685,571] <inf> host_comp: host_get_copy_bytes_normal: comp:1 0x40003 no bytes to copy, available bytes: 0, free_bytes: 768
[00:00:01.706,626] <inf> ll_schedule: zephyr_domain_thread_fn: ll timer avg 3117, max 10401, overruns 0
[00:00:02.730,626] <inf> ll_schedule: zephyr_domain_thread_fn: ll timer avg 3085, max 4258, overruns 0
[00:00:03.754,626] <inf> ll_schedule: zephyr_domain_thread_fn: ll timer avg 3084, max 4259, overruns 0
[00:00:04.778,628] <inf> ll_schedule: zephyr_domain_thread_fn: ll timer avg 3085, max 4251, overruns 0
[00:00:05.802,626] <inf> ll_schedule: zephyr_domain_thread_fn: ll timer avg 3085, max 4262, overruns Terminated
```

**To Reproduce**
~/sof-test/test-case/check-suspend-resume-with-audio.sh -l 5 -m capture

**Reproduction Rate**
~30%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {[b64d8f76d5d6](https://github.com/thesofproject/linux/commit/b64d8f76d5d6)}
    * SOF: {[650423feb197](https://github.com/thesofproject/sof/commit/650423feb197)}
2) Name of the topology file
    * Topology: {sof-ace-tplg/sof-mtl-nocodec.tplg }
3) Name of the platform(s) on which the bug is observed.
    * Platform: {MTL-NOCODEC}

[mtrace.txt](https://github.com/thesofproject/sof/files/11165652/mtrace.txt)

[dmesg.txt](https://github.com/thesofproject/sof/files/11165653/dmesg.txt)

RanderWang, 2023-04-11:
the major reason is that GP-DMA doesn't copy data or SSP doesn't transmit data after a couple cycles of suspend-resume test.

wszypelt, 2023-04-21:
@keqiaozhang Unfortunately, the logs are missing Payloads data, request for access to IPC with Payloads

RanderWang, 2023-04-24:
It is very hard to reproduce it and even we can we don't get any error

fredoh9, 2023-04-24:
Most recent failure report from daily test is on 2023-04-20.

planresultdetail/23905?model=MTLP_RVP_NOCODEC&testcase=check-suspend-resume-with-capture-5

Reproduction rate is lowered. Please advise if we can lower to P2.

RanderWang, 2023-04-25:
> 

> Most recent failure report from daily test is on 2023-04-20.
> 
> planresultdetail/23905?model=MTLP_RVP_NOCODEC&testcase=check-suspend-resume-with-capture-5
> 
> Reproduction rate is lowered. Please advise if we can lower to P2.

thanks. We can keep eye on it for some days

mengdonglin, 2023-05-09:
Reproduce rate is less than 10%. Lower priority.

keqiaozhang, 2023-06-26:
This issue cannot be reproduced in CI for a month. Closing it.

closed 2023-06-26 by keqiaozhang