thesofproject/sof #7418: [BUG] Headphone cannot output sound with SN6140 in RPL

MrBook2019, 2023-04-07:
**Describe the bug**
Thinkpad L14 got a senary SN6140 codec, speaker, Dmic, headset mic works fine.
Only no sound output from headphone.
When disable Audio DSP，the issue is reproduced too.
The same os and codec, but cpu change to ADL, headphone can output sound.

**To Reproduce**
Play music and then plugin headphone, or plugin headphone and then play music.

**Reproduction Rate**
100%

**Expected behavior**
Headphone can output sound.

**Impact**
I can't use headphone to listen music.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: Linux version 5.19.17 (root@book-ThinkPad-L14-Gen-4) (gcc (Ubuntu 12.2.0-3ubuntu1) 12.2.0, GNU ld (GNU Binutils for Ubuntu) 2.39) #1 SMP PREEMPT_DYNAMIC Mon Mar 13 10:36:48 CST 2023
    * OS:   Ubuntu 22.10
    * SOF: [    6.341055] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
               [    6.341059] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:21:0
2) Name of the topology file
    * Topology: default
3) Name of the platform(s) on which the bug is observed.
    * Platform: 
        CPU : 13th Gen Intel(R) CoreT i5-1340P
        audio codec : Conexant SN6140

**Screenshots or console output**
book@book-ThinkPad-L14-Gen-4:~$ inxi -A
Audio:
  Device-1: Intel driver: sof-audio-pci-intel-tgl
  Sound Server-1: ALSA v: k5.19.17 running: yes
  Sound Server-2: PipeWire v: 0.3.58 running: yes

book@book-ThinkPad-L14-Gen-4:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 3: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 4: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 5: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 31: HDA Analog Deep Buffer (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
book@book-ThinkPad-L14-Gen-4:~$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 6: DMIC (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 7: DMIC16kHz (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

When headphone plugin and play music:
book@book-ThinkPad-L14-Gen-4:~$ sudo hdajacksensetest 
[sudo] password for book: 
Pin 0x16 (Black Headphone, Left side): present = Yes
Pin 0x19 (Black Mic, Right side): present = Yes

When headphone plugout and play music:
book@book-ThinkPad-L14-Gen-4:~$ sudo hdajacksensetest 
Pin 0x16 (Black Headphone, Left side): present = No
Pin 0x19 (Black Mic, Right side): present = No

MrBook2019, 2023-04-07:
sof log as below:
[    6.202409] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[    6.226174] sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 5
[    6.226181] sof-audio-pci-intel-tgl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    6.226205] sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 2
[    6.229055] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[    6.229060] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:21:0
[    6.229061] sof-audio-pci-intel-tgl 0000:00:1f.3: FW ABI is more recent than kernel
[    6.229070] sof-audio-pci-intel-tgl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
[    6.341055] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[    6.341059] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:21:0
[    6.341062] sof-audio-pci-intel-tgl 0000:00:1f.3: FW ABI is more recent than kernel
[    6.344256] sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:22:1 Kernel ABI 3:21:0
[    6.344260] sof-audio-pci-intel-tgl 0000:00:1f.3: warn: topology ABI is more recent than kernel
[    6.344494] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Parent card not yet available, widget card binding deferred
[    6.375980] snd_hda_codec_conexant ehdaudio0D0: SN6140: BIOS auto-probing.
[    6.376568] snd_hda_codec_conexant ehdaudio0D0: autoconfig for SN6140: line_outs=1 (0x17/0x0/0x0/0x0/0x0) type:speaker
[    6.376571] snd_hda_codec_conexant ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    6.376573] snd_hda_codec_conexant ehdaudio0D0:    hp_outs=1 (0x16/0x0/0x0/0x0/0x0)
[    6.376574] snd_hda_codec_conexant ehdaudio0D0:    mono: mono_out=0x0
[    6.376575] snd_hda_codec_conexant ehdaudio0D0:    inputs:
[    6.376576] snd_hda_codec_conexant ehdaudio0D0:      Mic=0x19
[    6.384678] snd_hda_codec_conexant ehdaudio0D0: ASoC: sink widget AIF1TX overwritten
[    6.384687] snd_hda_codec_conexant ehdaudio0D0: ASoC: source widget AIF1RX overwritten
[    6.384839] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget hifi3 overwritten
[    6.384844] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget hifi2 overwritten
[    6.384849] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget hifi1 overwritten
[    6.384853] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: source widget Codec Output Pin1 overwritten
[    6.384857] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget Codec Input Pin1 overwritten
[    6.384863] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget Analog Codec Playback overwritten
[    6.384869] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget Digital Codec Playback overwritten
[    6.384874] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget Alt Analog Codec Playback overwritten
[    6.384882] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: source widget Analog Codec Capture overwritten
[    6.384888] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: source widget Digital Codec Capture overwritten
[    6.384893] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: source widget Alt Analog Codec Capture overwritten
[    6.384906] skl_hda_dsp_generic skl_hda_dsp_generic: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 3
[    6.409257] input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input16
[    6.409311] input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input17
[    6.409343] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input18
[    6.409372] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input19
[    6.409413] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input20

lgirdwood, 2023-04-07:
@MrBook2019 

> When headphone plugin and play music: book@book-ThinkPad-L14-Gen-4:~$ sudo hdajacksensetest [sudo] password for book: Pin 0x16 (Black Headphone, Left side): present = Yes Pin 0x19 (Black Mic, Right [...]

@MrBook2019 sorry, can you confirm if it's teh speakers or heaphone that does not work. It seems above when headphone is plugged in you can hear audio on teh headphones but no audio from speakers when headphones removed?

If this is correct then I can see the SN6140 has a class D amplifier and it may not be enabled by the SN6140 codec driver.
https://en.senarytech.com/product/9.html