thesofproject/sof #7419: [BUG] waveform deformation with headset capture on IPC4 SDW platforms

keqiaozhang, 2023-04-07:
**Describe the bug**
We recently enabled alsabat test for headset capture on all IPC4 platforms, but found that the waveform is not perfect on all IPC4 SDW platforms.
[image]

**To Reproduce**
~/sof-test/test-case/check-alsabat.sh -c hw:sofsoundwire,1 -p hw:Device,0 -C 2 -F 599

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {[cb01c5d89222](https://github.com/thesofproject/linux/commit/cb01c5d89222)}
    * SOF: {[8f5ee926e20c](https://github.com/thesofproject/sof/commit/8f5ee926e20c)}
2) Name of the topology file
    * Topology: {sof-ace-tplg/sof-mtl-rt711-4ch.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {IPC4 SDW platform}

ranj063, 2023-04-08:
@keqiaozhang can we try without the EQiir in the capture pipeline? @singalsu FYI

fredoh9, 2023-04-10:
I'm trying to reproduce this with/without EQIIR in capture pipeline. But I'm seeing silence issue first. looks switch/gain settings are not right. Looking at it now.

fredoh9, 2023-04-10:
I tried with TGLU_RVP_SDW without any change(=with EQIIR), alsabat capture tests are all passed. I saw the failures and waveform (planresultdetail/23296) but today's daily test(planresultdetail/23323) also passed with same SHA1 build.

MTLP_RVP_SDW issue is different. I will take a look at alsa controls.

fredoh9, 2023-04-10:
> I'm trying to reproduce this with/without EQIIR in capture pipeline on MTL SDW. But I'm seeing silence issue first. looks switch/gain settings are not right. Looking at it now.

It looks like device specific issue,
- sh-tglu-rvp-sdw-02: i tried on it, it failed. saw same waveform
- sh-tglu-rvp-sdw-01: tested works fine
- sh-tglu-rvp-sdw-03: today's daily test passed

@keqiaozhang can you replace the USB soundcard/loopback cables and try again? Removed P1

fredoh9, 2023-04-10:
for MTLP_RVP_SDW, I'm able to verify capture using headset(4-pole plug).
When I use loopback cables, Headset Mic Jack/Headset Mic Switch became off, I can make the Switch ON but 'Headset Mic Jack' is read only.

```
numid=11,iface=CARD,name='Headset Mic Jack'
  ; type=BOOLEAN,access=r-------,values=1
  : values=off
numid=9,iface=MIXER,name='Headset Mic Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=off
```

keqiaozhang, 2023-04-11:
> @keqiaozhang can you replace the USB soundcard/loopback cables and try again? Removed P1

I did a check and found that this issue is not related to the USB sound card, it caused by one amixer control `rt711 AMIC Volume`. If I set `rt711 AMIC Volume` to 0, then it can always pass the test. 
```
amixer -c sofsoundwire cset name='rt711 AMIC Volume' 0
```

keqiaozhang, 2023-04-11:
> When I use loopback cables, Headset Mic Jack/Headset Mic Switch became off

This behavior is designed  in the codec driver. If you connect a real headset with a mic, this switch will automatically turn on.
https://github.com/thesofproject/linux/blob/topic/sof-dev/sound/soc/intel/boards/sof_sdw_rt711.c#LL109C1-L120C3

But it should be not a problem for alsabat test, we use amixer to turn on this switch before testing, it works.

lgirdwood, 2023-04-24:
@keqiaozhang any update ?

keqiaozhang, 2023-05-04:
Closing this bug, https://github.com/thesofproject/sof-test/pull/1021 has been merged and this issue no longer exists in CI.

closed 2023-05-04 by keqiaozhang