thesofproject/sof #7476: [BUG] toplogy1 : error: tuple SOF_TKN_BUF_SIZE: Numerical result out of range when pipeline period size increases beyond 40ms

Vamshigopal, 2023-04-20:
**Describe the bug**
when we increase the period size beyond 40ms in PIPELINE_PCM_ADD and DAI_ADD in *.m4 , it fails to build the topology with below error.

ALSA lib data.c:854:(parse_tuple_set) tuple SOF_TKN_BUF_SIZE: Numerical result out of range
ALSA lib parser.c:212:(tplg_load_config) failed to parse topology
Unable to load configuration: Invalid argument

Branch: main
alsatplg version 1.2.8
libasound version 1.2.8
libatopology version 1.2.8

**To Reproduce**

"sof-tgl-max98357a-rt5682\;sof-adl-max98360a-rt5682\;-DCODEC=MAX98360A\;-DFMT=s32le\;-DPLATFORM=adl\;-DLINUX_MACHINE_DRIVER=sof_rt5682\;-DAMP_SSP=1\;-DBT_OFFLOAD\;-**DSPK_MIC_PERIOD_US=500000**"

./scripts/build-tools.sh

**Reproduction Rate**
10/10

**Expected behavior**
Build should work without any errors.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: main
2) Name of the topology file
    * Topology: tools/topology/topology1/sof-tgl-max98357a-rt5682.m4
3) Name of the platform(s) on which the bug is observed.
    * Platform: RPL

**Screenshots or console output**
[tplg-build.log](https://github.com/thesofproject/sof/files/11284640/tplg-build.log)

sathyap-chrome, 2023-04-20:
@kv2019i  is there any change in RPL-001 or cavs-2.5 release ? I recollect we could change it in older releases

kv2019i, 2023-04-20:
@Vamshigopal @sathyap-chrome No change that at least I know about. I also tried this locally with our SOF docker image and I could not trigger this error. Will continue to look at this... maybe something in my test procedure.

kv2019i, 2023-04-20:
Hmm, this would have to be a -ERANGE from tplg_get_unsigned() in alsa-lib, and that only happens if the value is negative or beyond UINT_MAX . 50000 clearly isn't above UINT_MAX, so something is missing here. Continuing debug...

Vamshigopal, 2023-04-20:
Thanks @kv2019i for the quick fix.
Tested the fix #7481, it works fine with larger buffers size.

plbossart, 2023-04-20:
This is an important things to fix, but seriously does this make any sense to use period sizes > 40ms? This will have a very large latency which will become detectable by users, e.g. for A/V sync.

I mean, there are limits to this, 10ms was already pushing the reasonable. Just the fact that we test this makes me a bit uncomfortable, this can't be a product solution, can it?

@cujomalainey @kv2019i what am I missing?

sathyap-chrome, 2023-04-21:
@plbossart this is for pnp analysis and for large buffer usage - for deep buffer kind of use case. This is not planned for productization yet. We are aware of the latency issue with high buffers.

cujomalainey, 2023-04-21:
@plbossart agreed which is why this is being implemented a completely separate path where we can measure latency and AV sync media. I.e. we are leveraging multiple endpoints.

plbossart, 2023-04-21:
phew, you guys scared me ;-)

All good to experiment.

We did the same thing some 12 years ago in the phone space, the buffers were increased and then user-experience forced the sizes the be pulled back to more reasonable values, typical issues are A/V sync, dropped samples or wait when re-routing to different endpoints, deferred volume changes, etc.

There's also a typical 'plateau' in power levels, usually after 100ms of buffering the returns on power consumption become marginal.

cujomalainey, 2023-04-24:
> There's also a typical 'plateau' in power levels, usually after 100ms of buffering the returns on power consumption become marginal.

Even when doing things like compressed offload? I am just trying to understand how a plateau is hit if the audio server can stop waking up

plbossart, 2023-04-24:
@cujomalainey most of the power improvements will happen when you move from e.g. 20ms to 100ms. After that usually the returns are not that important. This is true of both compressed and PCM data.

It makes sense in that the sleep times are now one or two orders of magnitude what it takes to enter low-power states, and the 'active' power becomes less significant compared to 'idle' power.

Vamshigopal, 2023-05-05:
@kv2019i we see a hw params ipc failed error while verifying the audio with larger buffers (>44ms). Looks like host failed to allocate large dma buffers.

[dmesg.log](https://github.com/thesofproject/sof/files/11405085/dmesg.log)

[  528.412709] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  528.413719] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x60010000 (msg/reply size: 108/20): -12
[  528.413737] sof-audio-pci-intel-tgl 0000:00:1f.3: HW params ipc failed for stream 1

[sof-trace.log](https://github.com/thesofproject/sof/files/11405090/sof-trace.log)

[     1787215.501899] (          14.895833) c1 buffer                   src/audio/buffer.c:55   ERROR buffer_alloc(): could not alloc size = 4294963712 bytes of type = 32
[     1787236.074815] (          20.572916) c1 dma-trace                 src/trace/trace.c:141  INFO Suppressed 22 similar messages:    Number of Blocks: total % used %d free %d
[     1787249.303981] (          13.229166) c1 host         1.0           src/audio/host.c:845  ERROR host_params(): failed to alloc dma buffer
[     1787271.335230] (          22.031250) c1 pipe         1.7   ....../pipeline-params.c:268  ERROR pipeline_params(): ret = -12, host->comp.id = 0
[     1787293.783146] (          22.447916) c1 ipc                  src/ipc/ipc3/handler.c:304  ERROR ipc: pipe 1 comp 0 params failed -12

kv2019i, 2023-05-16:
@Vamshigopal Sorry, missed this comment. "ROR buffer_alloc(): could not alloc size = 4294963712 bytes of type = 32" clear indicates an invalid value passed from the host. If you can repeat this, can you share the ".conf" file matching the ".tplg" and I can check this.

Vamshigopal, 2023-05-17:
@kv2019i here is the .conf and .tplg used,
[conf-tplg.zip](https://github.com/thesofproject/sof/files/11495338/conf-tplg.zip)

cujomalainey, 2023-05-17:
would probably be good to dump the IPC message on the kernel side so you can identify if the math error is on the host or dsp

kv2019i, 2023-05-19:
@Vamshigopal let me open a new item to track this. I think I found the problem (it's on FW side).