thesofproject/sof #7478: [BUG] Audio playback gets heavily distorted after rapid pause/resume on HDA analog output

ujfalusi, 2023-04-20:
**Describe the bug**
While testing audio playback on hw:0,0 on UPX-i11 board I have checked pause/resume with aplay -i (holding <SPACE> key to rapid pause/resume) and after  few seconds the audio playback got corrupted.
The playback continued normally, firmware reported no issues, but the audio is completely distorted, high pitched and garbled.

**To Reproduce**
aplay -iv -Dhw:0,0 <wav file>
# Listen to it, sounds ok.

Now, press and HOLD the <SPACE> key to rapid pause/resume aplay
after few seconds you can hear that the sound starts to be up-shifted pitched, now release <SPACE> and if it is paused, press it again to resume.
audio is pitched up, distorted.
Not affecting HDMI (ChainDMA).

The same issue present on DeepBuffer PCM (-Dhw:0,31)

**Reproduction Rate**
10/10

**Expected behavior**
audio is not distorted

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: topic/sof-dev or upstream linux-next
    * SOF: ebaa1429f9d40dc17a28438193789a771cbab2cf
2) Name of the topology file
    * Topology: sof-hda-generic.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL (UPX-i11)

**Screenshots or console output**
nothing out of ordinary in logs, all looks error free.

plbossart, 2023-04-20:
Someone was talking about finding something to keep themselves busy yesterday @ujfalusi.

Humm, wish granted, eh? wow, nice bug, this will be a tough one to crack.

ujfalusi, 2023-04-21:
Update: it is not the rapid pause/resume, it can be a slow, pasue/resume, small wait, pause/resume.
After the 10th or so iteration the audio starts to be distorted.

ujfalusi, 2023-04-24:
Update: it is not reproducible with the reference firmware.

ujfalusi, 2023-04-24:
this is the original tplg path:
[image]

I have tried with these variations:
[image] [image] [image]

all of them exhibiting the same issue, the last one (without gains) interestingly will reproduce it much faster (4th pause/resume sequence)

ujfalusi, 2023-04-25:
I'm using GCC to build the firmware, I got an XCC built binary and that works OK, I cannot reproduce the issue

PR #7517 (2023-04-26): audio: mixin_mixout_generic: normal_mix: Update src/dst after each copy
The copying part of the functions need to update the src/dst buffer pointers after each iteration to correctly account for the possible wrap of the buffer.

This was discovered while doing pause - 1sec_wait - resume - 3sec_wait cycles and around the 10th iteration audio became 'metallic' sounding.

Fixes: #7478 

Thanks @ranj063 to test with XCC which is not affected and pointing me to the correct path.
closed 2023-04-26 by kv2019i