thesofproject/sof #7506: [BUG][MTL] DMIC capture returned "Input/output error"

yongzhi1, 2023-04-24:
**Describe the bug**
DMIC capture returned error immediately,

**To Reproduce**
```
arecord -Dhw:0,99 /tmp/deep.wav -c4 -r48000 -fs32_le 
```
**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel:  https://chromium-review.googlesource.com/c/chromiumos/third_party/kernel/+/4459333/1 + https://github.com/thesofproject/linux/pull/4313/commits/9dcdef43efba05d75eba013f6840cddc677b4ee4
    * SOF: {https://github.com/thesofproject/sof/commit/ca68ba68f22c2bf3c209b62466a2aaac6b09ea2d }

2) Name of the topology file
    * Topology: { sof-ace-tplg/sof-mtl-max98357a-rt5682-ssp2-ssp0.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {MTL}

[dmesg-arecord-io-err.txt](https://github.com/thesofproject/sof/files/11313725/dmesg-arecord-io-err.txt)
[mtrace-dmic-4xch-capture-01.txt](https://github.com/thesofproject/sof/files/11313731/mtrace-dmic-4xch-capture-01.txt)

abonislawski, 2023-04-25:
@yongzhi1 only on main or mtl-003/004 too?

yongzhi1, 2023-04-25:
Hi, @abonislawski , yes, it happened on main only, one observation found by accident is, if HDMI playback gets started first, then DMIC capture works fine. Here is the trace captured with the following seq:

Start DMIC capture and observe Input/output error
```arecord -Dhw:0,99 /tmp/test.wav -c4 -r48000 -fs32_le```

Start playback on HDMI from another console:
```aplay -Dhw:0,5 -c2 -r48000 -fS32_LE /dev/zero```

Run the DMIC capture again:
```arecord -Dhw:0,99 /tmp/test.wav -c4 -r48000 -fs32_le```
[mtrace-dmic-hdmi-dmic-dump.txt](https://github.com/thesofproject/sof/files/11324604/mtrace-dmic-hdmi-dmic-dump.txt)

yongzhi1, 2023-05-06:
Will re-open when receiving more sightings.

yongzhi1, 2023-05-09:
@udaymb also observed the same issue, so re-open and updated the build environment to reflect the latest recipe, thanks!!

yongzhi1, 2023-05-15:
Thanks to the following patch, the IO error is rarely seen:
https://github.com/softwarecki/sof/commit/d75d44f0c9ad285a92215fa80f3ff3c44b5f39e8

yongzhi1, 2023-05-17:
Patch merged to mtl-003-drop-stable, thanks @softwarecki @abonislawski for your great support!!

jairaj-arava, 2023-06-01:
@abonislawski @softwarecki   the issue is appeared on multiple boards eventhough we have #7654  fix in the fw used. Hence, re-opened.

bardliao, 2023-06-08:
After chatting with @yongzhi1 and @udaymb , it seems that the issue only happens after suspend test.
To reproduce:
1. run "suspend_stress_test -c10"
2. run arecord -Dhw:0,99 -fs32_le -r48000 -c4 /tmp/test.wav -vv

I see IPC error in freeing pipeline, but it seems not be the reason of IO error.
[dmesg.txt](https://github.com/thesofproject/sof/files/11685905/dmesg.txt)
```
[ 3243.730471] sof-audio-pci-intel-mtl 0000:00:1f.3: FW reported error: 12 - Required resource is in invalid state
[ 3243.741836] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc error for msg 0x12000000|0x0
[ 3243.750364] sof-audio-pci-intel-mtl 0000:00:1f.3: failed to free pipeline widget pipeline.14
[ 3243.760475] sof-audio-pci-intel-mtl 0000:00:1f.3: FW reported error: 12 - Required resource is in invalid state
[ 3243.771914] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc error for msg 0x12010000|0x0
[ 3243.780414] sof-audio-pci-intel-mtl 0000:00:1f.3: failed to free pipeline widget pipeline.13
[ 3243.789854] sof-audio-pci-intel-mtl 0000:00:1f.3: Failed to free connected widgets
[ 3243.798364] sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_free on 0000:00:1f.3: -22
```

And from mtrace, it shows 0 avail in buffer.
```
[00:00:00.201,063] <wrn> dai_comp: comp:0 0x40000 dai_copy(): nothing to cople_common: task add 0xa00f8500 0x20180U priority 0 flags 0x0
[00:00:00.141,060] <inf> host_comp: host_get_copy_bytes_normal: comp:1 0x40002 no bytes to copy, 0 avail in buffer, 3072 free in DMA
[00:00:00.141,098] <inf> dai_intel_dmic: dai_dmic_start: dmic_start(), dmic->fifo_a
[00:00:00.141,106] <inf> dai_intel_dmic: dai_dmic_update_bits: dai_dmic_update_bits base 10000, reg 0, mask 6000000, value 4000000
[00:00:00.141,115] <inf> dai_intel_dmic: dai_dmic_start: dmic_start(), pdm0 mic_a = 1, mic_b = 1
[00:00:00.141,121] <inf> dai_intel_dmic: dai_dmic_update_bits: dai_dmic_update_bits base 10000, reg 1000, mask c000, value c000
[00:00:00.141,130] <inf> dai_intel_dmic: dai_dmic_update_bits: dai_dmic_update_bits base 10000, reg 100c, mask 3, value 3
[00:00:00.141,138] <inf> dai_intel_dmic: dai_dmic_update_bits: dai_dmic_update_bits base 10000, reg 1020, mask 80, value 80
[00:00:00.141,145] <inf> dai_intel_dmic: dai_dmic_start: dmic_start(), pdm1 mic_a = 1, mic_b = 1size 0x600 id 0.0 flags 0x0
[00:00:00.135,896] <inf> pipe: pipeline_connect: comp:0 0x60000 connect buffer 0 as sink
[00:00:00.135,905] <inf> pipe: pipeline_connect: comp:0 0x40001 connect buffer 0
```

abonislawski, 2023-06-22:
Investigating this now, L1 changes looks promising

yongzhi1, 2023-06-29:
Issue fixed on https://github.com/thesofproject/sof/tree/mtl-005-drop-stable, thank you @abonislawski again for the great support!

mengdonglin, 2023-08-03:
This issue is a duplicate of https://github.com/thesofproject/sof/issues/7646

mengdonglin, 2023-09-04:
Zephyr fixes got merged https://github.com/zephyrproject-rtos/zephyr/pull/62101 and SOF updated for v2.7

commit 5ad5da5b (2023-05-29): dai: always reload dma ll chain

A problem was observed with launching capture with dmic. Sometimes the host
starts reading data too late, so that the capture buffer becomes full.
In this case, the dma descriptors are not reloaded. This leads to stopping
the dma. When the host starts reading and reads the accumulated data the
stream is already stopped. With this patch, the descriptors are always
reloaded and the stream will not be stopped. A glitch may occur but it is
better than no data at all.

Fixes #7506.