thesofproject/sof #7534: [BUG] Firmware exception: assertion failure checking for address validity

lyakh, 2023-04-28:
**Describe the bug**
An invalid address has been detected in `intel_adsp_hda_set_buffer()` https://sof-ci.01.org/sofpr/PR7526/build6566/devicetest/index.html?model=ADLP_RVP_SDW_IPC4ZPH&testcase=multiple-pipeline-all A possible duplicate exists as https://github.com/thesofproject/linux/issues/4329 but unconfirmed

**To Reproduce**
Happened during a multiple-pipeline-all test

**Reproduction Rate**
Has been seen once during a PR test

**Screenshots or console output**
Firmware:
```
[    0.580876] <inf> ipc: rx	: 0x13080003|0x0
[    0.581300] <inf> host_comp: comp:10 0x4000a no bytes to copy, available samples: 0, free_samples: 384
[    0.581316] <wrn> dai_comp: comp:9 0x40009 dai_copy(): nothing to copy
[    0.581845] <inf> ipc: rx	: 0x13080004|0x0
[    0.582305] <inf> host_comp: comp:10 0x4000a no bytes to copy, available samples: 0, free_samples: 384
[    0.582321] <wrn> dai_comp: comp:9 0x40009 dai_copy(): nothing to copy
[    0.582376] <inf> dai_comp: comp:8 0x40008 dai_playback_params() dest_dev = 0 stream_id = 0 src_width = 4 dest_width = 4
ASSERTION FAIL [aligned_addr >= (3187671040) && aligned_addr < (3187671040) + (1966080)] @ /srv/home/jenkins/workspace/sof_generic_build/zephyr/soc/xtensa/intTerminated
```
Kernel:
```
[ 1144.779088] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13080004|0x0: GLB_SET_PIPELINE_STATE
[ 1144.779285] kernel: snd_soc_rt711:rt711_sdw_write: rt711 sdw:0:025d:0711:00: [rt711_sdw_write] 00e0 <= 0003
[ 1144.779344] kernel: snd_soc_rt711:rt711_sdw_write: rt711 sdw:0:025d:0711:00: [rt711_sdw_write] 00f0 <= 0003
[ 1144.779346] kernel: snd_soc_rt711:rt711_clock_config: rt711 sdw:0:025d:0711:00: rt711_clock_config complete, clk_freq=4800000
[ 1144.780035] kernel: snd_soc_rt711:rt711_sdw_write: rt711 sdw:0:025d:0711:00: [rt711_sdw_write] 00e0 <= 0003
[ 1144.780097] kernel: snd_soc_rt711:rt711_sdw_write: rt711 sdw:0:025d:0711:00: [rt711_sdw_write] 00f0 <= 0003
[ 1144.780098] kernel: snd_soc_rt711:rt711_clock_config: rt711 sdw:0:025d:0711:00: rt711_clock_config complete, clk_freq=4800000
[ 1144.938621] kernel: soundwire_cadence:cdns_update_slave_status_work: soundwire_intel soundwire_intel.link.0: Slave status change: 0x40000
[ 1144.939054] kernel: soundwire_bus:sdw_handle_slave_alerts: rt711 sdw:0:025d:0711:00: Slave impl defined interrupt
[ 1144.939075] kernel: snd_soc_rt711:rt711_interrupt_callback: rt711 sdw:0:025d:0711:00: rt711_interrupt_callback control_port_stat=4
[ 1144.939725] kernel: soundwire_cadence:cdns_update_slave_status_work: soundwire_intel soundwire_intel.link.0: Slave status change: 0x20000
[ 1145.193507] kernel: snd_soc_rt711:rt711_sdw_read: rt711 sdw:0:025d:0711:00: [rt711_sdw_read] b921 0000 => 80000000
[ 1145.194989] kernel: snd_soc_rt711:rt711_sdw_read: rt711 sdw:0:025d:0711:00: [rt711_sdw_read] 7520 85a0 9c20 aca0 => 00000800
[ 1145.196236] kernel: snd_soc_rt711:rt711_sdw_read: rt711 sdw:0:025d:0711:00: [rt711_sdw_read] 7520 85a0 9c20 aca0 => 00000030
[ 1145.196262] kernel: snd_soc_rt711:rt711_jack_detect_handler: rt711 sdw:0:025d:0711:00: in rt711_jack_detect_handler, jack_type=0x3
[ 1145.196280] kernel: snd_soc_rt711:rt711_jack_detect_handler: rt711 sdw:0:025d:0711:00: in rt711_jack_detect_handler, btn_type=0x0
[ 1145.272645] kernel: snd_sof:sof_pcm_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 10 dir 1 cmd 0
[ 1145.272649] kernel: snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-tgl 0000:00:1f.3: trigger cmd: 0 state: 3
[ 1145.280553] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x13080004|0x0
[ 1145.280583] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Attempting to prevent DSP from entering D3 state to preserve context
[ 1145.280592] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 1145.280617] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc0000d83 rirb 00
[ 1145.280630] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 1145.280658] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Host IPC initiator: 0x93080004|0x0|0x0, target: 0x0|0x0|0x0, ctl: 0x3
[ 1145.280672] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 1145.280680] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 1145.280688] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: IPC timeout
[ 1145.280696] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[ 1145.280717] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x00000005: module: ROM, state: FW_ENTERED, running
[ 1145.280753] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: extended rom status:  0x5 0x0 0x0 0x0 0x0 0x0 0x0 0x1
[ 1145.280762] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 1145.280817] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_dai_trigger on iDisp2 Pin: -110
[ 1145.280834] kernel:  HDMI2: ASoC: error at dpcm_be_dai_trigger on HDMI2: -110
[ 1145.280845] kernel:  HDMI2: ASoC: trigger FE cmd: 1 failed: -110
```

lgirdwood, 2023-07-04:
@lyakh closed ? IIUC this one was recently fixed.

lyakh, 2023-07-05:
> @lyakh closed ? IIUC this one was recently fixed.

@lgirdwood yes, I think this was just another emanation of the pointer corruption bugs, fixed by #7660 - @keqiaozhang FYI, I understand we've never seen this specific form in CI and aren't seeing it now

closed 2023-07-05 by lyakh