thesofproject/sof #7537: [ADL] BT headset mic no sound when BT offload enabled

macchian, 2023-04-28:
**To Reproduce**
1. login in an Google account.
2. type Chrome://flags in browser, search "Bluetooth Offload", enabled it. (in order to keep the BT offload On in ChromeOS)
3. pair BT headset and connected BT headset.
4. create a Video call. e.g. Google Meet.
5. BT headset mic doesn't work on DUT. So that remote side can not hear the BT mic voice from DUT.

catch the sof fw logs:
[      178753.847064] (        1596.197876) c0 dw-dma                 src/drivers/dw/dma.c:1077 INFO dw_dma_avail_data_size() size is 0!
[      178791.815812] (          37.968750) c0 dai          14.62          src/audio/dai.c:951  WARN dai_copy(): nothing to copy
[      178842.284560] (          50.468746) c0 host         14.60         src/audio/host.c:470  INFO no bytes to copy, 0 avail in buffer, 64 free in DMA

**Reproduction Rate**
How often does the issue happen ? i.e. 1/10 (once in ten attempts), 1/1000 or all the time.
100%

**Expected behavior**
Host side, BT headset mic capturing work.

**Impact**
In any Video conference Call , client side can not hear the host side's BT headset mic voice.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
kernel: https://chromium.googlesource.com/chromiumos/third_party/kernel/+/refs/heads/chromeos-5.10
sof FW: https://github.com/thesofproject/sof/tree/adl-004-drop-stable

2) Name of the topology file
sof topology: sof-adl-rt1019-rt5682.tplg

3) Name of the platform(s) on which the bug is observed.
    * Platform: ADL

kv2019i, 2023-04-28:
@macchian This seems like the feature is not working at all. 

Does this happen on all devices (of this type?)
Did this work in the past on this device (if yes, what was the last known good version)?
Does the same sw+fw version work on other ADL devices?

FW log shows no data coming, so one thing to check is the pinmux setup in coreboot. Incorrectly set RX pin would look like this.

kv2019i, 2023-05-09:
@macchian The FW log in the bug report has usage of the non-BT PCM0 as well. And then for the BT PCM, there is:

[      180124.003259] (          38.593750) c0 pipe         14.63 ....../pipeline-params.c:239  INFO pipe params dir 1 frame_fmt 0 buffer_fmt 0 **rate 16000**
[      180147.128258] (          23.125000) c0 pipe         14.63 ....../pipeline-params.c:243  INFO pipe params stream_tag 2 channels 1 sample_valid_bytes 2 sample_
container_bytes 2
[      180460.201162] (         313.072906) c0 dai          14.62       src/ipc/ipc3/dai.c:103  INFO dai_data_config() dai type = 1 index = 2 dd 0x9e11e200
[      180538.326159] (          78.125000) c0 dai          14.62          src/audio/dai.c:414  INFO dai_capture_params() src_dev = 7 stream_id = 0 src_width = 2 dest_width = 2
[      180561.867825] (          23.541666) c0 dai          14.62          src/audio/dai.c:420  INFO dai_capture_params() fifo 0x77410

In system log, we can see BT opened for 8kHz:
2023-04-19T06:22:00.898553Z DEBUG cras_server[2091]: Configure alsa device hw:0,7 rate 8000Hz, 1 channels
2023-04-19T06:22:00.898599Z DEBUG cras_server[2091]: buffer size set to 32768
....
2023-04-19T06:22:00.899171Z DEBUG kernel: [   84.958734] sof-audio-pci-intel-tgl 0000:00:1f.3: **rate_min: 8000 rate_max: 8000**
2023-04-19T06:22:00.899172Z DEBUG kernel: [   84.958735] sof-audio-pci-intel-tgl 0000:00:1f.3: channels_min: 1 channels_max: 1
2023-04-19T06:22:00.899173Z DEBUG kernel: [   84.958764] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG

... so the kernel and FW logs don't match now. One shows BT offload configured for NB-SCO (8kHz), but FW log shows WB-SCO (16kHz).  Are the logs covering same attempt? 

This also shows up as BCLK selection. In FW log we can clearly see the 16kHz BCLK being used:
[      184485.878086] (          29.114582) c0 mn                 ./drivers/intel/ssp/mn.c:328  INFO find_mn for freq 38400000 **bclk 256000**
[      184510.773918] (          24.895832) c0 mn                 ./drivers/intel/ssp/mn.c:625  INFO bclk_rate 256000, *out_scr_div 150, m 1, n 1
[      184535.878084] (          25.104166) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:215  INFO ssp_bclk_prepare_enable(): sscr0 = 0xc0c0957f
[      184561.294750] (          25.416666) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:987  INFO ssp_early_start(): SSE set for SSP2
[      184735.721826] (         174.427078) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:1102 INFO ssp_trigger() cmd 1
[      184758.846825] (          23.125000) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:1001 INFO ssp_start()
[      186370.930094] (        1612.083252) c0 dw-dma                 src/drivers/dw/dma.c:1077 INFO dw_dma_avail_data_size() size is 0!
[      186408.690509] (          37.760414) c0 dai          14.62          src/audio/dai.c:951  WARN dai_copy(): nothing to copy

macchian, 2023-05-09:
[bt_capture_fw.log](https://github.com/thesofproject/sof/files/11433130/bt_capture_fw.log)
sorry to confuse you, my bad to upload the inconsistent NB and WB bt headsets logs.

attached the new logs aligning NB headset (8K rate) I catch today.
[bt_capture_kernel.log](https://github.com/thesofproject/sof/files/11433128/bt_capture_kernel.log)

kv2019i, 2023-05-10:
New log seems all fine.

@macchian Could you try this debugging patch and share the log:
https://github.com/kv2019i/sof/commit/87c012fc95d637f5b7288c3f128628bd6c45c0ee

The git commit message shows example print out on a system with valid data coming in from the SSP RX pins. This only covers the first 64 words.

macchian, 2023-05-11:
@kv2019i , applied the debug patch https://github.com/kv2019i/sof/commit/87c012fc95d637f5b7288c3f128628bd6c45c0ee
catch the fw logs for your analyzing.

[bypass_dma_fw.log](https://github.com/thesofproject/sof/files/11450001/bypass_dma_fw.log)

macchian, 2023-05-11:
> @kv2019i , applied the debug patch [kv2019i@87c012f](https://github.com/kv2019i/sof/commit/87c012fc95d637f5b7288c3f128628bd6c45c0ee) catch the fw logs for your analyzing.
> 
> [bypass_dma_fw.log](https://github.com/thesofproject/sof/files/11450001/bypass_dma_fw.log)

looks like no data ?

[      123165.411773] (          13.749999) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      123179.682605] (          14.270833) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      123193.536771] (          13.854166) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      123207.807604] (          14.270833) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      123221.765937] (          13.958333) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      123235.984686] (          14.218749) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      123250.047186] (          14.062499) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      123264.318019] (          14.270833) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      123278.484685] (          14.166666) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      123292.443017] (          13.958333) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      123306.661767] (          14.218749) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      123320.568016] (          13.906249) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      123334.786766] (          14.218749) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      123348.693015] (          13.906249) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      123362.963848] (          14.270833) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      123377.026347] (          14.062499) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff

macchian, 2023-05-11:
@kv2019i , as your request by using LBM topology, I see the same value

[      146156.296276] (         176.718750) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:1160 INFO ssp_trigger() cmd 1
[      146179.473358] (          23.177082) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:1056 INFO ssp_start()
[      162407.181047] (       16227.708008) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:173  INFO RX: 64 items, sample tick 4800

[      162421.347713] (          14.166666) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      162435.931045] (          14.583333) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      162450.462295] (          14.531249) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      162464.993544] (          14.531249) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      162479.681044] (          14.687499) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      162494.368543] (          14.687499) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      162509.160209] (          14.791666) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      162523.587292] (          14.427083) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      162538.378958] (          14.791666) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      162552.962291] (          14.583333) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      162567.701873] (          14.739583) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      162582.337290] (          14.635416) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      162596.920622] (          14.583333) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      162611.556038] (          14.635416) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      162626.139371] (          14.583333) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      162641.295621] (          15.156249) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff

[loopback.log](https://github.com/thesofproject/sof/files/11452346/loopback.log)

macchian, 2023-05-12:
@kv2019i , I capture the logs as your request with LBM topology + real bt headset connection (vim GMeet)

[lbm_bt.log](https://github.com/thesofproject/sof/files/11459104/lbm_bt.log)

It looks like the same fixed value in register...

[      212575.460303] (          25.260416) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:1042 INFO ssp_early_start(): SSE set for SSP2
[      212750.564463] (         175.104156) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:1160 INFO ssp_trigger() cmd 1
[      212773.741545] (          23.177082) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:1056 INFO ssp_start()
[      221073.324549] (        8299.583008) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:173  INFO RX: 64 items, sample tick 2400

[      221087.751631] (          14.427083) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      221102.126631] (          14.374999) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      221116.657880] (          14.531249) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      221131.293296] (          14.635416) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      221145.824546] (          14.531249) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      221160.616212] (          14.791666) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      221175.095378] (          14.479166) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      221189.782877] (          14.687499) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      221204.157877] (          14.374999) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      221218.897460] (          14.739583) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      221233.584959] (          14.687499) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      221248.116208] (          14.531249) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      221262.699541] (          14.583333) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      221277.074541] (          14.374999) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      221291.762040] (          14.687499) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      221306.189123] (          14.427083) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:175  INFO RX: fefff7ff fefff7ff fefff7ff fefff7ff
[      221445.199534] (         139.010406) c0 sa                          src/lib/agent.c:79   WARN validate(), ll drift detected, delta = 175266

kv2019i, 2023-05-12:
@macchian That doesn't make sense that LBM case shows those fixed bytes! Did you have a playback stream running when you capture the LBM log? If there was not concurrent playback, then the above would be still values from the physical RX pin.

I made one more debug patch. This creates a lot of log and can only be used with SSP2 (in this form), but it should get us more detail what happens on this device:
https://github.com/kv2019i/sof/tree/202305-bt-ssp-poll-rx2
Specific patch (can be applied on top of the previous ones) -> https://github.com/kv2019i/sof/commit/d1b136dfb642bedde0ea8d3772cf90a89ceb4de2

macchian, 2023-05-15:
> @macchian That doesn't make sense that LBM case shows those fixed bytes! Did you have a playback stream running when you capture the LBM log? If there was not concurrent playback, then the above would be still values from the physical RX pin.
I realized that I should run concurrent playback and capture command at a time. Here is my command when I catch the logs for your analyzing. The logs were WITHOUT patch [DNM][RD] drivers: ssp: very verbose debug for SSP2 capture 

[lbm_bt_print_rx_at_start.log](https://github.com/thesofproject/sof/files/11474008/lbm_bt_print_rx_at_start.log)

localhost ~ # aplay -Dplughw:0,7 a2002011001-e02-8kHz-mono.wav -vv | arecord -Dplughw:0,7 -r8000 -c1 -fS16_LE -vv loopback-rec.wav
Recording WAVE 'loopback-rec.wav' : Signed 16 bit Little Endian, Rate 8000 Hz, Mono
Plug PCM: Hardware PCM card 0 'sof-rt5682' device 7 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 1
  rate         : 8000
  exact rate   : 8000 (8000/1)
  msbits       : 16
  buffer_size  : 4000
  period_size  : 1000
  period_time  : 125000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 1000
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 4000
  silence_threshold: 0
  silence_size : 0
  boundary     : 9007199254740992000
  appl_ptr     : 0
  hw_ptr       : 0
Playing WAVE 'a2002011001-e02-8kHz-mono.wav' : Signed 16 bit Little Endian, Rate 8000 Hz, Mono
Plug PCM: Hardware PCM card 0 'sof-rt5682' device 7 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 1
  rate         : 8000
  exact rate   : 8000 (8000/1)
  msbits       : 16
  buffer_size  : 4000
  period_size  : 1000
  period_time  : 125000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 1000
  period_event : 0
  start_threshold  : 4000
  stop_threshold   : 4000
  silence_threshold: 0
  silence_size : 0
  boundary     : 9007199254740992000
  appl_ptr     : 0
  hw_ptr       : 0
############           +                           | 45%arecord: pcm_read:2161: read error: Input/output error
########+                                          | 14%aplay: pcm_write:2069: write error: Input/output error

> I made one more debug patch. This creates a lot of log and can only be used with SSP2 (in this form), but it should get us more detail what happens on this device: https://github.com/kv2019i/sof/tree/202305-bt-ssp-poll-rx2 Specific patch (can be applied on top of the previous ones) -> [kv2019i@d1b136d](https://github.com/kv2019i/sof/commit/d1b136dfb642bedde0ea8d3772cf90a89ceb4de2)

I catch the logs with LBM topology +commit d1b136d and aplay+arecord command as above. The ssp dump logs were below

[lbm_bt_dump_ssp_reg.log](https://github.com/thesofproject/sof/files/11474016/lbm_bt_dump_ssp_reg.log)

kv2019i, 2023-05-15:
Hmm, loopback test is now correct, but results is not at all what they should be. Did you have BT offload setup (connected to a headset while running this test? should not matter just checking). Here's one more patch (pushed to the same branch). If you could do the LBM test with this (no need to setup BT offload at all for this test, just run arecord+aplay with the LBM topology and this patch). https://github.com/kv2019i/sof/commits/202305-bt-ssp-poll-rx2 and patch https://github.com/kv2019i/sof/commit/bf804516c9457eef72d0808a54c74074c50602bc

kv2019i, 2023-05-15:
@macchian Ok, the 0xfefff7ff mystery solved. There was a bug in the dump logic and it tried to read one extra item from the RX FIFO when the FIFO in fact was empty. A fix pushed to the https://github.com/kv2019i/sof/commits/202305-bt-ssp-poll-rx2 branch as https://github.com/kv2019i/sof/commit/f448f6a3b5feb05fad9c78b827f7c8d36a626195 

So the findings again match with the results we got earlier -> we are not getting any  data from the SSP2 port. The dump_sssr.log also confirms there are no error conditions flagged by the SSP RX receiver, ADSP is just not getting data.

@macchian  Could you try to do a loopback test with unmodified cavs2.5-001-drop-stable FW (with topology modified to enable LBM)? Bypassing the DMA (like the R&D patches do) can interfere with the use-case, but a clean test with loopback should work.

- If the loopback test does not work -> this needs to be debugged first (e.g. with plain nocodec tplg, test loopback with SSP0 or SSP1)
- If the loopback works -> potentially issue with BCLK/FSYNC/RX signals.

kv2019i, 2023-05-16:
>     * If the loopback test does not work -> this needs to be debugged first (e.g. with plain nocodec tplg, test loopback with SSP0 or SSP1)

So this is the case, vanilla FW with loopback test fails

lgirdwood, 2023-07-04:
@kv2019i @juimonen can we close ?

macchian, 2023-07-05:
> @kv2019i @juimonen can we close ?

yes, I can close it. The CL submits for reviewing https://review.coreboot.org/c/coreboot/+/76247

closed 2023-07-05 by macchian