thesofproject/sof #7548: [ADL/RPL] [Bug] BT HFP offload capture stops working from second time

Vamshigopal, 2023-05-02:
**Description**
with BToffload enabled , when we connect HFP-only supported bt headset, mic stops working from second try,  from sof trace logs we see 
[    85209608.124491] (        1869.218628) c0 dw-dma                 src/drivers/dw/dma.c:1062 INFO dw_dma_avail_data_size() size is 0!
[    85209633.593240] (          25.468748) c0 dw-dma                 src/drivers/dw/dma.c:1128 ERROR dw_dma_get_data_size(): xrun detected
[dmesg.log](https://github.com/thesofproject/linux/files/11333182/dmesg.log)

[sof-logger.log](https://github.com/thesofproject/linux/files/11333132/sof-logger.log)

[btsnoop.log](https://github.com/thesofproject/linux/files/11333133/btsnoop.log)

**steps to reproduce**
1. Connect to an HFP-only bt headset
2. Start a youtube clip via BT output
3. Start mic recording via BT input
4. Stop mic recording via BT input
5. Start mic recording via BT input

**Expected**
 Mic should start at step 5

**Result**
 Mic fails to start 

**repro rate**
10/10

**SW Configuration**
kernel:  https://chromium.googlesource.com/chromiumos/third_party/kernel/+/refs/heads/chromeos-5.15
sof FW: https://github.com/thesofproject/sof/tree/cavs2.5-001-drop-stable
sof topology:  sof-adl-max98360a-rt5682.tplg

kv2019i, 2023-05-03:
Tested the same FW with nocodec tplg and the problem does not occur. Stopping and restarting SSP record while playback is kept running, works without issues with the nocodec tplg. Only delta between working and non-working configuration is the data format (48kHz/2ch/32bit versus 8kHz/1ch/16bit), which makes this very curious.

With the BT HFP configuration, I can also see the SSP RX FIFOs is getting filled, so the problem would appear to be with DW-DMA not moving data from SSP FIFO to SRAM (on second capture start).

kv2019i, 2023-05-03:
Narrowed it now down to:.
 - 8kHz 1ch 16bit -> fails the test sequence
 - 8kHz 2ch 16bit -> passes the sequence

Dumps of hw state in passing/failing cases have not revealed anything yet, so still no explanation why this happens.

kv2019i, 2023-05-04:
Minimal test setup to reproduce the issue on any cAVS2.5 system (two topmost patches):
https://github.com/kv2019i/sof/commits/202305-bt-test-ipc3 

No new findings yet. Multiple workarounds tried out, but no reliable fix yet.

kv2019i, 2023-05-05:
@Vamshigopal A draft fix now available at https://github.com/thesofproject/sof/pull/7564 . Please test if this helps in your original scenario. We will continue to test and see if we can find a a better fix still, but this seems to be fixing this specific scenario.

Vamshigopal, 2023-05-08:
@kv2019i  thanks for the fix, tested the fix #7564  with original scenario, recording works fine. checked start and stop captures mutiple times by playback is going on.

kv2019i, 2023-05-09:
Thanks @Vamshigopal for verifying. I tried one alternative approach to clean up the fix, but that didn't provide results so I now polished up the current fix and submitted for mainline as #7577 . When/if merged, I'll backport to the stable series.

kv2019i, 2023-05-12:
Fix merged to mainline, ok to close @Vamshigopal ?

sathyap-chrome, 2023-05-12:
Great. thanks @kv2019i 
Can you move the fix to cavs-2.5 release too

Vamshigopal, 2023-05-12:
> Fix merged to mainline, ok to close @Vamshigopal ?

Thanks @kv2019i , Yes we can close the issue.

Vamshigopal, 2023-05-30:
Hi @kv2019i 

Re-opened as we observe issue reproduced again with same error signature ,
```
[    65181836.472407] (          20.104166) c0 ssp-dai      1.2   /drivers/intel/ssp/ssp.c:1007 INFO ssp_start()
[    65182083.295313] (         246.822906) c0 sa                          src/lib/agent.c:96   WARN validate(), ll drift detected, delta = 85992
[    65182172.409893] (          89.114578) c0 dw-dma                 src/drivers/dw/dma.c:1062 INFO dw_dma_avail_data_size() size is 0!
[    65182197.149476] (          24.739582) c0 dw-dma                 src/drivers/dw/dma.c:1128 ERROR dw_dma_get_data_size(): xrun detected
[    65182219.180725] (          22.031250) c0 dai          14.72          src/audio/dai.c:884  ERROR dai_report_xrun(): overrun due to no space available
[    65182241.680724] (          22.500000) c0 dai          14.72          src/audio/dai.c:806  ERROR comp_overrun(): sink->free = 64, copy_bytes = 0
[    65182259.649473] (          17.968750) c0 pipe         14.73 ....../pipeline-stream.c:270  INFO pipe trigger cmd 6
[    65182293.607805] (          33.958332) c0 pipe         14.73 ....../pipeline-stream.c:143  ERROR pipeline_copy(): ret = -61, start->comp.id = 72, dir = 0
[    65182315.795304] (          22.187500) c0 pipe         14.73 ..../pipeline-schedule.c:208  ERROR pipeline_task(): xrun recovery failed! pipeline is stopped.

```

steps to reproduce

1. Connect to an HFP-only bt headset
2. Start a youtube clip via BT output
3. Start mic recording via BT input
4. Stop mic recording via BT input
5. repeat 3,4 steps for couple of times
6. reboot the device
7. start youtube pb via BT output 
8. repeat 3,4 
9. you will hit issue In second repeat.

Vamshigopal, 2023-05-30:
@kv2019i is it with nocodec tplg ?

kv2019i, 2023-05-31:
Updated patch -> https://github.com/thesofproject/sof/pull/7717

ujfalusi, 2023-06-07:
I think there is another angle to this issue which is brought to light by the TX FIFO drain simplifications.
The setup is
**SSP1** - **clock provider**, playback only to speaker, want MCLK as **19200000**
**SSP2** - **clock receiver**, playback/capture to BT, want MCLK as  **38400000**

We see this sequence in case of a failure:
SSP2 start (sets MCLK to 38400000)
```
ssp-dai      1.2   /drivers/intel/ssp/ssp.c:1111 INFO ssp_trigger() cmd 7
ssp-dai      1.2   /drivers/intel/ssp/ssp.c:900  INFO ssp_pre_start()
mn                 ./drivers/intel/ssp/mn.c:259  INFO mclk_rate 38400000, mclk_source_clock 1
mn                 ./drivers/intel/ssp/mn.c:220  INFO mclk_id 0 mdivr_val 1
mn                 ./drivers/intel/ssp/mn.c:330  INFO find_mn for freq 38400000 bclk 128000
mn                 ./drivers/intel/ssp/mn.c:628  INFO bclk_rate 128000, *out_scr_div 300, m 1, n 1
ssp-dai      1.2   /drivers/intel/ssp/ssp.c:234  INFO ssp_bclk_prepare_enable(): sscr0 = 0xc0c12b7f
ssp-dai      1.2   /drivers/intel/ssp/ssp.c:993  INFO ssp_early_start(): SSE set for SSP2
ssp-dai      1.2   /drivers/intel/ssp/ssp.c:1111 INFO ssp_trigger() cmd 1
ssp-dai      1.2   /drivers/intel/ssp/ssp.c:1008 INFO ssp_start()
```
SSP2 start (clocks have been configured for SSP2 already)
```
ssp-dai      1.2   /drivers/intel/ssp/ssp.c:1111 INFO ssp_trigger() cmd 7
ssp-dai      1.2   /drivers/intel/ssp/ssp.c:900  INFO ssp_pre_start()
ssp-dai      1.2   /drivers/intel/ssp/ssp.c:993  INFO ssp_early_start(): SSE set for SSP2
ssp-dai      1.2   /drivers/intel/ssp/ssp.c:1111 INFO ssp_trigger() cmd 1
ssp-dai      1.2   /drivers/intel/ssp/ssp.c:1008 INFO ssp_start()
```
SSP2 RX stop
SSP2 start
```
ssp-dai      1.2   /drivers/intel/ssp/ssp.c:1111 INFO ssp_trigger() cmd 7
ssp-dai      1.2   /drivers/intel/ssp/ssp.c:900  INFO ssp_pre_start()
ssp-dai      1.2   /drivers/intel/ssp/ssp.c:993  INFO ssp_early_start(): SSE set for SSP2
ssp-dai      1.2   /drivers/intel/ssp/ssp.c:1111 INFO ssp_trigger() cmd 1
ssp-dai      1.2   /drivers/intel/ssp/ssp.c:1008 INFO ssp_start()
```
SSP2 TX stop

At this point we have `SSP2 RX` still running, **MCLK** is configured for **38400000**

SSP1 start (fails to set the MCLK to 19200000)
```
ssp-dai      1.1   /drivers/intel/ssp/ssp.c:1111 INFO ssp_trigger() cmd 7
ssp-dai      1.1   /drivers/intel/ssp/ssp.c:900  INFO ssp_pre_start()
mn                 ./drivers/intel/ssp/mn.c:180  INFO MCLK 19200000, source = 1
mn                 ./drivers/intel/ssp/mn.c:192  ERROR Can't set MCLK 0 to 19200000, it is already configured to 38400000
ssp-dai      1.1   /drivers/intel/ssp/ssp.c:161  ERROR ssp_mclk_prepare_enable(): invalid mclk_rate = 19200000 for mclk_id = 0
ssp-dai      1.1   /drivers/intel/ssp/ssp.c:993  INFO ssp_early_start(): SSE set for SSP1
demux        1.2        src/audio/mux/mux.c:693  INFO mux_trigger(), command = 1
ssp-dai      1.1   /drivers/intel/ssp/ssp.c:1111 INFO ssp_trigger() cmd 1
ssp-dai      1.1   /drivers/intel/ssp/ssp.c:1008 INFO ssp_start()
```
Note that the MCLK request fails since the same clock is already used by `SSP2` and it is set to **38400000**, the MCLK request fail is handled by skipping the BCLK setup for SSP1 but the error from `ssp_pre_start()` is ignored, so we go on and start the SSP1 anyways.
This will lead to:
```
dw-dma                 src/drivers/dw/dma.c:1093 INFO dw_dma_free_data_size() size is 0!
dw-dma                 src/drivers/dw/dma.c:1128 ERROR dw_dma_get_data_size(): xrun detected
dai          1.6            src/audio/dai.c:881  ERROR dai_report_xrun(): underrun due to no data available
dai          1.6            src/audio/dai.c:775  ERROR comp_underrun(): dev->comp.id = 6, source->avail = 192, copy_bytes = 0
pipe         1.7   ....../pipeline-stream.c:270  INFO pipe trigger cmd 6
pipe         1.7   ....../pipeline-stream.c:143  ERROR pipeline_copy(): ret = -61, start->comp.id = 6, dir = 1
pipe         1.7   ..../pipeline-schedule.c:187  ERROR pipeline_task(): xrun recovery failed! pipeline is stopped.
```
Which is not a surprise without BCLK setup for a clock provider.

I think this setup is not correct. SSP1 and SSP2 can be used at the same time and they want the very same MCLK to run in different speeds. A clock cannot be 38400000 and 19200000 at the same time.
Either both SSPs should be asking for the same MCLK rate
or
the two SSP should use different mclk_id from topology (if it is possible at all).

kv2019i, 2023-06-14:
Closing this issue, let's follow-up on MCLK configuration question in separate bug #7803 as that seems to be a separate issue (albeit uncovered by same testing steps).

closed 2023-06-14 by kv2019i