thesofproject/sof #7553: Mux component returning 0 bytes of data during device switching between Headset and speaker with Mux usecase.

maruthi-machani, 2023-05-03:
**Describe the bug**
During switching between the speaker and headset, observed at one instance of time mux seeing both speaker and HS active but the data in speaker is not yet placed.
This condition will occur when HS is in pending state and DSP receives IPC to start the speaker before the execution of mux pipe task. After IPC when mux pipe task gets execution it will see data only in the HS.
Mux will process the data only if data available at both the input sources, else will return as no data available.
Eventhough the HS data is present, mux is returning as no data avail.
Attaching the log.
[debuglog_Initial8.txt](https://github.com/thesofproject/sof/files/11379895/debuglog_Initial8.txt)

[image]

**To Reproduce**
1) play youtube on headset
2) open browser, start google meet, from google meet settings switch the audio device to Headset and speaker (repeat untill issue is reproduced).

**Reproduction Rate**
issue occurs at 1/30 device switchings.

**Detatils of Topology**
Three pipelines are used to support the mux usecase.
two input host pipelines connected to the mux-dai pipeline.
Attaching the topology for reference.
[sof-acp-rmb-tdm8ch-dmic4ch.m4.txt](https://github.com/thesofproject/sof/files/11379869/sof-acp-rmb-tdm8ch-dmic4ch.m4.txt)
[image]

lgirdwood, 2023-07-04:
@maruthi-machani is this now resolved with recent v2.6 fixes for mux ?

maruthi-machani, 2023-07-05:
@lgirdwood this issue is not related to the recent patches merged https://github.com/thesofproject/sof/pull/7590.

lgirdwood, 2023-07-05:
@maruthi-machani can you describe how the pipelies above are scheduled, which use DMA IRQ and whic use timer IRQ.

maruthi-machani, 2023-07-06:
Details of pipelines scheduling, mentioned by @bhiregoudar in
https://github.com/thesofproject/sof/issues/7525#issuecomment-1621859267

closed 2025-07-24 by github-actions[bot]