thesofproject/sof #7562: module_adapter.c: warning: dereference of NULL mod->output_buffers[i].data

marc-hb, 2023-05-04:
@plbossart asked me to try `gcc -fanalyzer`. One of the few warnings is this. @ranj063 is this a false positive? Sorry but your name is on the `git blame` the most :-)

**To Reproduce**
```
cd sof
git checkout 43b08c62b523
rm -rf ../build-*
./scripts/xtensa-build-zephyr.py tgl -C=-DEXTRA_CFLAGS=-fanalyzer 
```

The real version has colors!

```
SOF/sof/src/audio/module_adapter/module_adapter.c:336:45: warning: dereference of NULL '0' [CWE-476] [-Wanalyzer-null-dereference]
  336 |                 mod->output_buffers[i].data =
      |                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
  337 |                         (__sparse_force void __sparse_cache *)rballoc(0, SOF_MEM_CAPS_RAM,
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  338 |                                                                       md->mpd.out_buff_size);
      |                                                                       ~~~~~~~~~~~~~~~~~~~~~~
  'module_adapter_prepare': events 1-14
    |
    |  172 |         if (ret) {
    |      |            ^
    |      |            |
    |      |            (1) following 'false' branch (when 'ret == 0')...
    |......
    |  182 |         sink = list_first_item(&dev->bsink_list, struct comp_buffer, source_list);
    |      |         ~~~~
    |      |         |
    |      |         (2) ...to here
    |......
    |  194 |         if (dev->state == COMP_STATE_ACTIVE)
    |      |            ~
    |      |            |
    |      |            (3) following 'false' branch...
    |......
    |  198 |         ret = comp_set_state(dev, COMP_TRIGGER_PREPARE);
    |      |         ~~~ 
    |      |         |
    |      |         (4) ...to here
    |  199 |         if (ret < 0)
    |      |            ~
    |      |            |
    |      |            (5) following 'false' branch (when 'ret >= 0')...
    |......
    |  202 |         if (ret == COMP_STATUS_STATE_ALREADY_SET) {
    |      |         ~~ ~
    |      |         |  |
    |      |         |  (7) following 'false' branch (when 'ret != 1')...
    |      |         (6) ...to here
    |......
    |  207 |         mod->deep_buff_bytes = 0;
    |      |         ~~~ 
    |      |         |
    |      |         (8) ...to here
    |......
    |  241 |         if (mod->num_input_buffers) {
    |      |            ~
    |      |            |
    |      |            (9) following 'true' branch...
    |  242 |                 mod->input_buffers =
    |      |                 ~~~
    |      |                 |
    |      |                 (10) ...to here
    |......
    |  245 |                 if (!mod->input_buffers) {
    |      |                    ~
    |      |                    |
    |      |                    (11) following 'false' branch...
    |......
    |  254 |         if (mod->num_output_buffers) {
    |      |         ~~ ~
    |      |         |  |
    |      |         |  (13) following 'false' branch...
    |      |         (12) ...to here
    |......
    |  264 |                 mod->output_buffers = NULL;
    |      |                 ~~~
    |      |                 |
    |      |                 (14) ...to here
    |
  'module_adapter_prepare': events 15-17
    |
    |  264 |                 mod->output_buffers = NULL;
    |      |                                     ^
    |      |                                     |
    |      |                                     (15) '0' is NULL
    |......
    |  271 |         if (mod->simple_copy)
    |      |            ~                         
    |      |            |
    |      |            (16) following 'false' branch...
    |......
    |  281 |         if (md->mpd.in_buff_size > mod->period_bytes) {
    |      |         ~~                           
    |      |         |
    |      |         (17) ...to here
    |
  'module_adapter_prepare': event 18
    |
    |SOF/sof/xtos/include/sof/list.h:93:40:
    |   93 |         for (item = (list)->next; item != (list); item = item->next)
SOF/sof/src/audio/module_adapter/module_adapter.c:320:9: note: in expansion of macro 'list_for_item'
    |  320 |         list_for_item(blist, &dev->bsource_list) {
    |      |         ^~~~~~~~~~~~~
    |
  'module_adapter_prepare': event 19
    |
    |  334 |         i = 0;
    |      |         ^
    |      |         |
    |      |         (19) ...to here
    |
  'module_adapter_prepare': event 20
    |
    |SOF/sof/xtos/include/sof/list.h:93:40:
    |   93 |         for (item = (list)->next; item != (list); item = item->next)
SOF/sof/src/audio/module_adapter/module_adapter.c:335:9: note: in expansion of macro 'list_for_item'
    |  335 |         list_for_item(blist, &dev->bsink_list) {
    |      |         ^~~~~~~~~~~~~
    |
  'module_adapter_prepare': events 21-24
    |
    |  336 |                 mod->output_buffers[i].data =
    |      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    |      |                 |  |               |        |
    |      |                 |  |               |        (24) dereference of NULL '*mod.output_buffers + (unsigned int)i * 8'
    |      |                 |  |               (23) '0' is NULL
    |      |                 |  (22) '0' is NULL
    |      |                 (21) ...to here
    |  337 |                         (__sparse_force void __sparse_cache *)rballoc(0, SOF_MEM_CAPS_RAM,
    |      |                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    |  338 |                                                                       md->mpd.out_buff_size);
    |      |                                                                       ~~~~~~~~~~~~~~~~~~~~~~
    |

```

paulstelian97, 2023-05-08:
mod->output_buffers cannot be NULL, yet the analyzer cannot figure this out (it's a conservative analyzer and null checks aren't sufficient for any non-local variable for it to _know_ it's not null; the fact that _we_ know that mod->output_buffers won't be concurrently modified isn't enough for the compiler).

To make the analyzer happy one would need to make a local copy of mod->output_buffers and do a NULL check (eventually an assert, possibly) before using said local copy.

marc-hb, 2023-05-08:
> mod->output_buffers cannot be NULL

Not even with a (future?) mistake? Where is this initialized?

> To make the analyzer happy one would need to make a local copy of mod->output_buffers and do a NULL check (eventually an assert, possibly) before using said local copy.

This sounds like it would be a small change, wouldn't it? 

> Do we do that or do we make a sort of ignore on this issue?

The `-fanalyzer` flag is brand new to me, I don't know if you can exclude specific spots. @plbossart ?

Note this is only one of a very small number of warnings right now, so it's probably worth some effort. `-fanalyzer` is being added to the kernel CI:
- https://github.com/thesofproject/linux/pull/4340

lgirdwood, 2023-07-04:
How can we tell GCC analyser about false positives ? i.e. so we can ignore these in the future

marc-hb, 2023-07-05:
@plbossart any idea? I looked at https://developers.redhat.com/articles/2022/04/12/state-static-analysis-gcc-12-compiler and https://gcc.gnu.org/onlinedocs/gcc/Static-Analyzer-Options.html and I didn't find anything about handling false positives.

lgirdwood, 2023-07-05:
> @plbossart any idea? I looked at https://developers.redhat.com/articles/2022/04/12/state-static-analysis-gcc-12-compiler and https://gcc.gnu.org/onlinedocs/gcc/Static-Analyzer-Options.html and I didn't find anything about handling false positives.

ok, lets keep this under TBD - the GCC folks will probably get around to this.

closed 2025-07-24 by github-actions[bot]