thesofproject/sof #7596: [BUG][MTL] failed to create module pipeline when enabling debug build on mtl-005-drop-stable

keqiaozhang, 2023-05-12:
**Describe the bug**
If we enable the debug build on mtl-005-drop-stable branch, the basic audio functionality will be broken.

For example, when checking playback, it will failed with:
```
[  215.475729] kernel: snd_sof:sof_ipc4_widget_setup: sof-audio-pci-intel-mtl 0000:00:1f.3: Create widget pipeline.15 instance 0 - pipe 15 - core 0
[  215.475737] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x11000007|0x1: GLB_CREATE_PIPELINE
[  215.976534] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x11000007|0x1
[  215.976565] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[  215.976600] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to create module pipeline.15
[  215.976616] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Failed to set up connected widgets
[  215.976637] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed widget list set up for pcm 31 dir 0
[  215.976647] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -110
[  215.976740] kernel:  Deepbuffer Jack Out: ASoC: error at __soc_pcm_hw_params on Deepbuffer Jack Out: -110
[  215.976767] kernel: sdw_deprepare_stream: subdevice #0-Playback: inconsistent state state 1
[  215.976782] kernel:  Deepbuffer Jack Out: ASoC: error at dpcm_fe_dai_hw_params on Deepbuffer Jack Out: -110
[  215.976800] kernel: snd_sof:sof_pcm_hw_free: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: free stream 31 dir 0
[  215.976835] kernel: sdw_deprepare_stream: subdevice #0-Playback: inconsistent state state 6
[  215.977766] kernel: snd_sof:sof_pcm_close: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: close stream 31 dir 0
```
But if we disable the debug build, everything works fine.

**To Reproduce**
Build mtl firmware binaries with `-d` option.

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF: mtl-005-drop-stable
3) Name of the platform(s) on which the bug is observed.
    * Platform: all MTL platforms.

[dmesg-sdw.txt](https://github.com/thesofproject/sof/files/11458523/dmesg-sdw.txt)

[dmesg-hda.txt](https://github.com/thesofproject/sof/files/11458524/dmesg-hda.txt)
[dmesg-nocodec.txt](https://github.com/thesofproject/sof/files/11458525/dmesg-nocodec.txt)

wszypelt, 2023-05-12:
We lower the priority, because it's a debug option, we reduce to p2
What is the reason for exposing this to p1?

kv2019i, 2023-05-12:
@wszypelt @keqiaozhang @abonislawski This is likely an assert firing (enabled with debug builds), so needs to be investigated.

marc-hb, 2023-05-15:
> We lower the priority, because it's a debug option, we reduce to p2

The priority could be lowered if this was some complex interaction affecting some corner case in debug only.

"Nothing works with the debug build" looks definitely like a P1 to me! The debug build is mostly identical to the production build.

aborisovich, 2023-05-17:
> @wszypelt @keqiaozhang @abonislawski This is likely an assert firing (enabled with debug builds), so needs to be investigated.

Reproduced the issue using test:
`"tests/ace/fw_113_ssp/test_02_ssp_internal_loop.py::TestSspInternalLoop::test_113_02_simple_internal_loopback[16000Hz_16in16bit_4ch-ssp_1_to_ssp_1]"` .
On FPGA there had been Zephyr kernel timer timeout during setting pipeline state to running (during copier copy operation).
On RVP issue is similar but occurs in later time in test - when setting pipeline state to paused (also during copier copy operation). For non-debug builds those issues are not reproduced.

> @wszypelt @keqiaozhang @abonislawski This is likely an assert firing (enabled with debug builds), so needs to be investigated.

It looks like there is no assertions firing, we are just late with audio processing due to performance overhead of debug builds.
I'm still looking for possible deadlocks/delays that could cause timeouts.

@marc-hb 

> "Nothing works with the debug build" looks definitely like a P1 to me! The debug build is mostly identical to the production build.

This is not true. Asserts add a lot of checks that provide performance penalty that may cause delays in processing.
We should certainly make sure this feature works and test it on PRs but I would restrain from calling it part of the release candidate. Anyway, work in progress.

mengdonglin, 2023-07-27:
Closing as this issue cannot be reproduced by mtl-005.0.3

closed 2023-07-27 by mengdonglin