thesofproject/sof #7603: [BUG] SSP loopback test fails if mtrace is NOT running with BT config

kv2019i, 2023-05-12:
**Describe the bug**
SSP loopback test fails on MTL if mtrace is NOT running.

For example, when checking playback, it will failed with:
```
alsabat -Dhw:0,2 -r48000 -c2 -fS16_LE
alsa-utils version 1.2.6

Entering playback thread (ALSA).
Get period size: 2048  buffer size: 16384
Playing generated audio sine wave
Entering capture thread (ALSA).
Get period size: 2048  buffer size: 16384
Recording ...
Write PCM device error: Input/output error(-5)
Exit playback thread fail: -5
```

If one runs mtrace-reader.py on the background, error does not happen.

**To Reproduce**
See above alsabat, use BT LBM topology

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF: mtl-005-drop-stable
    * topology: sof-nocodec-bt-lbm.tplg
2) kernel: sof-dev 90c8074e2cea (daily image for 20230512)
3) Name of the platform(s) on which the bug is observed.
    * Platform: all MTL platforms.

mwasko, 2023-05-15:
@wszypelt can you try to reproduce it in FW tests?

kv2019i, 2023-05-16:
Reproduces also with SOF main ( a675edf8e7eacff48699d3c4daec99963f19f868 ), so not specific to mtl-005 branch.

kv2019i, 2023-05-16:
The error only happens with 8kHz/16kHz 1ch test cases:
alsabat -Dhw:0,2 -r8000 -c1 -fS16_LE
alsabat -Dhw:0,2 -r16000 -c1 -fS16_LE

These match the HFP/SCO Bluetooth cases. The 48kHz 2ch case works (with and without mtrace):
alsabat -Dhw:0,2 -r48000 -c2 -fS16_LE
-> OK!

It seems the playback direction is the problematic one. This alone is enough to trigger the problem.
aplay -Dhw:0,2 -r8000 -c1 -fS16_LE -traw /dev/urandom 

With mtrace running, error is not hit. As a side-effect, it's not possible to get logs out of what happens in FW in the failing case (as there are no logs printed out).

kv2019i, 2023-05-16:
@juimonen suspected the link clocks. fix here:
https://github.com/thesofproject/sof/pull/7619

With that, the test is stll failing, but indeed the playback time is not correct:

```
time aplay -Dhw:0,2 -r48000 -c2 -fS16_LE -traw -d15 /dev/urandom
Playing raw data '/dev/urandom' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo

 

real    0m30.160s
user    0m0.005s
sys     0m0.049s
```

I.e. playback of 15sec of audio takes 30sec of wallclock time.

kv2019i, 2023-05-16:
One bug found in topology, the link clock source was incorrectly selected for LBM topology (0, should be 1). But even with this fixed, the playback time is still wrong.

kv2019i, 2023-05-17:
Now managed to get the error with mtrace running so can provide a log now. It seems the SSP-DW start is failing and this happens at higher reproduction rate when mtrace is NOT running. But you can still hit the issue even with mtrace.
[mtrace.txt](https://github.com/thesofproject/sof/files/11495543/mtrace.txt)

kv2019i, 2023-05-17:
It seems this bug is fixed by this commit in mtl-005-drop-stable -> https://github.com/thesofproject/sof/commit/87dc723abd965caeba19ead34f6d754f60e5672c
I was originally testing with mtl-005-rc1 and that does not have the patch (neither does main).
FYI @softwarecki @abonislawski 

Removing mtl-005 tag , but keeping bug open to ensure this gets fixed in main.

keqiaozhang, 2023-06-12:
@kv2019i https://github.com/thesofproject/sof/pull/7654 was merged, can we close this bug?

kv2019i, 2023-06-12:
Yes, let's close this. The delays with L1 exits are tracked in https://github.com/thesofproject/sof/issues/7646 and https://github.com/thesofproject/linux/issues/4408

closed 2023-06-12 by kv2019i