thesofproject/sof #7614: Zephyr main branch requires 0.16.x Zephyr SDK

marc-hb, 2023-05-15:
Since zephyr commit https://github.com/zephyrproject-rtos/zephyr/commit/9fc99928caf04c3ae78 and https://github.com/zephyrproject-rtos/zephyr/commit/49d0ad55206aa, the Zephyr main branch is not compatible with the Zephyr SDK 0.15.x anymore, it now requires 0.16.x. The Zephyr main branch now depends on different (and correct) toolchain names which are not present in 0.15.x

Every PR trying to build the Zephyr main branch (a.k.a. "zmain") started to fail like this, example in https://github.com/thesofproject/sof/actions/runs/4984592530

```
CMake Error at /zep_workspace/zephyr/cmake/compiler/gcc/target.cmake:12 (message):
  C compiler
  /home/cwd_user/zephyr-sdk-0.15.2/xtensa-intel_ace15_mtpm_zephyr-elf/bin/xtensa-intel_ace15_mtpm_zephyr-elf-gcc
  not found - Please check your toolchain installation
```

Fixing this for Linux compilation should be more or less a one-line change in the `zephyr/docker-run.sh` script (example in previous commit 11441d85409)

The Windows compilation that we've enjoyed thanks to @aborisovich's heroic work (commit df1ba222306d) is unfortunately a bit more complicated.

Note upgrading will also help with OpenSSL 1.x removal where this has already been discussed:
- https://github.com/thesofproject/rimage/pull/157

marc-hb, 2023-05-15:
P1 because this will block any Zephyr uprev past the commits listed above.

iuliana-prodan, 2023-05-16:
I know about this commit https://github.com/zephyrproject-rtos/hal_xtensa/commit/e636ce8fdf334addaae37fb2417dc2cb8bf7b50a which I reverted because it required an update in zephyr sdk - see [comment](https://github.com/zephyrproject-rtos/zephyr/pull/57084#issuecomment-1518092809)
I see now that [Zephyr SDK 0.16.1](https://github.com/zephyrproject-rtos/sdk-ng/releases) has support for xtensa-intel_ace15_mtpm_zephyr-elf. 

Wouldn't these 2 updates help?

marc-hb, 2023-05-16:
@iuliana-prodan this is an "infrastructure" task, there's no problem with the code I know of (yet).

closed 2023-05-19 by kv2019i