thesofproject/sof #7615: [BUG] [MTL] sound from deep buffer playback device is barely audible

yongzhi1, 2023-05-15:
**Describe the bug**
The volume from deepbuf playback is very low.

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug)

Plug in headset and run
```
 aplay -Dhw:0,31  -r 48000 -f s16 -c 2 /tmp/sample.wav
```
**Reproduction Rate**
100%

**Expected behavior**
The sound level should be on a par with headset.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {https://chromium.googlesource.com/chromiumos/third_party/kernel/+/5d271868}
    * SOF: {https://github.com/thesofproject/sof/tree/mtl-005-drop-stable }
2) Name of the topology file
    * Topology: {sof-mtl-max98357a-rt5682-ssp2-ssp0.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {MTL/rex}

[dmesg-deep-buf-playback.txt](https://github.com/thesofproject/sof/files/11482039/dmesg-deep-buf-playback.txt)
[mtrace-deepbuf-playback-volume-low.txt](https://github.com/thesofproject/sof/files/11482050/mtrace-deepbuf-playback-volume-low.txt)

RanderWang, 2023-05-16:
The ipc message data is correct. I first thought it was caused by format conversion, but I don't find any wrong

keqiaozhang, 2023-05-17:
I checked the deep buffer pipeline on MTL-NOCODEC_RVP, the audio quality is good with `gain.15.1 Deepbuffer [dB gain: 0.00, 0.00]`

[image]

Test cmd:
```aplay -Dhw:0,31 -f dat -c2 sine.wav  & arecord -Dhw:0,0 -f dat -c 2 record.wav -vvv -d 10```

plbossart, 2023-05-17:
not sure I understand @keqiaozhang, was this a case where the gain was not 0dB? We've had this discussion multiple times, I thought we had something in the test setup when the gains are always reset to 0dB

yongzhi1, 2023-05-17:
Sorry, the low volume turned out to be an test bench glitch, thanks @RanderWang for checking the audio formats and @keqiaozhang for verification, will open an seperate bug to track the deep-buf noise issue, thanks!!