thesofproject/sof #7646: [MTL] DMIC capture xruns on MTLP_RVP_HDA with GNA in D3hot

plbossart, 2023-05-18:
https://sof-ci.01.org/linuxpr/PR4349/build4857/devicetest/index.html shows multiple regressions on DMIC.

plbossart, 2023-05-18:
Additional data points with more experiments:

cSOF MTL-005 does NOT show the problem with the 'nocodec' build of the kernel

cSOF MTL-005 DOES show the problem with the 'regular' build of the kernel with
a) standard HDaudio + DMIC topology or
b the sof-mtl-nocodec topology selected with a kernel parameter `options snd-sof sof_debug=0x401`

In both case disabling pm_runtime with ` options snd_sof_pci sof_pci_debug=0x1` does not help: it's not a suspend-resume issue.

So this points to an intersection of UEFI setup, kernel HDaudio DMA initialization and cSOF behavior.

plbossart, 2023-05-23:
the statement "cSOF MTL-005 does NOT show the problem with the 'nocodec' build of the kernel" was incorrect, it was due to a missing kernel configuration that didn't include PMC settings. This was corrected with https://github.com/thesofproject/kconfig/pull/86

To recap, the problem happens with all kernel builds when:
a) GNA disabled in UEFI BIOS or GNA put to D3hot by PMC core, and
b) cSOF MTL-005 firmware is used

The problem does not happen with 
a) GNA disabled in Coreboot, or
b) IPC4 reference firmware.

@kv2019i @mwasko does this clarify the problem statement?

abonislawski, 2023-05-31:
@plbossart Im trying to reproduce this on our linux platform but I cant get error on DMIC arecord

Setup:
MTL-NOCODEC
Disabled in bios: GNA, IPU, VPU
Kernel: linux-image-gc8f67b29a153_6.4.0-rc1-nocodec-topic-sof-dev_amd64.deb (2023-05-30)
FW: MTL-005
TPLG: MTL-005

also tested with options snd-sof sof_debug=0x401

What could be missing here?

plbossart, 2023-05-31:
@abonislawski you are probably not using the 'right' image. I don't know what this tag c8f67b29a153 is. Just use a regular 'default' image and the 0x401 image, you'll see.

abonislawski, 2023-05-31:
Its just newest daily build from sof-ci:

Linux Branch: 
topic/sof-dev
Linux Commit: 
c8f67b29a153
Kconfig Branch: 
master
Kconfig Commit: 
4b61be9d5472
https://sof-ci.ostc.intel.com/#/buildresult/linuxbuildresult/188

Which one is "default"? Its hard to start without setup description.
Also, is there somewhere build with "mtl_set_device_d3(MTL_GNA_PCI_DEV)" to download?

plbossart, 2023-05-31:
follow the download button and use linux-image-gc8f67b29a153_6.4.0-rc1-default-topic-sof-dev_amd64.deb

plbossart, 2023-05-31:
oh wait, we've actually reverted the problematic commit to make things work in CI.

So you need to compile yourself topic/sof-dev with commit https://github.com/thesofproject/linux/pull/4349/commits/743c0c9b590899b78edcd24bf1999f948141d8ca reverted

abonislawski, 2023-05-31:
oh, so disabling GNA in BIOS is not enough? 
Good we clarified this, let me try

plbossart, 2023-05-31:
@abonislawski with GNA disabled you should see the issue. @fredoh9 disabled it on our device via the System Agent (SA) configuration. GNA device (B0:D8:F0) was moved to `<Disabled>`

fredoh9, 2023-06-01:
worked with @abonislawski @plbossart

latest mtl-005-drop-stable doesn't have the issue. When we first had this issue, mtl-005-drop-stable branch had 36a7a817fb8b
I did git bisect, it points this fix the problem.

```
commit 87dc723abd965caeba19ead34f6d754f60e5672c 
Author: Adrian Warecki <adrian.warecki@intel.com> 
Date:   Wed May 10 14:28:18 2023 +0200 

    dai: always reload dma ll chain 
     
    Reload the dma descriptor chain even if there are no bytes to copy. This 
    may result in the loss of some data, but at least the dma will not stop. 
     
    Signed-off-by: Adrian Warecki <adrian.warecki@intel.com> 

src/audio/dai-zephyr.c | 1 + 
1 file changed, 1 insertion(+) 
```

mengdonglin, 2023-06-01:
@abonislawski Does this mean putting GNA in D3 could make dai DMA stop?

abonislawski, 2023-06-01:
@mengdonglin thats why title says "DMIC capture xruns" but this is more likely side effect of host issues on linux in corner case, thats why this dma reload commit helps in several issues but was never needed on windows. Additional issue thread will follow up to investigate this.

One more thing not mentioned before on github: on windows GNA was always in D3 with SOF MTL tests and no issues here.

marc-hb, 2023-06-01:
> I did git bisect, it points https://github.com/thesofproject/sof/commit/87dc723abd965caeba19ead34f6d754f60e5672c fix the problem.

Note this is a cherry-pick [-x] of commit https://github.com/thesofproject/sof/commit/5ad5da5bc757 on the main branch. The commit message is different and missing the reference to the original commit but it is the same one-line change.

lgirdwood, 2023-07-04:
@plbossart can this be closed ?

mengdonglin, 2023-08-01:
@abonislawski You fixed this issue for mtl-005-drop-stable with https://github.com/thesofproject/zephyr/commit/c31e683295ebd77789f6b4e95c68b073b53b0112 
but I cannot find this fix in upstream Zephyr. However, we cannot reproduce this issue on MTLP_RVP_HDA recently (ww30 BIOS). Maybe the BIOS fix has been integrated and so FW fix is no longer needed?

mengdonglin, 2023-08-03:
@plbossart @abonislawski @lgirdwood 
BIOS fix to set L1 exit threshold to be 1/4 FIFO size were applied in early of June. I think that's why we cannot reproduce this issue with recent BIOS and SOF main branch. 

SOF main branch doesn't have FW fix https://github.com/thesofproject/zephyr/commit/c31e683295ebd77789f6b4e95c68b073b53b0112 for mtl-005-drop-stable branch. 

My understanding is that either BIOS or FW fix is enough. So I will close this bug. Please reopen it if you have any concerns.

closed 2023-08-03 by mengdonglin

mengdonglin, 2023-08-03:
@abonislawski @yongzhi1 Do you know if coreboot also applied the BIOS fix to set L1 exit threshold to be 1/4 FIFO size?
If coreboot hasn't applied the fix, SOF FW fix https://github.com/thesofproject/zephyr/commit/c31e683295ebd77789f6b4e95c68b073b53b0112 will be needed on main branch for next MTL-006 release.

mengdonglin, 2023-09-04:
Zephyr fixes got merged https://github.com/zephyrproject-rtos/zephyr/pull/62101/ and SOF updated for v2.7