thesofproject/sof #7649: [BUG][Multi-core]  ipc timed out for 0x13000004|0x1GLB_SET_PIPELINE_STATE when testing non-core 0 pipeline on MTL-NOCODEC

keqiaozhang, 2023-05-19:
**Describe the bug**
Testing mtl-005-stable-drop branch on MTL-NOCODEC platform with multi-core tplg, but all non-core0 pipelines failed the test with IPC times out. The reproduce rate is 100%.

**DMESG**
```
[  898.191889] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x13000003|0x1: GLB_SET_PIPELINE_STATE [data size: 12]
[  898.192296] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x33000000|0x1: GLB_SET_PIPELINE_STATE
[  898.192326] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx done : 0x13000003|0x1: GLB_SET_PIPELINE_STATE [data size: 12]
[  898.192333] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x13000004|0x1: GLB_SET_PIPELINE_STATE [data size: 12]
[  898.700002] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x13000004|0x1
[  898.700032] sof-audio-pci-intel-mtl 0000:00:1f.3: Attempting to prevent DSP from entering D3 state to preserve context
[  898.700041] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[  898.700059] sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0x93000004|0x1|0x0, target: 0x13000004|0x1|0x0, ctl: 0x3
[  898.700070] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[  898.700078] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  898.700084] sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[  898.700091] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[  898.700107] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0x5, ROM error: 0x0
[  898.700113] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
[  898.700123] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit enabled
[  898.700128] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[  898.700158] sof-audio-pci-intel-mtl 0000:00:1f.3: failed to set final state 4 for all pipelines
[  898.700168] sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[  898.700181]  Port2: ASoC: trigger FE cmd: 1 failed: -110

```
**TPLG**
[image]

**To Reproduce**
$ aplay   -Dhw:0,1 -r 48000 -c 2 -f S16_LE -d 3 /dev/zero -v -q
$ aplay   -Dhw:0,2 -r 48000 -c 2 -f S16_LE -d 3 /dev/zero -v -q

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Linux Branch: [topic/sof-dev](https://github.com/thesofproject/linux/tree/topic/sof-dev)
    * Linux Commit: [0035b02b10f1](https://github.com/thesofproject/linux/commit/0035b02b10f1ef6417d76f469c04572a3af5b9a3)
    * SOF Branch: [mtl-005-drop-stable](https://github.com/thesofproject/sof/tree/mtl-005-drop-stable)
    * SOF Commit: [7c46d9c6fea6](https://github.com/thesofproject/sof/commit/7c46d9c6fea68be3c0ee1ff426d24e12c8b9be15)
2) Name of the topology file
    * Topology: sof-mtl-nocodec-multicore.tplg (Built with PR https://github.com/thesofproject/sof/pull/7582)
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTL-RVP-NOCODEC

[dmesg.txt](https://github.com/thesofproject/sof/files/11514420/dmesg.txt)
[mtrace.txt](https://github.com/thesofproject/sof/files/11514422/mtrace.txt)

abonislawski, 2023-05-19:
Multi-core fix added to 005, it should cover case when several pipelines used on a single secondary core.
Please verify if it helps here or in #7650

keqiaozhang, 2023-05-19:
This issue is still reproducible on latest mtl-005 branch. But it won't fail immediately and the errors are also changed.

```
[  228.014879] kernel: snd_sof:sof_ipc4_prepare_copier_module: sof-audio-pci-intel-mtl 0000:00:1f.3: copier copier.SSP.4.1, IPC size is 188
[  228.014883] kernel: snd_sof:sof_ipc4_update_resource_usage: sof-audio-pci-intel-mtl 0000:00:1f.3: copier.SSP.4.1: ibs / obs / cpc: 384 / 384 / 6388
[  228.014891] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x47000000|0x0: MOD_SET_DX [data size: 8]
[  228.517095] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x47000000|0x0
[  228.517140] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Attempting to prevent DSP from entering D3 state to preserve context
[  228.517155] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[  228.517218] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0x47000000|0x0|0x0, target: 0x67000000|0x0|0x0, ctl: 0x3
[  228.517238] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[  228.517250] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  228.517261] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[  228.517269] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[  228.517288] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0x0, ROM error: 0x0
[  228.517294] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
[  228.517304] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit enabled
[  228.517309] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[  228.517369] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed to enable target core for widget pipeline.3
[  228.517381] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Failed to set up connected widgets
[  228.517402] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed widget list set up for pcm 1 dir 0
[  228.517410] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -110
[  228.517446] kernel:  Port1: ASoC: error at __soc_pcm_hw_params on Port1: -110
[  228.517462] kernel:  Port1: ASoC: error at dpcm_fe_dai_hw_params on Port1: -110
[  228.517478] kernel: snd_sof:sof_pcm_hw_free: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: free stream 1 dir 0
[  228.518201] kernel: snd_sof:sof_pcm_close: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: close stream 1 dir 0
```

abonislawski, 2023-05-22:
Probably now we hit this one
https://github.com/thesofproject/sof/issues/7108#issuecomment-1549019959

keqiaozhang, 2023-05-22:
I checked this issue again today and found that this issue only happened at the first/second round of test after system reboot.

But this issue cannot be reproduced after reloading kmod, 
Test steps:
```
1. rebooting the device
2. check aplay on non-core 0 pipelines. it fails at first or second round of test.
3. then run kmod reload
    $ sof-test/tools/kmod/sof_remove.sh
    $ sof-test/tools/kmod/sof_insert.sh
4. run test again on non-core0 pipelines, it will always pass the test. 
```

abonislawski, 2023-05-30:
Verified today with latest 005 version.
Original problem is fixed along with another one in 7108.

Right now it can hit another issue, aplay -Dhw:0,1 is working fine one by one but (as @keqiaozhang advised) we can add delay and break it.
That's because driver will put dsp to D3 state in such case:

> [  +0.001031] snd_sof_intel_hda_common:hda_dsp_state_log: sof-audio-pci-intel-mtl 0000:00:1f.3: Current DSP power state: D3
> [  +0.000004] snd_sof:sof_set_fw_state: sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state change: 7 -> 0

and restore when we start another aplay, core 1 will be power up successfully but hit timeout on the first IPC (create pipeline).

I think this is already third issue in this thread so it would be great close this and open another one for D3 restore scenario.

@mengdonglin FYI: IMR context save is enabled only on 005, thats why it can behave differently on main branch.
https://github.com/thesofproject/sof/commit/6c57b64b3494b371a60fa974ffebc03d7977decc

aborisovich, 2023-06-15:
Failed to reproduce this issue with:
https://github.com/thesofproject/sof/pull/7325 + https://github.com/zephyrproject-rtos/zephyr/pull/56168

aborisovich, 2023-06-18:
Reproduced the issue, checked sof main branch with my 2 modified PRs:
https://github.com/thesofproject/sof/pull/7325
https://github.com/zephyrproject-rtos/zephyr/pull/56168
Unfortunately this does not work and fails very differently (on system boot multiple IPC timeout error on pipeline creation).
[dmesg (2).txt](https://github.com/thesofproject/sof/files/11782875/dmesg.2.txt)
Attempting to cherry-pick those two PRs onto existing MTL-005 as next step.

serhiy-katsyuba-intel, 2023-06-27:
The root cause of the issue is that IDC interrupt was not re-enabled after exiting D3.
Fix PR for mtl-005-drop-stable branch:
https://github.com/thesofproject/zephyr/pull/68

keqiaozhang, 2023-06-30:
This issue has been fixed on latest mtl-005 branch, closing it.

closed 2023-06-30 by keqiaozhang