thesofproject/sof #7650: [BUG] ipc timed out for 0x13000003 GLB_SET_PIPELINE_STATE on MTL-HDA when testing multiple-pipeline

keqiaozhang, 2023-05-19:
**Describe the bug**
This issue only happens on mtl-005-stable-drop branch, no reproductions on sof-main branch. It's more like a regression and the reproduce rate is almost 100%.

**dmesg**
```
[ 1982.116838] kernel: snd_sof:sof_ipc4_route_free: sof-audio-pci-intel-mtl 0000:00:1f.3: unbind modules copier.host.15.1:0 -> gain.15.1:0
[ 1982.117021] kernel: snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc4 set pipeline 0 state 3
[ 1982.117039] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x13000003|0x0: GLB_SET_PIPELINE_STATE
[ 1982.622809] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x13000003|0x0
[ 1982.622912] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Attempting to prevent DSP from entering D3 state to preserve context
[ 1982.622924] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 1982.623028] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0x93000003|0x0|0x0, target: 0x33000000|0x0|0x0, ctl: 0x3
[ 1982.623159] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 1982.623205] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 1982.623251] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[ 1982.623283] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[ 1982.623334] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0x0, ROM error: 0x0
[ 1982.623376] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
[ 1982.623456] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit enabled
[ 1982.623494] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 1982.623562] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to pause all pipelines
[ 1982.623612] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[ 1982.623675] kernel:  HDA Analog: ASoC: trigger FE cmd: 0 failed: -110
```

**To Reproduce**
$ ~/sof-test/test-case/multiple-pipeline.sh -f p -l 50
or 
$ ~/sof-test/test-case/multiple-pipeline.sh -f c -l 50
or
$ ~/sof-test/test-case/multiple-pause-resume.sh -r 50

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Linux Branch: [topic/sof-dev](https://github.com/thesofproject/linux/tree/topic/sof-dev)
    * Linux Commit: [0035b02b10f1](https://github.com/thesofproject/linux/commit/0035b02b10f1ef6417d76f469c04572a3af5b9a3)
    * SOF Branch: [mtl-005-drop-stable](https://github.com/thesofproject/sof/tree/mtl-005-drop-stable)
    * SOF Commit: [7c46d9c6fea6](https://github.com/thesofproject/sof/commit/7c46d9c6fea68be3c0ee1ff426d24e12c8b9be15)
2) Name of the topology file
    * Topology: sof-mtl-nocodec-multicore.tplg (Built with PR https://github.com/thesofproject/sof/pull/7582)
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTL-RVP-NOCODEC

[mtrace.txt](https://github.com/thesofproject/sof/files/11514837/mtrace.txt)

keqiaozhang, 2023-05-19:
Checked with latest mtl-005/https://github.com/thesofproject/sof/commit/ddc2ef666872f07be639625652b20c877fee9f71, the results are the same. This issue can still be reproduced.

keqiaozhang, 2023-05-22:
This issue also can be reproduced after reverting https://github.com/thesofproject/sof/commit/f2fec7476d1f625c11c9383f921d5a1e6bf06c75 and https://github.com/thesofproject/sof/commit/ddc2ef666872f07be639625652b20c877fee9f71.

So this is not a regression after mtl-005-rc1.

serhiy-katsyuba-intel, 2023-05-23:
I'm trying to reproduce the issue on our Linux machine. On capture test ("-f c" option) I get "aplay: pcm_write:2127: write error: Input/output error" and I do not observe ipc timeouts. So, I'm not sure if this is same problem as reported here or something else.

Reverting this commit https://github.com/thesofproject/sof/commit/7c46d9c6fea68be3c0ee1ff426d24e12c8b9be15 solves input/output error problem and test successfully passes 50 iteration. That commit is only present on mtl-005-drop-stable branch.

@keqiaozhang, if its possible, could you try to revert the above commit and run the test to check if it also solves the problem you observe?
Just in case, tagging @abonislawski , @RanderWang .

keqiaozhang, 2023-05-24:
I tried https://github.com/abonislawski/sof/commits/mtl-005-drop-stable(commit ID:https://github.com/abonislawski/sof/commit/edc117c553f27611a2e508c7e464a555880d3f95) and issue cannot be reproduced anymore.

keqiaozhang, 2023-05-25:
Closing this bug, the fix has been landed on https://github.com/thesofproject/sof/commits/mtl-005-drop-stable.

closed 2023-05-25 by keqiaozhang