thesofproject/sof #7652: [BUG] toplogy1: FW pcm_params with large period size

kv2019i, 2023-05-19:
**Describe the bug**
Continuation of 
https://github.com/thesofproject/sof/issues/7476

See log in https://github.com/thesofproject/sof/issues/7476#issuecomment-1550896125

Vamshigopal, 2023-05-19:
@kv2019i  i took the conf file and dmesg at different instances, so im not sure, let me share the logs and conf file with 50ms period size.
i do agree with your assumption, as i see same error for 50ms, 80ms.. i.e SOF_TKN_SCHED_PERIOD > 44739

kv2019i, 2023-05-19:
@Vamshigopal Bugfix candidate at https://github.com/thesofproject/sof/pull/7655
I didn't realize we had similar signed 32bit arithmetic in FW to calculate the frame size. :( So this obviously fails in the same conditions as the m4 calculations failed.

kv2019i, 2023-08-08:
Fix merged, closing.

closed 2023-08-08 by kv2019i