thesofproject/sof #7658: [BUG] s24 playback on speaker and headset is silent with mtl-005 fw and tplg

udaymb, 2023-05-19:
**Describe the bug**
s24 playback is silent

**To Reproduce**
aplay with speaker or headset with form s24_le using s24 playback file

**Reproduction Rate**
100

**Expected behavior**
s24 playback should work

**Impact**
critical

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA} - https://chromium.googlesource.com/chromiumos/third_party/kernel/+/refs/heads/chromeos-6.1
    * SOF: {SHA} https://github.com/thesofproject/sof/tree/mtl-005-drop-stable
2) Name of the topology file
    * Topology: https://github.com/thesofproject/sof/tree/mtl-005-drop-stable
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTL

serhiy-katsyuba-intel, 2023-05-23:
@RanderWang , @ranj063 , @udaymb , @abonislawski ,  could someone tell how does 24 bit audio sent by Linux/ChromeOS driver to FW copier.host look like? Is it 24 bit in upper 3 bytes of 4 bytes container (bits 8 to 31) or is it 24 bits in lower 3 bytes of 4 bytes container (bits 0 to 23).

Windows driver sends S24_4LE audio as 24 bits in **UPPER** 3 bytes of 4 bytes container, i.e. in bits 8 to 31. Maybe Linux/ChromOS driver bahaves differently?

There was a fix not long time ago for 24_4LE format: https://github.com/thesofproject/sof/commit/d2b0b4716be7f5371d48993833e198c27d8c5d7e after which 24_4LE started to work properly on Windows and maybe stopped working properly on Linux/ChromeOS.

singalsu, 2023-05-23:
Confirming too, that revert fixes the s24 playback issue in git main with also TGL platform.

ranj063, 2023-05-23:
> I've tried to run FW with revert of [d2b0b47](https://github.com/thesofproject/sof/commit/d2b0b4716be7f5371d48993833e198c27d8c5d7e) and that solves the problem.
> 
> So now we should decide what is the correct representation of 24 bit audio send by driver to FW copier.host.

@serhiy-katsyuba-intel Can ew introduce a kconfig that we can set/unset for Linux and Windows that does things differently?

kv2019i, 2023-05-23:
Kconfig doesn't really seem very tempting, we are talking about distinct data formats, so I'd prefer if we could expand the FW interface (the IPC4 audio formats to be able to describe both variants) and be able to build a firmware that can support both (or at least say at runtime if the other variant is not supported).

@serhiy-katsyuba-intel wrote:
> ""Windows driver sends S24_4LE audio as 24 bits in UPPER 3 bytes of 4 bytes container, i.e. in bits 8 to 31. Maybe 
> Linux/ChromOS driver bahaves differently?"

This "Windows" format is SNDRV_PCM_FORMAT_S32_LE in Linux with msbits=24. Many apps ignore the msbits as for most purposes, it doesn't matter if the LSB byte is zeroed out (or ignored) -- i.e. just produce/consume 32bit signed samples. I'd argue this format is by far more common in Linux apps as well (as the other two are not very convenient if you need to do any audio processing on the host side).

The other formats in Linux:
- S24 in four byte containers, samples in lower three bytes SNDRV_PCM_FORMAT_S24 (and friends)
- S24 in three byte containers SNDRV_PCM_FORMAT_S24_3LE (and friends)

But Linux can describe all three variants, so we just need to decide what is correct interpretation of sof_ipc4_audio_format, and then only expose the matching formats in SOF ALSA driver. The ALSA plugin layer can convert between the formats, so we just need to advertize the correct formats.

abonislawski, 2023-05-23:
> * S24 in three byte containers SNDRV_PCM_FORMAT_S24_3LE (and friends)

This make sense and this is what I see in aplay when trying to play random 24b sample from windows

> Warning: format is changed to S24_3LE
> Playing WAVE '/home/24_bit_fixed.WAV' : Signed 24 bit Little Endian in 3bytes

Can we simply support both S24_LE and S24_3LE?

jairaj-arava, 2023-05-24:
@keqiaozhang 
Checked at our end: Without the fix,  the 24 bit playback was silent on both speakers and headset and 24 bit Headset capturing doesn't have any issue.

But with the FW that has the revert of [d2b0b47],  playback on both speaker and headset works.

RanderWang, 2023-05-24:
> @RanderWang , @ranj063 , @udaymb , @abonislawski , could someone tell how does 24 bit audio sent by Linux/ChromeOS driver to FW copier.host look like? Is it 24 bit in upper 3 bytes of 4 bytes container (bits 8 to 31) or is it 24 bits in lower 3 bytes of 4 bytes container (bits 0 to 23).
> 
> Windows driver sends S24_4LE audio as 24 bits in **UPPER** 3 bytes of 4 bytes container, i.e. in bits 8 to 31. Maybe Linux/ChromOS driver bahaves differently?
> 
> There was a fix not long time ago for 24_4LE format: [d2b0b47](https://github.com/thesofproject/sof/commit/d2b0b4716be7f5371d48993833e198c27d8c5d7e) after which 24_4LE started to work properly on Windows and maybe stopped working properly on Linux/ChromeOS.

Windows supports two types of 24bit format:   S24_4LE (0 ~ 23 bits) or S24_3LE( each sample uses 24bit container size). I validated on Win10 with TGL HDA codec

mengdonglin, 2023-05-25:
> @keqiaozhang Checked at our end: Without the fix, the 24 bit playback was silent on both speakers and headset and 24 bit Headset capturing doesn't have any issue.
> 
> But with the FW that has the revert of [[d2b0b47](https://github.com/thesofproject/sof/commit/d2b0b4716be7f5371d48993833e198c27d8c5d7e)], 24bit playback on both speaker and headset works.

@udaymb @jairaj-arava @serhiy-katsyuba-intel I closed this issue as the fix been merged by https://github.com/thesofproject/sof/commits/mtl-005-drop-stable commit https://github.com/thesofproject/sof/commit/24eca03573122e9018917a2d2fd2c0c7b8761461

closed 2023-05-25 by mengdonglin