thesofproject/sof #7663: [BUG] [MTL] Deepbuf playback is mixed with random artifacts/glitch

yongzhi1, 2023-05-21:
**Describe the bug**
Deep buffer playback has random artifacts. 

**To Reproduce**
```
aplay -Dhw:0,31 -c2 -r48000 -fS16_LE /tmp/sample.wav
```
**Reproduction Rate**
How often does the issue happen ? i.e. 1/10 (once in ten attempts), 1/1000 or all the time.
80%

**Expected behavior**
The sound from deep buffer device should be noise/glitch free.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
   * Kernel: {SHA} - https://chromium.googlesource.com/chromiumos/third_party/kernel/+/refs/heads/chromeos-6.1
   * SOF: {SHA} https://github.com/thesofproject/sof/tree/mtl-005-drop-stable 
2) Name of the topology file
    * Topology: sof-mtl-max98357a-rt5682-ssp2-ssp0
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTL

**Screenshots or console output**

[mtrace.log](https://github.com/thesofproject/sof/files/11523648/mtrace.log)
[dmesg-log.txt](https://github.com/thesofproject/sof/files/11523649/dmesg-log.txt)
[dp-spk-sine.zip](https://github.com/thesofproject/sof/files/11523653/dp-spk-sine.zip)

mengdonglin, 2023-05-22:
@yongzhi1 Can you reproduce this issue with headset playback?  We cannot reproduce this issue with SSP loopback on RVP.

yongzhi1, 2023-05-22:
Hi, @mengdonglin , yes, we can also reproduce the issue on "Deepbuffer Jack Out", @udaymb also observed drag in the sound. 
It seems the issue has more to do with workload or timing/scheduling, I am not able to detect irregularity with the loopback test either, but when running the deepbuff alone, it's easy to hear.

fredoh9, 2023-05-25:
I'm interested in this issue.

I have a setup and build my own kernel with chromeos-6.1 and loaded fw/tplg with [mtl-005-drop-stable](https://github.com/thesofproject/sof/tree/mtl-005-drop-stable)

But hard to tell whether I'm hearing any noise/glitch. I can play with pure sine wave and check again. Or using alsabat I can offload my detection part also.

fredoh9, 2023-05-26:
Interestingly I can't reproduce the problem, not sure what's wrong or good about my environment.

1. I flashed R115 of 2023/5/19
2. I was able to test playback on headset and spekaer and Deepbuffer on headphone, sound was good but I didn't try much
3. I buid [chromeos-6.1](https://chromium.googlesource.com/chromiumos/third_party/kernel/+/refs/heads/chromeos-6.1)
I had to pick up a patch otherwise FW loading failed. This way I was able to make sure I'm using my own build

```
$ git log --oneline -2
be3488fafdc5 (HEAD) ASoC: SOF: topology: Fix logic for copying tuples
e7f4d1476d2c (m/main, cros/chromeos-6.1) FROMGIT: x86/sched: Add the SD_ASYM_PACKING flag to the die domain of hybrid processors
```
4. overwrite fw/tplg from mtl-005-stable-drop
5. I did same thing, playback on headset and speaker and Deepbuffer. Play song and pure sine wave many time but sound okay.
6. decided to prove with alsabat. ChromeOS has old alsa-utils, copied latest alsa-lib/alsa-utils locally and test alsabat test. this return always pass.

```
./alsabat -Phw:sofrt5682,0 --standalone -n 240000 -r 48000 -c 2 -f S16_LE -F 599 -k 2.1
./alsabat -Chw:CODEC,0 -c 2 -r 48000 -f S16_LE -F 599 -k 2.1
```

yongzhi1, 2023-05-26:
Hi, thank you @fredoh9 , even our validation team confirmed that the deepbuf on headset is clean and glitch-free, however with the same FW(md5sum 16251efe48af), and with Deepbuf configured to SPK instead of HS for test purpose, it's very easy to tell the differences. and with same kernel, to repeat the test with MTL-03 release, the sound coming from Deepbuffer Amps is on a par with regular Spk. Attached clips captured by webcam fyi [comparision.zip](https://github.com/thesofproject/sof/files/11570658/comparision.zip)

plbossart, 2023-05-26:
@yongzhi1 when you use the speaker path, don't you have the additional load of speaker protection? What you see with glitches could be xruns or problems coming for the smartamp module, no?

yongzhi1, 2023-05-26:
Hi, @plbossart , the deepbuf & regular playback are mirrored, smart-amp module was not integrated yet:
[image]
To play same music from same speakers, under the same condition:
1. Play from hw:0,31 - captured by webcam on Host machine as deepbuf-amps.wav
2. Play from hw:0,1  - captured by webcam on Host machine as regular-spks.wav

Sound quality from 2 is consistently better than 1.

plbossart, 2023-05-26:
@yongzhi1 is the only difference in topology the SSP which is used? I don't really see how the firmware could behave differently between headphone and speaker outputs, it doesn't even know what the output is...

I also don't get what the 'captured by webcam on Host machine' means.

yongzhi1, 2023-05-26:
Hi, @plbossart , instead of depending on on-board DMIC, webcam on external Desktop is used to record the sound for review.
Actually with "Deepbuf Jack out" => 3.5 mm line-out => Logitech S120, one can also hear the differences between Headset and Deepbuff Jackout.

plbossart, 2023-05-26:
Right, so this has nothing to do with the selected endpoint and it's a clear issue with the firmware. Thanks @yongzhi1 for the clarification.

mwasko, 2023-06-05:
> Right, so this has nothing to do with the selected endpoint and it's a clear issue with the firmware. Thanks @yongzhi1 for the clarification.

@plbossart can you explain why you believe that it is clear that an issue is in the FW?

@softwarecki has investigated this issue and the problem has been root caused as a SW Driver issue not being able to keep up with data transmission in deep buffer scenario. As a deep buffer optimization the FW waits with DMA update till there is less then 4 periods of data. However, when this optimization is enabled the Driver has a problem with catching up over these remaining 4 periods. On MTL-005 this optimization has been turned off so you can check it, but I would say this is something we want to have for power improvements.

The question is if it is acceptable for the SW driver to not be able to produce data for 4 ms (resume time from low power state?)? If yes then indeed it is FW issue that wrongly assumed that it is a valid flow.

keqiaozhang, 2023-06-06:
I checked this issue with mtl-005 on mtl-soundwire ChromeBook, deepbuffer playback is clean, without any glitch. 

@yongzhi1 , have you seen this issue on soumdware devices? Or it only occurs on I2S devices?

RanderWang, 2023-06-06:
@yongzhi1  I check the dmesg log and find that 10 ms buffer size is used ( 0000780), do you try other number ? 

[  726.511567] sof-audio-pci-intel-mtl 0000:00:1f.3: Create widget copier.host.15.1 instance 0 - pipe 15 - core 0
[  726.511572] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x40000004|0x15 [data size: 84]
[  726.512033] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x60000000|0x15
[  726.512208] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx done : 0x40000004|0x15 [data size: 84]
[  726.512223] Message payload: 00000000: 00000000 000000c0 00000180 00000001
[  726.512227] Message payload: 00000010: 0000bb80 00000010 ffffff10 00000001
[  726.512229] Message payload: 00000020: 00000000 00001002 0000bb80 00000020
[  726.512232] Message payload: 00000030: ffffff10 00000001 00000000 00002002
[  726.512234] Message payload: 00000040: 00000000 00000000 **00000780** 00000001
[  726.512235] Message payload: 00000050: 00000001

RanderWang, 2023-06-06:
@yongzhi1  can you share your topology file ?  In my opinion, we tested different topology so got different result.  I discussed with Keqiao and found he thought he used the correct topology file but I found it was for headphone deep buffer.

mwasko, 2023-06-06:
> Where is the root cause analysis and where do these 4ms come from?

@softwarecki  was trying to explain this in [#4408](https://github.com/thesofproject/linux/issues/4408). The 4ms is coming from the FW optimization, while the buffer is 100+ms the FW to reduce Driver resume frequency is waiting with update of the DMA read/write registers till there is < 4ms of data left.

> The deep buffer scenario assumes a 100+ms buffer, I really don't get why we expect near real-time behavior when there is plenty of time to wake-up earlier. In a 4-period setup, the host driver should have 25+ ms to provide the data. If we play with timers on the host side, the timer should not go below the 80% empty mark in the ring buffer... Even if you make that 90% empty, that's still nowhere near the 4ms.

@plbossart FW is not expecting SW to wake up every 4ms, rather trigger him every (100+ms - 4ms). However when FW update DMA there is only 4ms of data left in the buffer and SW is expected to copy more within next 4ms (this optimization was reverted here: [8ddd262](https://github.com/thesofproject/sof/commit/8ddd262530042b10a5e566062433b5f2c294dd98)). I suspect that we might have two optimizations that conflict with each other. The SW Driver will not be able to fill more data till there is < 4ms because FW explicitly does not update read pointers to avoid unnecessary wake of the host. This is something worth investigating.

plbossart, 2023-06-06:
I suspect it's one optimization too far. It's natural that when the host needs to provide more 10x data it needs more time, especially if this involves fetching data from a file or something slower.

A good rule of thumb is not to expect the host stack to react in less than 10ms. That's what we do for regular playback, and with deep-buffer+S0i1 the expectation should be that the wake-up and reaction times are even longer.

mwasko, 2023-06-07:
> A good rule of thumb is not to expect the host stack to react in less than 10ms. That's what we do for regular playback, and with deep-buffer+S0i1 the expectation should be that the wake-up and reaction times are even longer.

That seems reasonable, but while the 4ms might be too aggressive I still believe we need this mechanism to improve power savings - @mmaka1 fyi. @softwarecki can you share details why exactly 4ms was chosen? @plbossart do you have any recommendations from the SW, maybe we should use % of the buffer left in conjunction with check that it will be not less then e.g. 10ms?

ujfalusi, 2023-06-07:
@mwasko, 10% of the in firmware 'host buffer' is a reasonable watermark, I would start from 30% then go lower to see where the issue surfaces, it might give us a hint also?

softwarecki, 2023-06-07:
> can you share details why exactly 4ms was chosen?

The delayed dma reload mechanism was introduced in this commit https://github.com/thesofproject/sof/commit/e1831cb0. This commit https://github.com/thesofproject/sof/commit/e95723ac added a default configuration where dma is reloaded 4ms before the end of data in the buffer. I think @lyakh would be able to tell us why this value was chosen.

plbossart, 2023-06-07:
@softwarecki we are confusing DMA requests and host wakeups. the host does NOT wake-up after every DMA request, it only wakes-up when the host buffer is becoming empty. The expectation is that the host buffer is bigger than the firmware buffer and has enough slack to wake-up.

lgirdwood, 2023-06-09:
Thinking aloud - if FW is doing host FW reload less frequently then FW pointer will be more out of sync with host LPIB. To commit/reload host we eventually do this
```c
/**
 * @brief Commit a number of bytes that have been transferred to/from host
 *
 * Writes the length to BFPI. For host transfers LLPI and LPIB are
 * also written to with the given length.
 *
 * This then updates the read or write position depending on the direction.
 *
 * LPIBI writes here can be seen on the host side of the transfer in the
 * matching LPIB register.
 *
 * LLPI seems to, from behavior, inform the hardware to actually read/write
 * from the buffer. Without updating BFPI AND LLPI, the transfer doesn't
 * happen in testing for host transfers.
 *
 * @param base Base address of the IP register block
 * @param regblock_size Register block size
 * @param sid Stream ID within the register block
 * @param len Len to increment postion by
 */
static inline void intel_adsp_hda_host_commit(uint32_t base,
					      uint32_t regblock_size,
					      uint32_t sid,
					      uint32_t len)
{
	*DGBFPI(base, regblock_size, sid) = len;
	*DGLLPI(base, regblock_size, sid) = len;
	*DGLPIBI(base, regblock_size, sid) = len;
}
```
It seems like we need a variation of this function so that we can
1) Independently update host LPIB without copying data
2) Independently update DSP position which may force data copy.

wszypelt, 2023-06-16:
@yongzhi1 does the issue still reproduce?

closed 2023-06-21 by yongzhi1