thesofproject/sof #7667: [BUG] The waveform deformed at the beginning of the recording onMTL-SDW

keqiaozhang, 2023-05-22:
**Describe the bug**
Observed this issue in daily test, alsabat test failed on MTL-SDW, the waveform deformed at the beginning of the recording:

[image]

**To Reproduce**
~/sof-test/test-case/check-alsabat.sh -c hw:sofsoundwire,1 -p hw:Device,0 -C 2 -F 599

**Reproduction Rate**
>50%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel Commit: [67e9dd79ff18](https://github.com/thesofproject/linux/commit/67e9dd79ff18)
    * SOF Commit: [38f3f5d4119e](https://github.com/thesofproject/sof/commit/38f3f5d4119e)
2) Name of the topology file
    * Topology: sof-mtl-rt711-4ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTL-RVP_SDW

keqiaozhang, 2023-05-23:
Confirmed that this is a hardware related issue, not all MTL-SDW RVPs have this issue. Let's track this issue in https://github.com/intel-innersource/drivers.audio.ci.sof-framework/issues/425.

keqiaozhang, 2023-05-25:
Reopening this bug,  since we have observed this issue on all MTL-SDW devices. Even some devices can pass the alsabat test, but the initial waveform still has varying degrees of deformation.

mengdonglin, 2023-05-25:
@bardliao @plbossart @aiChaoSONG  Can we have loopback of the onboard ALC711 to check the data?

bardliao, 2023-05-25:
> @bardliao @plbossart @aiChaoSONG Can we have loopback of the onboard ALC711 to check the data?

Yes, we can do loopback by the following commands.
sudo su
cd /sys/kernel/debug/soundwire/master-0-0/intel-sdw/
echo 3 > cdns-pdi-loopback-target
echo 2 > cdns-pdi-loopback-source
run aplay
run arecord

master-0-0 depends on which sdw link is rt711 on.

aiChaoSONG, 2023-05-30:
With Bard's instruction on soundwire loopback, I am not able to reproduce the issue with iteration 200. And after replace the usb sound card with a stereo one, I am not able to reproduce the issue, either. I can confirm that this issue is related to USB sound card.

fredoh9, 2023-05-30:
I didn't know there is a codec loopback mode for SDW soundcard. We could add loopback test also for eligible codecs.
Sure, I will work with @keqiaozhang for upgrading USB sound card

keqiaozhang, 2023-06-12:
Closing this bug, as we have replaced the USB sound card to a new model on all MTL-SDW devices. This issue is not reproducible in CI.

closed 2023-06-12 by keqiaozhang