thesofproject/sof #7683: [BUG] Speaker playback loses sound upon feedback capture stops

yongzhi1, 2023-05-24:
**Describe the bug**
When we abort echo ref capture in the middle of speaker playback, the playback continues but with no audible sound.

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug)

Start speaker playback
```
aplay -D hw:0,1 -c 2 -r 48000 -f s16_le /home/sample.wav -vv
```
Start echo ref capture
```
arecord -Dhw:0,27 /tmp/feedback.wav -c2 -r48000 -fs32_le -vv
```
Abort echo ref capture in the middle of playback with CTL+C and observe playback is running with sound muted immediately. 

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {https://chromium.googlesource.com/chromiumos/third_party/kernel/+/refs/heads/chromeos-6.1}
    * SOF: {https://github.com/thesofproject/sof/tree/mtl-005-drop-stable}
2) Name of the topology file
    * Topology: {sof-mtl-max98357a-rt5682-ssp2-ssp0.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {MTL}

**Screenshots or logs**
Please also include the relevant sections from the firmware log and kernel log in the report:
[dmesg-log.txt](https://github.com/thesofproject/sof/files/11559150/dmesg-log.txt)
[mtrace.txt](https://github.com/thesofproject/sof/files/11559151/mtrace.txt)

fredoh9, 2023-06-02:
This is 100% reproducible with my MTL REX setup too
@aiChaoSONG do you have any suggestion?

serhiy-katsyuba-intel, 2023-06-02:
I suspect this commit https://github.com/zephyrproject-rtos/zephyr/pull/54505/commits/d97abc931df78c644ce0e63bda8638ef6604425b merged into main at April 25 may be responsible to the problem.
However, I do not have now an access to platform to verify the guess. Do you maybe recall the problem is a regression and on older firmware (before April 25) there was no such problem?

fredoh9, 2023-06-02:
@serhiy-katsyuba-intel I can revert it and test again. I need prioritize other tasks first, will update here if I have some chance to try.

serhiy-katsyuba-intel, 2023-06-05:
Verified suspected commit https://github.com/zephyrproject-rtos/zephyr/commit/d97abc931df78c644ce0e63bda8638ef6604425b is NOT responsible for the problem: both with and without that commit the problem is reproducible.

mengdonglin, 2023-06-07:
@yongzhi1  Can you still reproduce this issue with MTL-005 Hot Fix2 (https://github.com/thesofproject/sof/commit/c08898c8ef2859a50d20af6758e408cb242e0373)?

lgirdwood, 2023-06-09:
@serhiy-katsyuba-intel it could be pipeline state machine, i.e. the STOP from echo ref could be impacting some parts the playback pipeline logic.  It seems the host DMA is still consuming (otherwise aplay would fail) but downstream from host (probably after echo ref) there could be a condition that is stopping new samples after STOP from the echo ref..

yongzhi1, 2023-06-09:
Hi, @mengdonglin , yes the issue is still reproducible with latest MTL-005 Hot Fix2.

mengdonglin, 2023-06-12:
> Hi, @mengdonglin , yes the issue is still reproducible with latest MTL-005 Hot Fix2.

@yongzhi1 Thank you for check! Can you please confirm sof-mtl-max98357a-rt5682-ssp2-ssp0.tplg and the highlighted pipelines in the diagram are being used in your test?

[image]

I'll check if we can simulate the topology on MTL RVP.

bardliao, 2023-06-13:
JFYI, The issue is still reproducible if loopback mode is disabled. I.e. remove "quirks          "lbm_mode"" from the topology.

yongzhi1, 2023-06-14:
Hi, @mengdonglin , yes, we are using the above highlighted blocks for testing.

serhiy-katsyuba-intel, 2023-06-15:
> @serhiy-katsyuba-intel it could be pipeline state machine, i.e. the STOP from echo ref could be impacting some parts the playback pipeline logic. It seems the host DMA is still consuming (otherwise [...]

Did some verifications. After stopping capture on SSP0, playback on SSP0 continues without any problem visible on SOF FW side: valid audio data is copied to dai-zephyr DMA buffer as before (but this time no audio is heard on speakers).

Tried to comment out SSP stop (so stopping SSP capture does not modify any SSP related registers) -- does not help. Tried to also comment out DMA stop (so basically capture stop does not modify any SSP and DMA related registers) -- does not help either.

There is a chance the issue might be related to some clock or power gating in SoC. However, maybe this is a driver issue, maybe driver does some power/clock off for codec after capture stop ignoring the playback is still in progress? Can someone from driver side have a look at this?

Tagging: @bardliao , @ranj063, @lgirdwood .

Just in case, if we have a possibility to view I2S lines on board via logic analyzer of JTAG boundary-scan it would be nice to check if audio is actually transmitted on I2S or not when the issue is reproduced. That will tell us if the issue is in FW or in driver.

lyakh, 2023-06-19:
@yongzhi1 has it worked before? If yes - what was the last working version of the firmware and the kernel?

plbossart, 2023-06-19:
@yongzhi1 also when does the audio become silent on the speaker output?
a) immediately when the capture stops?
b) with a small, e.g. 3s delay, after the capture stops?

The latter case would be a clear indicator of a driver/DAPM issue, the former of a hardware/firmware issue.

serhiy-katsyuba-intel, 2023-06-19:
> @yongzhi1 also when does the audio become silent on the speaker output? a) immediately when the capture stops? b) with a small, e.g. 3s delay, after the capture stops?
> 
> The latter case would be a clear indicator of a driver/DAPM issue, the former of a hardware/firmware issue.

Audio becomes silent immediately or about so. If there is any delay, it is not observable with naked eye (ear), less then 0.5 sec., I guess. Definitely much less then 3 sec.

plbossart, 2023-06-19:
Humm, that's interesting @serhiy-katsyuba-intel. We could confirm this by dumping the amplifier registers before and after killing the capture path, to see if there's indeed no difference. That would definitively rule out a DAPM/speaker amplifier configuration issue. @yongzhi1 is this something you can do with a regmap dump?

It wouldn't rule out the driver completely, there is still scope for some driver interference in the pipeline management, but that would be centered on the firmware configuration, not the 3rd party amp.

plbossart, 2023-06-19:
alternatively, hooking-up a LogicPro on the SSP line would tell you right away what happens, but it's more difficult to open-up the device and access SSP pins.

lyakh, 2023-06-19:
@yongzhi1 I think we also need to know where the loopback is applied: at the SSP or at the codec level? If it's an internal SSP loopback mode, then the codec and the amp shouldn't matter. If it's an external patch, then we need to check them all