thesofproject/sof #7696: [BUG] ipc timed out for 0x13000003|0x1GLB_SET_PIPELINE_STATE on MTL-NOCODEC

keqiaozhang, 2023-05-26:
**Describe the bug**
Observed this issue on mtl-005-stable-drop branch, no reproductions on main branch. This issue happens when testing multiple pipelines. The reproduction is about 30%.

**dmesg**
```
[ 7891.007695] snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-mtl 0000:00:1f.3: trigger cmd: 0 state: 3
[ 7891.007706] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x13000003|0x1: GLB_SET_PIPELINE_STATE [data size: 12]
[ 7891.007834] snd_sof:sof_pcm_hw_free: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: free stream 31 dir 0
[ 7891.007852] snd_sof_intel_hda_common:hda_dsp_stream_trigger: sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x200]=0x340000 successful
[ 7891.007862] snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-mtl 0000:00:1f.3: trigger cmd: 0 state: 2
[ 7891.514235] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x13000003|0x1
[ 7891.514274] sof-audio-pci-intel-mtl 0000:00:1f.3: Attempting to prevent DSP from entering D3 state to preserve context
[ 7891.514282] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 7891.514316] sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0x93000003|0x1|0x0, target: 0x32000000|0x0|0x0, ctl: 0x3
[ 7891.514348] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 7891.514371] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 7891.514393] sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[ 7891.514410] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[ 7891.514439] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0x0, ROM error: 0x0
[ 7891.514460] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
[ 7891.514487] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit enabled
[ 7891.514506] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 7891.514548] sof-audio-pci-intel-mtl 0000:00:1f.3: failed to pause all pipelines
[ 7891.514582] sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[ 7891.514613]  Port0: ASoC: trigger FE cmd: 0 failed: -110
```

**To Reproduce**
/sof-test/test-case/multiple-pipeline.sh -f a -c 20 -l 50

**Reproduction Rate**
30%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: sof-dev/[022b4bbfd067](https://github.com/thesofproject/linux/commit/022b4bbfd06731e81d04abb3b75ac17648fdca63)
    * SOF: [mtl-005-drop-stable](https://github.com/thesofproject/sof/tree/mtl-005-drop-stable)/[cb0f0daec6b8](https://github.com/thesofproject/sof/commit/cb0f0daec6b8975c4cb0947d241c6e530b29f3d5)
2) Name of the topology file
    * Topology: sof-ace-tplg/sof-mtl-nocodec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform:MTL-NOCODEC

[dmesg.txt](https://github.com/thesofproject/sof/files/11572248/dmesg.txt)

[mtrace.txt](https://github.com/thesofproject/sof/files/11572249/mtrace.txt)

keqiaozhang, 2023-07-04:
I also run to other IPC errors during the testing:
```
[  552.059650] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x45020002|0x20003: MOD_BIND
[  552.567682] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x45020002|0x20003
[  552.567723] sof-audio-pci-intel-mtl 0000:00:1f.3: Attempting to prevent DSP from entering D3 state to preserve context
[  552.567737] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[  552.567761] sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0xc5020002|0x20003|0x0, target: 0x65000000|0x20002|0x0, ctl: 0x3
[  552.567781] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[  552.567794] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  552.567805] sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[  552.567818] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[  552.567839] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0x0, ROM error: 0x0
[  552.567849] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
[  552.567856] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit enabled
[  552.567860] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[  552.567883] sof-audio-pci-intel-mtl 0000:00:1f.3: failed to bind modules mixin.3.1:0 -> mixout.4.1:0
```

```
[ 5391.113866] snd_sof:sof_ipc4_prepare_copier_module: sof-audio-pci-intel-mtl 0000:00:1f.3: copier copier.SSP.4.1, IPC size is 188
[ 5391.113882] snd_sof:sof_ipc4_update_resource_usage: sof-audio-pci-intel-mtl 0000:00:1f.3: copier.SSP.4.1: ibs / obs / cpc: 384 / 384 / 6388
[ 5391.113903] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x47000000|0x0: MOD_SET_DX [data size: 8]
[ 5391.313459] snd_sof:sof_pcm_trigger: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
[ 5391.313479] snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-mtl 0000:00:1f.3: trigger cmd: 0 state: 3
[ 5391.615988] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x47000000|0x0
[ 5391.616009] sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[ 5391.616039] sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed to enable target core for widget pipeline.3
[ 5391.616046] sof-audio-pci-intel-mtl 0000:00:1f.3: Failed to set up connected widgets
[ 5391.616062] sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed widget list set up for pcm 1 dir 0
[ 5391.616069] sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -110
[ 5391.616102]  Port1: ASoC: error at __soc_pcm_hw_params on Port1: -110
[ 5391.616119]  Port1: ASoC: error at dpcm_fe_dai_hw_params on Port1: -110
``

keqiaozhang, 2023-07-10:
The reproduce rate has become very low on latest 005-stable branch, less than 5%.

mengdonglin, 2023-08-16:
Closing as this issue is for single core only. IPC timeout with multi-core will be tracked by #7774

closed 2023-08-16 by mengdonglin