thesofproject/sof #7701: [BUG] ipc_pipeline_free failed on ADLP-SDW-IPC4 platform

keqiaozhang, 2023-05-29:
**Describe the bug**
Observed this bug on ADLP_SKU0B00_SDCA_IPC4ZPH when testing playback continuously on different pipelines.

Before there's a tplg regression https://github.com/thesofproject/sof/issues/7677, it has been fixed by https://github.com/thesofproject/sof/pull/7686. 
Not sure if this issue is hiding by https://github.com/thesofproject/sof/issues/7677 or a new one, need further investigation, 

**mtrace**
```
[   64.693036] <inf> ipc: rx	: 0x13000003|0x1
[   64.693058] <inf> pipe: pipe:0 0x0 pipe trigger cmd 2
[   64.693081] <inf> pipe: pipe:1 0x0 pipe trigger cmd 2
[   64.693098] <inf> pipe: pipe:0 0x0 pipe trigger cmd 2
[   64.693920] <inf> ll_schedule: task complete 0xbe0b6240 0x20180U
[   64.693930] <inf> ll_schedule: num_tasks 2 total_num_tasks 2
[   64.693966] <inf> ll_schedule: task complete 0xbe0b7780 0x20180U
[   64.693976] <inf> ll_schedule: num_tasks 1 total_num_tasks 1
[   64.693995] <inf> ll_schedule: zephyr_domain_unregister domain->type 1 domain->clk 4
[   64.703473] <err> ipc: failed to wait schedule thread
[   64.703490] <err> ipc: ipc4: failed to send delayed reply
[   64.704491] <inf> ipc: rx	: 0x46000002|0x3
[   64.704830] <inf> ipc: rx	: 0x12000000|0x0
[   64.704848] <err> ipc: ipc_comp_free(): comp id: 262144 state is 4 cannot be freed
[   64.704858] <err> ipc: ipc_pipeline_free(): module free () failed
[   64.704888] <err> ipc: ipc4: FW_GEN_MSG failed with err 12
[   64.705290] <inf> ipc: rx	: 0x12010000|0x0
[   64.705311] <err> ipc: ipc_comp_free(): comp id: 196608 state is 4 cannot be freed
[   64.705321] <err> ipc: ipc_pipeline_free(): module free () failed
[   64.705336] <err> ipc: ipc4: FW_GEN_MSG failed with err 12
Terminated
```
**dmesg**
```
[  544.122564] kernel: snd_sof_intel_hda_common:hda_dsp_stream_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x2014001e successful
[  644.128615] kernel: snd_sof:sof_pcm_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 2 dir 0 cmd 0
[  644.128622] kernel: snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-tgl 0000:00:1f.3: trigger cmd: 0 state: 3
[  644.128628] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13000003|0x1: GLB_SET_PIPELINE_STATE [data size: 16]
[  644.139419] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x33000006|0x1: GLB_SET_PIPELINE_STATE
[  644.139422] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW reported error: 6 - Unknown error while processing the request
[  644.139454] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc error for msg 0x13000003|0x1
[  644.139457] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to pause all pipelines
[  644.139460] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
[  644.139465] kernel:  Speaker: ASoC: trigger FE cmd: 0 failed: -22
```

**To Reproduce**
1. ~/sof-test/test-case/check-playback.sh -d 100 -l 1 -r 1
2. ~/sof-test/test-case/check-playback.sh -d 1 -l 100 -r 1
3. ~/sof-test/test-case/check-playback.sh -d 1 -l 1 -r 50

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {[cbf25c97cdda](https://github.com/thesofproject/linux/commit/cbf25c97cdda)}
    * SOF: {[a6984b85030f](https://github.com/thesofproject/sof/commit/a6984b85030f)}
2) Name of the topology file
    * Topology: {avs-tplg/sof-adl-rt711-l0-rt1316-l12-rt714-l3.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: ADLP_SKU0B00_SDCA_IPC4ZPH

[dmesg.txt](https://github.com/thesofproject/sof/files/11589722/dmesg.txt)

[mtrace.txt](https://github.com/thesofproject/sof/files/11589723/mtrace.txt)

keqiaozhang, 2023-05-29:
This issue cannot be reproduced after reverting https://github.com/thesofproject/sof/commit/601d2bc41ec7316231b8472d40b0e4a4b04436d3 and https://github.com/thesofproject/sof/commit/3835846836d43cf142e9e6420ee864b509540ec3

bardliao, 2023-05-30:
@keqiaozhang a potential fix is https://github.com/thesofproject/linux/pull/4389. I don't see FW report error with the PR. But I saw 
`[  202.641981] kernel: soundwire sdw-master-0: generic_new_peripheral_assigned: invalid dev_num 1`
I have no idea what did I miss, yet.

bardliao, 2023-05-30:
> @keqiaozhang a potential fix is [thesofproject/linux#4389](https://github.com/thesofproject/linux/pull/4389). I don't see FW report error with the PR. But I saw `[ 202.641981] kernel: soundwire sdw-master-0: generic_new_peripheral_assigned: invalid dev_num 1` I have no idea what did I miss, yet.

BTW, I can't reproduce the `invalid dev_num` error on my local sdw devices.

ranj063, 2023-05-31:
@keqiaozhang doe https://github.com/thesofproject/linux/pull/4389 help?

fredoh9, 2023-06-02:
we can close this

closed 2023-06-02 by fredoh9