thesofproject/sof #7703: [BUG] Issues with using Zephyr CONFIG_LOG_MIPI_SYST_USE_CATALOG with SOF

kv2019i, 2023-05-29:
**Describe the bug**
Currently SOF Zephyr builds use "TEXT" logging output by default.

One alternative to text logging is the MIPI SYS-T Catalog logging backend which is found in Zephyr upstream:
https://github.com/zephyrproject-rtos/zephyr/issues/19841
https://docs.zephyrproject.org/latest/samples/subsys/logging/syst/README.html

There are currently some issues enabling this build with SOF:

- [ ] Builds with Cadence xtensa tool chain fail with MIPI_SYST_USE_CATALOG (work/zephyr/lib/os/cbprintf_packaged.c:22:2: error: "CONFIG_CBPRINTF_PACKAGE_SUPPORT_TAGGED_ARGUMENT)
- [ ] No logs from SOF emitted to backend (is_in_log_strings_section() check always returns false for SOF prints, potential issue in linker scripts)

**To Reproduce**
Add following to work/sof/app/prj.conf

```
CONFIG_LOG_BACKEND_ADSP_MTRACE=y
CONFIG_LOG_BACKEND_ADSP_MTRACE_OUTPUT_SYST=y
CONFIG_LOG_MIPI_SYST_ENABLE=y
CONFIG_LOG_MIPI_SYST_USE_CATALOG=n
```

And rebuild SOF. Capture log with "mtrace-reader.py" to a file "dsplog.bin" and post-process with upstream mipi-syst tools like:

systprint -c ../../work/build-tgl/zephyr/mipi_syst_collateral.xml dsplog.bin

**Reproduction Rate**
100%

**Expected behavior**
Sys-T Catalog logging can be enabled for all support logging backends.

**Impact**
Inability to use more effiecient standard dictionary format.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: n/a
    * SOF: e7fc334bda41da7e1c611057e6ccda258595a150
    * Zephyr: 7d54586751cfe405d2cdcfd850f8100bc9844f63
2) Name of the topology file
    * Topology: n/a
3) Name of the platform(s) on which the bug is observed.
    * Platform: Intel MTL
  
**Screenshots or console output**

kv2019i, 2023-05-29:
FYI @abonislawski @marc-hb @dbaluta @nashif @lgirdwood @mwasko

kv2019i, 2023-05-29:
Hmm, interesting, I tried SYS_USE_CATALOG with SOF2.5 (uses 9ff0732e8b17a80773009f1ab435912a3a5e432e , Zephyr ~3.3 e3ae110a05d3427647a03fcbeff4478c195c5a48 and Zephyr SDK 0.15), and SYST_USE_CATALOG works with that combination. So this is potentially a regression on Zephyr side (I see no changes on SOF side). FYI @dcpleung

kv2019i, 2023-05-31:
Tested revering the "logging: Use TYPE_SECTION macros" , but did not seem to help. Filed a bug now to Zephyr side for better tracking. I'll try to bisect better when I just have the chance.

dcpleung, 2023-05-31:
Any instructions on how to reproduce it on hardware? I have a UPX11 to test on. Meanwhile, I will try running Sys-T under QEMU to see if I can reproduce it using Zephyr's sample app.

kv2019i, 2023-06-01:
We now have at least RFC level solutions for all issues. Using these, I did a quick comparison of the code size impact of enabling MIPI_SYST_USE_CATALOG logging (SYST_CAT). Following is a Intel MTL build with xt-clang toolchain default params from SOF main:

- Runtime performance: ~ around same cost to calling context
- Overall image size: 810 KiB (SYST-CAT) versus 867 KiB (TEXT), difference of -57 KiB
- Example sections with notable differences
      - .text -> 317 KiB vs 385 KiB, difference of -68 KiB
      - .rodata -> 317 KiB vs 360 KiB, difference of -43 KiB
      - .log_strings -> 55 KiB vs 0 KiB, difference of +55KiB
     
This is with https://github.com/zephyrproject-rtos/zephyr/pull/58757 that allows to use minimal libc with mipi-syst-lib.

kv2019i, 2023-06-20:
One issue solved (ability to use mipi-syst library with minimal libc now possible with Zephyr mainline). 

Unfortunately a new issue found on MTL where DSP crashes when catalog feature is enabled -> https://github.com/thesofproject/sof/issues/7839

kv2019i, 2023-09-05:
I'm happy to report the last blocking bug has been solved in mainline (just did a retest today), so we can close this.

kv2019i, 2023-09-05:
FYI @dcpleung finally it all works! :)

closed 2023-09-05 by kv2019i