thesofproject/sof #7705: [BUG][IPC4] MTL RVP hangs when core1 & core2 are used simultaneously

RanderWang, 2023-05-30:
**Describe the bug**
MTL RVP hangs when core1 & core2 are used simultaneously. We need to rescue the MTL RVP by cold boot.

**To Reproduce**
Do two streams playback on core1 & core2

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: main branch : 0035b02b10f1ef6417d76f469c04572a3af5b9a3
    * SOF: 005-drop-stable
2) Name of the topology file
    * Topology: development/sof-mtl-nocodec-multicore.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTL RVP B0

abonislawski, 2023-05-30:
Same on main branch?

abonislawski, 2023-05-30:
@RanderWang checked this today with our validation and we have good results when core1 & core2 are used simultaneously.
Looks like driver is not programming something correctly for this scenario?

RanderWang, 2023-05-31:
> @RanderWang checked this today with our validation and we have good results when core1 & core2 are used simultaneously. Looks like driver is not programming something correctly for this scenario?

Thanks! As you know, driver doesn't touch core1 & core2, which are maintained by FW. SOC team is checking the scan dump. I will share you the finding

RanderWang, 2023-06-01:
@abonislawski We got the reply from SOC team and you are in the loop. Do you have any idea ? Thanks!

RanderWang, 2023-06-06:
the scandump show DSP0 and DSP2 are issuing exclusive read to L2SRAM address 0x400ED940. however the transaction is not seen at fabric-to-L2SRAM interface. 

question: are these valid program counter: DSP0 = 0x4006 600C, DSP2 = 0x400E CE82

RanderWang, 2023-06-06:
@abonislawski  yes,  the address is for log_buffer for this scan dump.  I disable log but make no effect on 005 branch.  I tested main branch last week, but to my surprise this bug doesn't happen on latest main branch !

marcinszkudlinski, 2023-06-06:
0xAxxxxx is cached alias
0x4xxxxx is uncached alias
Same SRAM, just accessed through L1 or directly

RanderWang, 2023-06-07:
@abonislawski  no issue on latest main branch. The issue was fixed or workaround by [e2d0d1](https://github.com/zephyrproject-rtos/zephyr/commit/e2d0d1a801df2889fbe68de458fdd98918849924)
````
 kernel: allow for arch specific processing within busy loops

Give architectures that need it the ability to perform special checks
while e.g. waiting for a spinlock to become available.

````

This is proved that our SOC team reported that DSP0 and DSP2 are issuing exclusive read to L2SRAM address 0x400ED940 (log_buffer).

mengdonglin, 2023-06-14:
> @abonislawski no issue on latest main branch. The issue was fixed or workaround by [e2d0d1](https://github.com/zephyrproject-rtos/zephyr/commit/e2d0d1a801df2889fbe68de458fdd98918849924)
> 
> ```
>  kernel: allow for arch specific processing within busy loops
> 
> Give architectures that need it the ability to perform special checks
> while e.g. waiting for a spinlock to become available.
> 
> Signed-off-by: Nicolas Pitre <npitre@baylibre.com>
> ```
> 
> This is proved that our SOC team reported that DSP0 and DSP2 are issuing exclusive read to L2SRAM address 0x400ED940 (log_buffer).

wszypelt, 2023-06-16:
@nashif the problem is not on the FW side, please assign the right person to issue

wszypelt, 2023-06-23:
Fixed with latest zephyr version, @RanderWang can you check?

RanderWang, 2023-06-26:
fixed

closed 2023-06-26 by RanderWang