thesofproject/sof #7714: [BUG] Buffer must be in L2 address space

plbossart, 2023-05-30:
https://sof-ci.ostc.intel.com/#/result/planresultdetail/26668?model=TGLU_RVP_SDW_IPC4ZPH&testcase=multiple-pipeline-all-25

ASSERTION FAIL [aligned_addr >= (3187671040) && aligned_addr < (3187671040) + (1966080)] @ /srv/home/jenkins/workspace/sof_generic_build/zephyr/soc/xtensa/intel_adsp/common/include/intel_adsp_hda.h:166
	Buffer must be in L2 address space
[   50.060696] <err> os:  ** FATAL EXCEPTION
[   50.060706] <err> os:  ** CPU 0 EXCCAUSE 63 (zephyr exception)
[   50.060716] <err> os:  **  PC 0xbe052dbf VADDR (nil)
[   50.060723] <err> os:  **  PS 0x60e20
[   50.060736] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:14 CALLINC:2)
[   50.060746] <err> os:  **  A0 0xbe015028  SP 0xbe0a54d0  A2 0x4  A3 0xbe0a54e0
[   50.060756] <err> os:  **  A4 0xbe0a54c0  A5 0x4  A6 0x30  A7 (nil)
[   50.060766] <err> os:  **  A8 0xbe013531  A9 0xbe0a5440 A10 0xbe0762a8 A11 0xbe0a54e0
[   50.060776] <err> os:  ** A12 0xbe0a54c0 A13 0x4 A14 0x1 A15 0x1
[   50.060786] <err> os:  ** LBEG 0xbe016599 LEND 0xbe0165a8 LCOUNT 0xbe016e25
[   50.060795] <err> os:  ** SAR 0x20

plbossart, 2023-05-30:
this looks like a new regression @kv2019i

lgirdwood, 2023-05-31:
@lyakh fyi

marc-hb, 2023-05-31:
Duplicate of this?
- #7191

lgirdwood, 2023-06-01:
@kv2019i high impact on Linux topologies ? if not, pls move to v2.7

kv2019i, 2023-06-02:
This is a potential duplicate, but I'll keep this separate until we get more data. Let's keep #7191 with the v2.6 tag and I'll move this to v2.7 (as this sighting is on a system configuration we are not planning to support with SOF2.6 and IPC4).

lgirdwood, 2023-07-04:
@plbossart @kv2019i I think the original issue is solved and this duplicate can be closed ?

plbossart, 2023-07-04:
Not sure @lgirdwood, best to ask @keqiaozhang or @fredoh9 if this sighting was fixed, I haven't tracked daily test results in a long time.

fredoh9, 2023-07-05:
I haven't seen FATAL EXCEPTION for a few weeks. I think we can close this.

closed 2023-07-05 by fredoh9