thesofproject/sof #7718: [BUG] Rimage fails to sign fw image using custom private key

aborisovich, 2023-05-31:
**Describe the bug**
During firmware building using xtensa-build-zephyr.py script, rimage fails to generate final sof-mtl.ri image file with error

> "mtl_private_key.pem:    error: unable to open file for reading. No such file or directory (errno = 2)"

**To Reproduce**
Run building command (in "west top" directory):
`python sof\scripts\xtensa-build-zephyr.py mtl -k mtl_private_key.pem`

**Reproduction Rate**
100%.

**Expected behavior**
Rimage generates signed sof-mtl.ri file in build-sof-staging directory.

**Workaround**
Run building command (in "west top" directory) with absolute path to the private key:
`python sof\scripts\xtensa-build-zephyr.py mtl -k C:\Workspace\my_sof_workspace\mtl_private_key.pem`

**Impact**
Annoyance.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: N/A
    * SOF: Main SHA: 40625f0de30e3746cbf59c9ad317b2666fec1b7e
2) Name of the topology file
    * Topology: N/A
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTL (other not tested)
4) Operating system:
    * Windows

**Screenshots or console output**
[error.log](https://github.com/thesofproject/sof/files/11614753/error.log)

aborisovich, 2023-05-31:
Bug is NOT reproduced on the commit: 2df3cba533055db568a5a25295897f607cc2d013 .
Based on the git history of `xtensa-build-zephyr.py` there are 2 more commits forward - that's where would start looking for the bug cause: 5a82b0ab531845cf1ba7c6f4bd42fa1dbb98b773 or a769d3941d81f6d487e1a2bb6c3f99a193ebdcae .
@marc-hb I think it may be related to your recent changes to `west sign` command and process changes.

marc-hb, 2023-05-31:
thanks for @aborisovich for the very good bug report. Simple fix submitted in #7721

commit c81efcf2 (2023-06-02): xtensa-build-zephyr.py: resolve relative --key arguments

Fixes commit a769d3941d81 ("xtensa-build-zephyr.py: stop calling west
sign, rely on west build")

Before that commit, `xtensa-build-zephyr.py` used to invoke `west sign`
directly and in the same current directory. This allowed relative key
paths to just work. Now that `west sign` is indirectly invoked by `west
build`, there is no possible way to make relative key paths work. So we
must resolve them before using them.

Fixes #7718
closed 2023-06-02 by kv2019i