thesofproject/sof #7774: [BUG] Multi-core feature doesn't support 3 cores run at the same time

keqiaozhang, 2023-06-09:
**Describe the bug**
We enabled multi-core tplg in CI and observed that it doesn't support 3 cores run at the same time. It failed with IPC timed out 0x40030004|MOD_INIT_INSTANCE.

**dmesg**
```
[ 2398.696058] kernel: snd_sof:sof_ipc4_widget_setup: sof-audio-pci-intel-mtl 0000:00:1f.3: Create widget dai-copier.SSP.NoCodec-2.playback instance 3 - pipe 6 - core 2
[ 2398.696064] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x40030004|0x203002f: MOD_INIT_INSTANCE [data size: 188]
[ 2399.201595] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x40030004|0x203002f
[ 2399.201612] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Attempting to prevent DSP from entering D3 state to preserve context
[ 2399.201618] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 2399.201643] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0xc0030004|0x203002f|0x0, target: 0x40030004|0x203002f|0x0, ctl: 0x3
[ 2399.201650] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 2399.201655] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 2399.201659] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[ 2399.201663] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[ 2399.201677] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0x5, ROM error: 0x0
[ 2399.201682] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
[ 2399.201689] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit enabled
[ 2399.201693] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
```

**To Reproduce**
TPLG=/lib/firmware/intel/sof-ace-tplg/sof-mtl-nocodec-multicore.tplg MODEL=MTLP_RVP_NOCODEC_MULTICORE ~/sof-test/test-case/multiple-pipeline.sh -f p -l 50

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {[0efaf69a10d7](https://github.com/thesofproject/linux/commit/0efaf69a10d7)}
    * SOF: {[5f118b761bf0](https://github.com/thesofproject/sof/commit/5f118b761bf0)}
2) Name of the topology file
    * Topology: {sof-ace-tplg/sof-mtl-nocodec-multicore.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {MTL-NOCODEC-MULTICORE}

[dmesg.txt](https://github.com/thesofproject/sof/files/11701840/dmesg.txt)
[mtrace.txt](https://github.com/thesofproject/sof/files/11701841/mtrace.txt)

keqiaozhang, 2023-06-21:
New error in CI. Test ID:28022
```
[ 2375.146587] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x45020003|0x50006: MOD_BIND
[ 2375.146874] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x65000000|0x50006: MOD_BIND
[ 2375.146904] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx done : 0x45020003|0x50006: MOD_BIND
[ 2375.146916] kernel: snd_sof:sof_ipc4_route_setup: sof-audio-pci-intel-mtl 0000:00:1f.3: bind gain.4.1:0 -> dai-copier.SSP.NoCodec-1.playback:0
[ 2375.146921] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x45050006|0x50004: MOD_BIND
[ 2375.651700] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x45050006|0x50004
[ 2375.651719] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Attempting to prevent DSP from entering D3 state to preserve context
[ 2375.651724] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 2375.651745] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0x45050006|0x50004|0x0, target: 0x45050006|0x50004|0x0, ctl: 0x3
[ 2375.651753] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 2375.651758] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 2375.651762] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[ 2375.651767] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[ 2375.651781] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0x5, ROM error: 0x0
[ 2375.651786] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
[ 2375.651793] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit enabled
[ 2375.651797] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 2375.651816] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to bind modules gain.4.1:0 -> dai-copier.SSP.NoCodec-1.playback:0
```

fredoh9, 2023-07-26:
Instead of MOD_INIT_INSTANCE, now IPC timeout with MOD_BIND.

```
[ 2495.308510] kernel: snd_sof:sof_ipc4_route_setup: sof-audio-pci-intel-mtl 0000:00:1f.3: bind gain.4.1:0 -> dai-copier.SSP.NoCodec-1.playback:0
[ 2495.308538] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x45050006|0x50004: MOD_BIND
[ 2495.813887] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x45050006|0x50004
[ 2495.813911] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Attempting to prevent DSP from entering D3 state to preserve context
[ 2495.813922] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 2495.813939] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0x45050006|0x50004|0x0, target: 0x45050006|0x50004|0x0, ctl: 0x3
[ 2495.813947] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 2495.813951] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 2495.813955] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[ 2495.813960] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[ 2495.813970] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0x0, ROM error: 0x0
[ 2495.813975] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
[ 2495.813982] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit enabled
[ 2495.813986] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 2495.814011] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to bind modules gain.4.1:0 -> dai-copier.SSP.NoCodec-1.playback:0
```

keqiaozhang, 2023-07-31:
The highest probability of IPC timed out is  "MOD_INIT_INSTANCE" when testing multiple pipelines, The reproduction rate is about 90%.
The second one is "GLB_SET_PIPELINE_STATE", the reproduction rate is about 10%.

mengdonglin, 2023-08-01:
> @keqiaozhang can you retry Tuesday, a multicore fix PR has been merged today (after you ran this test). Thanks !!

The multicore fix is https://github.com/thesofproject/sof/pull/7995/, PR test result https://sof-ci.ostc.intel.com/#/result/planresultdetail/29601

mengdonglin, 2023-08-01:
@lgirdwood We can still reproduce this issue after #7995 was merged in daily test on ww31.2 [here](https://sof-ci.ostc.intel.com/#/result/planresultdetail/29617)

keqiaozhang, 2023-08-01:
@lgirdwood @mengdonglin 
I still can reproduce this issue on MTL-NOCODEC with multi-core tplg, but the issue has improved:
1.  "MOD_BIND" is hard to reproduce, before the repro rate was ~90%.
2.  The major IPC TIMEOUT now is  "GLB_SET_PIPELINE_STATE". repro rate is about 50% or less.

I'm running more tests to confirm it.

keqiaozhang, 2023-08-02:
@lgirdwood  interesting, I cannot reproduce this issue with PR https://github.com/thesofproject/sof/pull/8000 plus debug logging enabled.

lgirdwood, 2023-08-02:
> @lgirdwood interesting, I cannot reproduce this issue with PR #8000 plus debug logging enabled.

ok, can you switch debug OFF and try again. Could be timing related ? Thanks !

keqiaozhang, 2023-08-03:
I applied PR https://github.com/thesofproject/sof/pull/8000 only and tested more than 1K iterations, only failed once with "GLB_SET_PIPELINE_STATE". I think PR https://github.com/thesofproject/sof/pull/8000 is good to merge.

keqiaozhang, 2023-08-03:
BTW, this issue is very easy to reproduce when mtrace is disabled, even with PR https://github.com/thesofproject/sof/pull/8000.
Here is the test results:https://sof-ci.ostc.intel.com/#/result/planresultdetail/29740

keqiaozhang, 2023-08-10:
Unfortunately, this issue still can be reproduced on tip main branch even with mtrace in CI testing. There is a slight difference in reprod rate among different devices. All MTL in CI are upgraded to C0 SI.  In theory, there is no significant difference in hardware configuration.

wszypelt, 2023-08-11:
I will try to reproduce it in python tests

prolkows, 2023-08-21:
> BTW, this issue is very easy to reproduce when **mtrace is DISABLED**, even with PR #8000. Here is the test results:https://sof-ci.ostc.intel.com/#/result/planresultdetail/29740

It's hard to reproduce this issue locally in python tests without IPC payloads.

Could you please rerun this PR run with IPC payload enabled? For me, it will be easier to reproduce if I get passed tests results :)

RanderWang, 2023-08-28:
I did many tests and found that if IPC4 logging is disabled, multi-core test can pass. I found one infinite loop in fw on one device but it can't be reproduced on another one. I will update the status when I get any hint

RanderWang, 2023-08-28:
> It's hard to reproduce this issue locally in python tests without IPC payloads.
>
> Could you please rerun this PR run with IPC payload enabled? For me, it will be easier to reproduce if I get passed [...]

For the HW hang with multicore feature, it can't be reproduced by python test and only found with Linux audio test. Finally found the root reason by HW team

lgirdwood, 2023-08-28:
> I did many tests and find that if IPC4 logging is disabled, multi-core test can pass. I found one infinite loop in fw on one device but it can't be reproduced on another one. I will update the status when I get any hint

@RanderWang can you fix infinite loop with a timout check anbd log error. We think some were introduced when drivers were added to Zephyr. @kv2019i @abonislawski fyi.

kv2019i, 2023-08-28:
@RanderWang Which infinite loop did you see? Let's add a timeout to these, so we can exclude them out. I pushed first such fix now -> https://github.com/zephyrproject-rtos/zephyr/pull/61967

RanderWang, 2023-08-29:
Maybe it is cause by https://github.com/thesofproject/sof/issues/8115

lyakh, 2023-08-30:
Can we test the same on TGL?

keqiaozhang, 2023-09-06:
This issue has been improved a lot after https://github.com/thesofproject/sof/issues/8115 merged. Only multiple-pause-resume test failed in today's CI testing.

mengdonglin, 2023-09-08:
After PR #8149 got merged, this issue cannot be reproduced on ww36.4

wszypelt, 2023-09-08:
@keqiaozhang request for confirmation and closing the bug

fredoh9, 2023-09-08:
this issue cannot be reproduced on ww36.5 either. Let me do some manually test before closing the issue.

fredoh9, 2023-09-08:
re-run multicore test with daily build(planresultdetail/31515) and manually ran multiple-pipeline.sh. Looks very stable.
I think we can close this.

closed 2023-09-08 by fredoh9