thesofproject/sof #7776: [BUG] [Multi-core] alsabat test fails on core 0 pipelines

keqiaozhang, 2023-06-09:
**Describe the bug**
It seems that no internal loopback support on core 0 pipelines. All captured data is 0. But no such issue on core1/2 pipelines.

**To Reproduce**
 ~/sof-test/test-case/check-alsabat.sh -p hw:sofnocodec,0 -c hw:sofnocodec,0 -C 2 -F 821

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {[0efaf69a10d7](https://github.com/thesofproject/linux/commit/0efaf69a10d7)}
    * SOF: {[5f118b761bf0](https://github.com/thesofproject/sof/commit/5f118b761bf0)}
2) Name of the topology file
    * Topology: {sof-ace-tplg/sof-mtl-nocodec-multicore.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {MTL-NOCODEC-MULTICORE}

[image]

wszypelt, 2023-06-15:
@keqiaozhang Request for kernel logs with complete IPC payloads

keqiaozhang, 2023-06-20:
Confirmed that this issue was fixed by https://github.com/thesofproject/sof/pull/7818/

closed 2023-06-20 by keqiaozhang