thesofproject/sof #7780: Sound card modprobe fails on Dell XPS13 (9320)

e8johan, 2023-06-11:
**Describe the bug**
The default Debian testing kernel does not initialize the sound card properly, so I've built a custom 6.4.0-rc3 kernel based on the SOF project kernel. I've also installed the latest (v2.2.5) sof-bin firmwares

Both at boot and subsequent modprobes, the same error message is given.

**To Reproduce**
Either reboot or cold boot, or run:

`sudo modprobe -r snd-sof-pci-intel-tgl; sudo modprobe snd-sof-pci-intel-tgl`

**Reproduction Rate**
Always reproducable. Cold boot, reboot, modprobe while running.

**Expected behavior**
I expect the soundcard to initialize and mic and speakers to be available.

**Impact**
This means no sound, so a showstopper.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 6.4.0-rc3
    * SOF: v2.2.5
2) Name of the topology file
    * Topology: sof-adl-rt1316-l1-mono-rt714-l0.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: Debian testing with a custom kernel and custom installed SOF binary firmware.

**Screenshots or console output**

```
[  878.316176] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[  878.316301] sof-audio-pci-intel-tgl 0000:00:1f.3: SoundWire enabled on CannonLake+ platform, using SOF driver
[  878.316546] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[  878.316657] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[  878.323366] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[  878.342066] sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 0
[  878.342070] sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 4
[  878.344398] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[  878.344400] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[  878.344402] sof-audio-pci-intel-tgl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
[  878.461624] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[  878.461639] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[  878.479105] sof-audio-pci-intel-tgl 0000:00:1f.3: Direct firmware load for intel/sof-tplg/sof-adl-rt1316-l1-mono-rt714-l0.tplg failed with error -2
[  878.479114] sof-audio-pci-intel-tgl 0000:00:1f.3: error: tplg request firmware intel/sof-tplg/sof-adl-rt1316-l1-mono-rt714-l0.tplg failed err: -2
[  878.479121] sof-audio-pci-intel-tgl 0000:00:1f.3: you may need to download the firmware from https://github.com/thesofproject/sof-bin/
[  878.479124] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load DSP topology -2
[  878.479127] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -2
[  878.479145] sof_sdw sof_sdw: ASoC: failed to instantiate card -2
[  878.479320] sof_sdw sof_sdw: snd_soc_register_card failed -2
[  878.479326] sof_sdw: probe of sof_sdw failed with error -2
```

plbossart, 2023-06-12:
well the problem is that the topology file is not present on your system.

````
[  878.479105] sof-audio-pci-intel-tgl 0000:00:1f.3: Direct firmware load for intel/sof-tplg/sof-adl-rt1316-l1-mono-rt714-l0.tplg failed with error -2
[  878.479114] sof-audio-pci-intel-tgl 0000:00:1f.3: error: tplg request firmware intel/sof-tplg/sof-adl-rt1316-l1-mono-rt714-l0.tplg failed err: -2
[  878.479121] sof-audio-pci-intel-tgl 0000:00:1f.3: you may need to download the firmware from https://github.com/thesofproject/sof-bin/
````

We've provided this support but it's not been released just yet. It should be part of SOF 2.6 that's going to be released in a couple of weeks. In the mean time, you can copy and unzip this file in /lib/firmware/intel/sof-tplg/

[sof-adl-rt1316-l1-mono-rt714-l0.tplg.gz](https://github.com/thesofproject/sof/files/11725738/sof-adl-rt1316-l1-mono-rt714-l0.tplg.gz)

@mengdonglin please make sure the last topology files have been back ported and are on the menu for 2.6, thanks!

e8johan, 2023-06-12:
**Thank you!** That solved the modprobe scenario. However, in the initramfs (I run an encypted root filesystem), I now run into the following:

```
[    3.853133] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[    3.853151] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[    3.860910] sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:26:0 Kernel ABI 3:23:0
[    3.861014] sof-audio-pci-intel-tgl 0000:00:1f.3: error: can't connect DAI HDA0.OUT stream iDisp1
[    3.861128] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to add widget id 0 type 27 name : HDA0.OUT stream iDisp1
[    3.861249] sof_sdw sof_sdw: ASoC: failed to load widget HDA0.OUT
[    3.861319] sof_sdw sof_sdw: ASoC: topology: could not load header: -22
[    3.861407] sof-audio-pci-intel-tgl 0000:00:1f.3: error: tplg component load failed -22
[    3.861500] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load DSP topology -22
[    3.861590] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[    3.861718] sof_sdw sof_sdw: ASoC: failed to instantiate card -22
[    3.861916] sof_sdw sof_sdw: snd_soc_register_card failed -22
[    3.861977] sof_sdw: probe of sof_sdw failed with error -22
[    8.972910] PM: Image not found (code -22)
```

I copy all files from `/lib/firmware/intel/sof/intel-signed`, `/lib/firmware/intel/sof/community`, `/lib/firmware/intel/sof` and `/lib/firmware/intel/sof-tplg` into the initramfs, but I guess I'm missing something.

plbossart, 2023-06-12:
I don't understand what the initramfs has to do with the topology. Topology files should not be there.

e8johan, 2023-06-13:
That is probably my ignorance. I'm trying to apply these instructions from Arch in Debian: https://wiki.archlinux.org/title/Dell_XPS_13_Plus_(9320) .

This is what a cold boot looks like if I don't put any files into the initramfs, so I guess I need to find what files to include, or somehow make the modprobe happen later in the boot sequence.

```
[    3.564472] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[    3.577947] sof-audio-pci-intel-tgl 0000:00:1f.3: codec #2 probe error, ret: -2
[    3.578487] sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 0
[    3.578489] sof-audio-pci-intel-tgl 0000:00:1f.3: no hda codecs found!
[    3.578513] sof-audio-pci-intel-tgl 0000:00:1f.3: Direct firmware load for intel/sof/sof-adl.ri failed with error -2
[    3.578516] sof-audio-pci-intel-tgl 0000:00:1f.3: error: sof firmware file is missing, you might need to
[    3.578575] sof-audio-pci-intel-tgl 0000:00:1f.3:        download it from https://github.com/thesofproject/sof-bin/
[    3.578635] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load DSP firmware -2
[    3.579354] sof-audio-pci-intel-tgl 0000:00:1f.3: error: sof_probe_work failed err: -2
```

And then during a forced modprobe using `sudo modprobe -r snd-sof-pci-intel-tgl; sudo modprobe snd-sof-pci-intel-tgl` :

```
[   27.339319] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[   27.339396] sof-audio-pci-intel-tgl 0000:00:1f.3: SoundWire enabled on CannonLake+ platform, using SOF driver
[   27.339582] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[   27.339648] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   27.345449] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[   27.363623] sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 0
[   27.363627] sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 4
[   27.365926] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[   27.365928] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[   27.365931] sof-audio-pci-intel-tgl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
[   27.481030] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[   27.481040] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[   27.490634] sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:26:0 Kernel ABI 3:23:0
[   27.490952] sof_sdw sof_sdw: ASoC: Parent card not yet available, widget card binding deferred
[   27.526253] sof_sdw sof_sdw: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 3
[   27.546933] input: sof-soundwire HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input36
[   27.546967] input: sof-soundwire HDMI/DP,pcm=6 as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input37
[   27.546984] input: sof-soundwire HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input38
```

(but I guess I digress from the original bug, so feel free to close this issue if you feel that I'm heading off-topic)

closed 2023-06-13 by plbossart

plbossart, 2023-06-15:
@nezia1 I wonder if you are missing a module in your list, if you need to explicitly list them all in your MODULES definition above then you are obviously missing snd-soc-hdac-hda in your list. This module is required for HDMI support.

nezia1, 2023-06-19:
@plbossart Adding snd-soc-hdac-hda unfortunately did not work. I wonder if there might be something up with the order of modules? I'm not too experienced with working with drivers so I have no idea what could be wrong with my setup. It gave me the same output as before.
My MODULES definition currently looks like this: `MODULES=(snd_soc_rt715_sdca snd_soc_rt1316_sdw snd_sof_pci_intel_tgl snd_hda_intel snd_soc_sof_sdw snd-soc-hdac-hda i915)`