thesofproject/sof #7781: [BUG] Rimage revision defined in git submodule does not match the one defined in west.yml manifest

aborisovich, 2023-06-12:
**Describe the bug**
Git submodule rimage revision had not been updated during last update of the west.yml file.
Looks like regression introduced in the https://github.com/thesofproject/sof/pull/7630 .
It had been changed in west manifest from ab0429fdbe563ef6abe499c69b2483e96c4762d0 to
4ce79b152e23c7b3937a09ef57cdd69d125a9214 however, during work with git rimage submodule shows changes.

**To Reproduce**
1. execute command `west update`
2. execute command `git status` (now rimage submodule has changes - dirty state)
3. execute command `git submodule update --init --recursive` and confirm that SHA ab0429fdbe563ef6abe499c69b2483e96c4762d0 is checked out (bad revision).

**Reproduction Rate**
100%.

**Expected behavior**
`Git status` does not show changes in rimage submodule.

**Impact**
Annoyance.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: N/A
    * SOF: main
2) Name of the topology file
    * Topology: N/A
3) Name of the platform(s) on which the bug is observed.
    * Platform: Zephyr platforms

**Screenshots or console output**
[rimage bad revision.log](https://github.com/thesofproject/sof/files/11716180/rimage.bad.revision.log)

aborisovich, 2023-06-12:
@fredoh9 please double check whether your PR introduces this and fix if I am right.
Guys please take cautious with this. I know it's tiresome to track revision in both git and west but please do so until we migrate to west completely.

marc-hb, 2023-06-12:
I think it was not @fredoh9 but @andyross in 7511a1da153364c0e6ba2f990d43a272889af96e

marc-hb, 2023-06-12:
99.99% sure this has just been fixed by #7782 but please re-open if I got it wrong.

closed 2023-06-12 by marc-hb