thesofproject/sof #7801: [FEATURE][SMART_AMP] Revamp smart_amp component for adaptive solutions

johnylin76, 2023-06-14:
**Is your feature request related to a problem? Please describe.**
The current smart_amp component implementation [source code](https://github.com/thesofproject/sof/tree/main/src/audio/smart_amp) is named in a generic matter. It contains smart_amp.c as the role of common adapter (handling component API), and smart_amp_[generic|maxim_dsm].c for generic and Maxim DSM solution respectively. However, the implementation inside doesn't perform in the same manner. The design is deeply bound to Maxim DSM solution, which is not surprising since Maxim is the initial author and there is no other solution supported for a long time.

**Describe the solution you'd like**
Revamp smart_amp to be in a generic matter, and adaptive to other solutions.
The expected framework would be alike module_adapter.

**Describe alternatives you've considered**
1. ask the new solution provider to create another smart_amp component and keep the present one untouched.
     - cons: will produce code duplication
2. implement the new smart_amp solution by module_adapter. Given that there is the on-going project to convert existing components into modules with module_adapter
     - cons: not sure about the feasibility since amart_amp is multi-input (2 source, 1 sink) component

johnylin76, 2023-06-15:
Hi @lgirdwood ,
This issue needs your input especially for the 2nd point of alternatives.

If the on-going plan in mainstream is to migrate the processing components to module_adapter form, would you suggest to do smart_amp revamp task for adapting new smart_amp solution?
Does module_adapter support multiple source or sink already?

johnylin76, 2023-06-15:
+ @cujomalainey for viz

cujomalainey, 2023-06-27:
@ranj063 I believe I heard your name come up last time I talked about this, feel free to unsubscribe if I am having a mental cache miss.

lgirdwood, 2023-07-05:
@johnylin76 sorry for delay, yeah I agree - a generic smart amp layer would be the best option. We already have the "dummy" smart amp logic which could be modified to become a generic layer. Today the dummy smart amp just copies data, so its probably a good starting point. Also adding @lyakh @aiChaoSONG @mengdonglin. Think it may be too tight to squeeze this in for v2.7, v2.8 probably more resonable.

johnylin76, 2023-07-07:
@lgirdwood Thanks for the advice.
As for the dummy smart amp mentioned above, you mean [smart_amp_test](https://github.com/thesofproject/sof/blob/main/src/samples/audio/smart_amp_test_ipc4.c) implementation, is that correct?
That is indeed a good example to make the nifty re-cut for the present smart_amp implementation, picking out maxim-specific codes from the generic layer.
It is probably not a complicated task, I am happy to help if you don't mind.

lgirdwood, 2023-07-07:
> @lgirdwood Thanks for the advice. As for the dummy smart amp mentioned above, you mean [smart_amp_test](https://github.com/thesofproject/sof/blob/main/src/samples/audio/smart_amp_test_ipc4.c) [...]

@lyakh can you point @johnylin76 to your latest/correct code. Thanks

johnylin76, 2023-07-08:
https://github.com/thesofproject/sof/pull/7907 is created for Kconfig hierarchy change. The code is not touched yet.
Will wait for the pointer provided by @lyakh

lyakh, 2023-07-10:
@johnylin76 yes, https://github.com/thesofproject/sof/blob/main/src/samples/audio/smart_amp_test_ipc4.c is the smart amp test, that's currently used. What @lgirdwood meant was probably a conversion of that module to a loadable module as performed in #7867 . But so far in "main" we only have https://github.com/thesofproject/sof/blob/main/src/samples/audio/smart_amp_test_ipc4.c and https://github.com/thesofproject/sof/tree/mtl-005-drop-stable/src/audio/smart_amp

johnylin76, 2023-07-10:
Thanks for the information.
With reference to #7867 the long-term goal looks to me is using module_adapter for all types of processing including smart_amp. I suppose that the project for unifying module_adapter use has a long way to go. Due to the production request on our side for the adaptive smart_amp solutions which targets for the near future, the code work for present smart_amp component should be still valued.

@lyakh @lgirdwood Does that make sense to you? If yes I will put it to my To-Do list then.

Thanks

lgirdwood, 2023-07-10:
> Thanks for the information. With reference to #7867 the long-term goal looks to me is using module_adapter for all types of processing including smart_amp. 

Long term goal is to use the module API for all modules.

> I suppose that the project for unifying module_adapter use has a long way to go. Due to the production request on our side for the adaptive smart_amp solutions which targets for the near future, the [...]

Yes, at a high level all smart amps will look very similar at the module level (to the module API, behaviour and topology), the differences will be at the 3rd party blob API. So agreed, having having a generic part for module API, topology and behaviour makes good sense, it will need a generic 3rd party API to plug into blobs.

cujomalainey, 2023-07-11:
> Yes, at a high level all smart amps will look very similar at the module level (to the module API, behaviour and topology), the differences will be at the 3rd party blob API. So agreed, having [...]

Couple of Qs

1. do we plan to support all/multiple smart amps in a single build
2. do we want to support this via dynamic modules
3. is the smart amp interface its own UUID or a middle layer? The former will possibly have loading implications, i.e. if dynamic modules we may need to build into a single blob all needed DSM and load them all.

johnylin76, 2023-07-11:
> Couple of Qs
>
> 1. do we plan to support all/multiple smart amps in a single build
> 2. do we want to support this via dynamic modules
> 3. is the smart amp interface its own UUID or a middle layer? The [...]

Share with my ideas:
1. Supporting at most 1 smart amp solution in one build should be fair enough. I cannot think of any practical case requiring 2 or more smart amp solutions.
2. As a result of 1, my plan for smart amp revamp is still as a static module
3. My plan is that each solution owns its UUID for smart_amp. On FW building only one solution can be selected via Kconfig, UUID is used for making sure the selection on both FW and topology side are aligned. So the individual solution won't be misused or abused.

Unlike module_adapter which can runtime determine the inner processing, the modular design for smart_amp is only selective while building (more like the implementation choice e.g. EQIIR in generic, HIFI3, or HIFI4)

cujomalainey, 2023-07-11:
> 1. Supporting at most 1 smart amp solution in one build should be fair enough. I cannot think of any practical case requiring 2 or more smart amp solutions.

I mean, shipping 1 SOF for base firmware across all boards for both chromeos and linux upstream. Think if Maxim added a new DSM that is incompatible with MAX98373 but a platform supported both. Would make firmware management a nightmare.

> 2. As a result of 1, my plan for smart amp revamp is still as a static module
> 3. My plan is that each solution owns its UUID for smart_amp. On FW building only one solution can be selected via [...]

If you are giving them their own UUID you are doing a middle layer approach and then there is no shipping 1 limitation since they are identified individually.

cujomalainey, 2023-07-11:
Also I'm assuming, in the context of dynamic modules, we would have to duplicate code in memory if we did have >1 smart amp loaded on a machine as the code would be loaded for each module. Unlikely but a consequence of that particular scenario.

johnylin76, 2023-07-16:
> I mean, shipping 1 SOF for base firmware across all boards for both chromeos and linux upstream. Think if Maxim added a new DSM that is incompatible with MAX98373 but a platform supported both. [...]

Yes. I believe that is our plan and the reason we need to revamp smart_amp design.

As for the base firmware across all boards for both chromeos and linux upstream, that should be the SOF build with the dummy smart_amp solution. We are not able to provide (or ask partner to provide) the 3P smart_amp solution across all boards.

> If you are giving them their own UUID you are doing a middle layer approach and then there is no shipping 1 limitation since they are identified individually.

Indeed. Although my original plan is to propose the smart_amp design a bit simpler than module_adapter, by  making solutions mutually exclusive, a.k.a. 1 smart_amp solution in 1 build. It needs to be re-evaluated after taking the maintenance effort into consideration. 

> Also I'm assuming, in the context of dynamic modules, we would have to duplicate code in memory if we did have >1 smart amp loaded on a machine as the code would be loaded for each module. Unlikely but a consequence of that particular scenario.

I do see the possibilities that one platform may need to support multliple smart_amp solutions. given that the solution is majorly determined by OEMs, e.g. assuming MTL ChromeOS devices, Acer wants smart_amp solution A while HP wants smart_amp solution B. Then it will turn to the design structure similar to module_adapter. However, before we have the dynamic library support, I don't think we would like to make both solutions available in one build as it produces the memory redundancy. It is hardly possible in my view to have a ChromeOS project wanting both solutions to be loaded simultaneously

lgirdwood, 2023-09-29:
@johnylin76 can we close now or are more PRs needed ?

closed 2023-10-02 by johnylin76