thesofproject/sof #7813: [BUG][Multi-core] DMIC capture failed when module-copier added to the pipeline

yongzhi1, 2023-06-15:
**Describe the bug**
It's the same IPC timeout error reported in https://github.com/thesofproject/sof/issues/7649, however, the issue is only reproducible when PASSTHROUGH set to false for simple DMIC capture.

**To Reproduce**
Upon boot, run arecord on DMIC, the 1st time is always successful, wait 5 sec and run the 2nd time and observe the failure.

**Reproduction Rate**
100%

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://chromium.googlesource.com/chromiumos/third_party/kernel/+/refs/heads/chromeos-6.1
    * SOF: 005-drop-stable
2) Name of the topology file
    * Topology: development/sof-mtl-nocodec-multicore-4ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTL Rex

Please also include the relevant sections from the firmware log and kernel log in the report.
[image]

[dmic-core1-ipc-timedout.txt](https://github.com/thesofproject/sof/files/11761843/dmic-core1-ipc-timedout.txt)
[mtrace-dmic-core1-deadlock.txt](https://github.com/thesofproject/sof/files/11761860/mtrace-dmic-core1-deadlock.txt)

serhiy-katsyuba-intel, 2023-06-27:
Most probably, the root cause for this issue is same as for https://github.com/thesofproject/sof/issues/7649
The PR with fix for mtl-005-drop-stable branch: https://github.com/thesofproject/zephyr/pull/68

I was not able to properly test it as in recent development/sof-mtl-nocodec-multicore-4ch.tplg I have DMIC pipeline is on core 0 but on the attached image DMIC pipeline is on core 2. So, probably, development/sof-mtl-nocodec-multicore-4ch.tplg was modified some time ago? With the fix, arecord on DMIC works with no problem.

keqiaozhang, 2023-06-30:
@yongzhi1 https://github.com/thesofproject/zephyr/pull/68 has been merged to mtl-005 branch, can we close this bug?

closed 2023-07-06 by yongzhi1