thesofproject/sof #7814: [BUG] failed to prepare widget dai-copier.SSP.NoCodec-0.capture with 2 cores tplg

keqiaozhang, 2023-06-16:
**Describe the bug**
We added a simple multicore tplg under development folder for MTL: SSP0 playback on core 0 and SSP2 playback on core 1,capture uses core 0 only atm. But we found that capture test failed with:
```
[  233.401417] kernel: snd_intel_dspcfg:nhlt_check_ep_match: sof-audio-pci-intel-mtl 0000:00:1f.3: Endpoint: vbus_id=0 link_type=3 dir=0 dev_type = 0
[  233.401421] kernel: snd_intel_dspcfg:nhlt_check_ep_match: sof-audio-pci-intel-mtl 0000:00:1f.3: Endpoint: vbus_id=0 link_type=3 dir=1 dev_type = 0
[  233.401425] kernel: snd_intel_dspcfg:nhlt_check_ep_match: sof-audio-pci-intel-mtl 0000:00:1f.3: Endpoint: vbus_id=2 link_type=3 dir=0 dev_type = 0
[  233.401429] kernel: snd_intel_dspcfg:nhlt_check_ep_match: sof-audio-pci-intel-mtl 0000:00:1f.3: Endpoint: vbus_id=2 link_type=3 dir=1 dev_type = 0
[  233.401432] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: no matching blob for sample rate: 48000 sample width: 32 channels: 2
[  233.401463] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to prepare widget dai-copier.SSP.NoCodec-0.capture
[  233.401485] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Failed to prepare connected widgets
[  233.401530] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed widget list set up for pcm 0 dir 1
[  233.401553] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
[  233.401599] kernel:  Port0: ASoC: error at __soc_pcm_hw_params on Port0: -22
[  233.401620] kernel:  Port0: ASoC: error at dpcm_fe_dai_hw_params on Port0: -22
```
No such issue with 3cores tplg. 

**To Reproduce**
TPLG=/lib/firmware/intel/development/sof-mtl-nocodec-multicore-ssp0-ssp2.tplg MODEL=MTLP_RVP_NOCODEC-multicore-2cores ~/sof-test/test-case/check-capture.sh -d 3 -l 1 -r 1 -F

**Reproduction Rate**
100%

[image]

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {[2776754a4cd9](https://github.com/thesofproject/linux/commit/2776754a4cd984a390a17a0c96635beb2ad408d8)}
    * SOF: {[75e8f4b63c16](https://github.com/thesofproject/sof/commit/75e8f4b63c168fb7e7ee11b9db60812a57c9f915)}
2) Name of the topology file
    * Topology: {development/sof-mtl-nocodec-multicore-ssp0-ssp2.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {MTLP_RVP_NOCODEC}

[dmesg.txt](https://github.com/thesofproject/sof/files/11765589/dmesg.txt)

juimonen, 2023-06-16:
I guess the dai_index 1 is not matching the vbus_id in the nhlt blob (where you have only 0 and 2), not sure where this is coming.

btw. in the development dir makefile you have same nhlt.bin names for several of these multicore topologies. It doesn't matter that much, but they are overriding each other and you can't see what's inside the previous topology...

juimonen, 2023-06-16:
Object.Pipeline.io-gateway-capture [
        {
                index 8
                direction capture

                Object.Widget.dai-copier."1" {
                        dai_index       1

Quite sure that dai_index should be 2

juimonen, 2023-06-16:
and I meant it should be 0.... sorry

keqiaozhang, 2023-06-16:
I tried to change the dai_index to 0.
```
diff --git a/tools/topology/topology2/cavs-nocodec-multicore.conf b/tools/topology/topology2/cavs-nocodec-multicore.conf
index d0007895d..acd5b5164 100644
--- a/tools/topology/topology2/cavs-nocodec-multicore.conf
+++ b/tools/topology/topology2/cavs-nocodec-multicore.conf
@@ -243,7 +243,7 @@ Object.Pipeline.io-gateway-capture [
                direction capture

                Object.Widget.dai-copier."1" {
-                       dai_index       1
+                       dai_index       0
                        dai_type        "SSP"
                        type            dai_out
                        copier_type     "SSP"

```
 but it failed in the same way:

```
[28125.083515] snd_intel_dspcfg:nhlt_check_ep_match: sof-audio-pci-intel-mtl 0000:00:1f.3: Endpoint: vbus_id=0 link_type=3 dir=0 dev_type = 0
[28125.083520] snd_intel_dspcfg:nhlt_check_ep_match: sof-audio-pci-intel-mtl 0000:00:1f.3: Endpoint: vbus_id=0 link_type=3 dir=1 dev_type = 0
[28125.083525] snd_intel_dspcfg:nhlt_check_ep_match: sof-audio-pci-intel-mtl 0000:00:1f.3: Endpoint: vbus_id=2 link_type=3 dir=0 dev_type = 0
[28125.083530] snd_intel_dspcfg:nhlt_check_ep_match: sof-audio-pci-intel-mtl 0000:00:1f.3: Endpoint: vbus_id=2 link_type=3 dir=1 dev_type = 0
[28125.083534] sof-audio-pci-intel-mtl 0000:00:1f.3: no matching blob for sample rate: 48000 sample width: 32 channels: 2
[28125.083591] sof-audio-pci-intel-mtl 0000:00:1f.3: failed to prepare widget dai-copier.SSP.NoCodec-0.capture
[28125.083638] sof-audio-pci-intel-mtl 0000:00:1f.3: Failed to prepare connected widgets
[28125.083684] sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed widget list set up for pcm 0 dir 1
[28125.083741] sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
[28125.083816]  Port0: ASoC: error at __soc_pcm_hw_params on Port0: -22
[28125.083846]  Port0: ASoC: error at dpcm_fe_dai_hw_params on Port0: -22
```

juimonen, 2023-06-16:
fix here https://github.com/thesofproject/sof/pull/7818, includes commit also for nhlt bin names

keqiaozhang, 2023-06-20:
Closing this bug as https://github.com/thesofproject/sof/pull/7818 has been merged.

closed 2023-06-20 by keqiaozhang

yw662, 2024-06-17:
I am facing a simila issue with `sof-audio-pci-intel-mtl` and `dai-copier.DMIC.dmic01.capture`, on `fedora desktop 40 linux 6.9.4-200.fc40.x86_64`, redmi book pro 16 2024, `Intel Meteor Lake-P HD Audio Controller (rev 20)`, `Intel Core Ultra 5 125H`:
```
00:1f.3 Multimedia audio controller: Intel Corporation Meteor Lake-P HD Audio Controller (rev 20)
        Subsystem: Xiaomi Device 2309
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 32, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 207
        IOMMU group: 15
        Region 0: Memory at 48192c0000 (64-bit, non-prefetchable) [size=16K]
        Region 4: Memory at 4819000000 (64-bit, non-prefetchable) [size=2M]
        Capabilities: [50] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=55mA PME(D0-,D1-,D2-,D3hot+,D3cold+)
                Status: D3 NoSoftRst+ PME-Enable+ DSel=0 DScale=0 PME-
        Capabilities: [c0] Vendor Specific Information: Len=14 <?>
        Capabilities: [60] MSI: Enable+ Count=1/1 Maskable- 64bit+
                Address: 00000000fee00c78  Data: 0000
        Kernel driver in use: sof-audio-pci-intel-mtl
        Kernel modules: snd_hda_intel, snd_sof_pci_intel_mtl
```

and dmesg:
```
[    5.214572] sof-audio-pci-intel-mtl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    5.214605] sof-audio-pci-intel-mtl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    5.214618] sof-audio-pci-intel-mtl 0000:00:1f.3: enabling device (0000 -> 0002)
[    5.214871] sof-audio-pci-intel-mtl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    5.214962] sof-audio-pci-intel-mtl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    5.221480] sof-audio-pci-intel-mtl 0000:00:1f.3: use msi interrupt mode

[    5.252447] sof-audio-pci-intel-mtl 0000:00:1f.3: hda codecs found, mask 5
[    5.252449] sof-audio-pci-intel-mtl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    5.252453] sof-audio-pci-intel-mtl 0000:00:1f.3: DMICs detected in NHLT tables: 4
[    5.290415] sof-audio-pci-intel-mtl 0000:00:1f.3: Firmware paths/files for ipc type 1:
[    5.290419] sof-audio-pci-intel-mtl 0000:00:1f.3:  Firmware file:     intel/sof-ipc4/mtl/sof-mtl.ri
[    5.290420] sof-audio-pci-intel-mtl 0000:00:1f.3:  Firmware lib path: intel/sof-ipc4-lib/mtl
[    5.290421] sof-audio-pci-intel-mtl 0000:00:1f.3:  Topology file:     intel/sof-ace-tplg/sof-hda-generic-4ch.tplg
[    5.319254] sof-audio-pci-intel-mtl 0000:00:1f.3: Loaded firmware library: ADSPFW, version: 2.9.0.1

[    5.455876] sof-audio-pci-intel-mtl 0000:00:1f.3: Booted firmware version: 2.9.0.1
[    5.477375] sof-audio-pci-intel-mtl 0000:00:1f.3: Topology: ABI 3:29:0 Kernel ABI 3:23:0

[    5.555820] input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input12
[    5.555912] input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input13
[    5.555983] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input14
[    5.556034] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input15
[    5.556075] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input16

...

[   12.791191] sof-audio-pci-intel-mtl 0000:00:1f.3: failed to prepare widget dai-copier.DMIC.dmic01.capture
[   12.791192] sof-audio-pci-intel-mtl 0000:00:1f.3: Failed to prepare connected widgets
[   12.791193] sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed widget list set up for pcm 6 dir 1
[   12.791194] sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
[   12.791282] sof-audio-pci-intel-mtl 0000:00:1f.3: no matching blob for sample rate: 48000 sample width: 32 channels: 4
```

Should I open a new issue for that ?

plbossart, 2024-06-17:
@yw662 for the redmi issues see https://github.com/thesofproject/linux/issues/5053, the 'no matching blob' is a known error and there's already a fix shared upstream.

yw662, 2024-06-18:
@plbossart Thanks I am so glad to find that it is already solved 5 days ago. You guys are great. I will be back to confirm once fedora shipped the updates.

yw662, 2024-09-06:
Fedora is now on Linux 6.10.7 SOF 2024.06 and the redmi issue is fixed