thesofproject/sof #7827: [BUG][stable-v2.6] fatal exception CPU 0 EXCCAUSE 13 (load/store PIF data error) on TGLH-ipc4 platform

keqiaozhang, 2023-06-19:
**Describe the bug**
Observed this bug on stable-v2.6 branch. This issue happened when testing check-capture-100times.

**dmesg**
```
[ 1698.860448] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx      : 0x1b0a0000|0x0: GLB_NOTIFICATION|EXCEPTION_CAUGHT
[ 1698.860449] kernel: snd_sof:sof_ipc4_rx_msg: sof-audio-pci-intel-tgl 0000:00:1f.3: Unhandled DSP message: 0x1b0a0000|0x0
[ 1698.860451] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done : 0x1b0a0000|0x0: GLB_NOTIFICATION|EXCEPTION_CAUGHT
[ 1699.355176] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x13000004|0x1
[ 1699.355190] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Attempting to prevent DSP from entering D3 state to preserve context
[ 1699.355194] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 1699.355242] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[ 1699.355248] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 1699.355299] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Host IPC initiator: 0x93000004|0x1|0x0, target: 0x0|0x0|0x80000000, ctl: 0x3
[ 1699.355305] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 1699.355309] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 1699.355312] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: IPC timeout
[ 1699.355316] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[ 1699.355354] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x00000005: module: ROM, state: FW_ENTERED, running
[ 1699.355393] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: extended rom status:  0x5 0x0 0x0 0x0 0x0 0x0 0x0 0x1
[ 1699.355397] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 1699.355437] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to set final state 4 for all pipelines
[ 1699.355444] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[ 1699.355451] kernel:  DMIC Raw: ASoC: trigger FE cmd: 1 failed: -110
```

**mtrace**
```
[   90.929671] <inf> dai_intel_dmic: dai_dmic_update_bits base 10000, reg 2000, mask 10000, value 0
[   90.929678] <inf> dai_intel_dmic: dmic_start(), cic 0x00000001
[   90.929686] <err> os:  ** FATAL EXCEPTION
[   90.929693] <err> os:  ** CPU 0 EXCCAUSE 13 (load/store PIF data error)
[   90.929700] <err> os:  **  PC 0xbe01a234 VADDR 0x4
[   90.929705] <err> os:  **  PS 0x60420
[   90.929715] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:4 CALLINC:2)
[   90.929723] <err> os:  **  A0 0xbe0172bb  SP 0xbe08ea90  A2 0xbe072ed8  A3 (nil)
[   90.929730] <err> os:  **  A4 0x18c0  A5 0x9e083378  A6 0x9e0906bc  A7 (nil)
[   90.929736] <err> os:  **  A8 0xbe0155d2  A9 0xbe0874d0 A10 0x9e0906a0 A11 0x18c0
[   90.929745] <err> os:  ** A12 (nil) A13 0xbe087490 A14 0xbe0874a0 A15 0xbe066bb8
[   90.929751] <err> os:  ** LBEG 0xbe01a648 LEND 0xbe01a65c LCOUNT 0xbe01a8b4
[   90.929756] <err> os:  ** SAR 0x18

Backtrace:0xbe01a231:0xbe08ea90 0xbe0172b8:0xbe08eac0 0xbe016a32:0xbe08eae0 0x7e043eaf:0xbe08eb00  |<-CORRUPTED

[   90.929805] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[   90.929811] <err> os: Current thread: 0x9e0922f8 (unknown)
[   90.932048] <err> zephyr: Halting system
[   90.932105] <err> os:  ** FATAL EXCEPTION
[   90.932158] <err> os:  ** CPU 0 EXCCAUSE 13 (load/store PIF data error)
[   90.932226] <err> os:  **  PC 0xbe01a234 VADDR 0x4
[   90.932286] <err> os:  **  PS 0x60820
[   90.932343] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:8 CALLINC:2)
[   90.932426] <err> os:  **  A0 0xbe0172bb  SP 0xbe08ea90  A2 0xbe072ed8  A3 (nil)
[   90.932511] <err> os:  **  A4 0x60e25  A5 0x9e083378  A6 (nil)  A7 0x17
[   90.932586] <err> os:  **  A8 0xbe0168b3  A9 0xbe08e958 A10 0xbe08d8b8 A11 (nil)
[   90.932678] <err> os:  ** A12 0x9e091cbc A13 0x1 A14 0xbe08ea18 A15 0x60e27
[   90.932758] <err> os:  ** LBEG 0xbe041186 LEND 0xbe041195 LCOUNT 0xbe0427bb
[   90.932835] <err> os:  ** SAR 0x18

Backtrace:0xbe01a231:0xbe08ea90 0xbe0172b8:0xbe08eac0 0xbe016a32:0xbe08eae0 0x7e043eaf:0xbe08eb00  |<-CORRUPTED

[   90.933121] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[   90.933186] <err> os: Current thread: 0x9e0917c0 (unknown)
[   90.933250] <err> zephyr: Halting system
[   90.933308] <err> os:  ** FATAL EXCEPTION
[   90.933361] <err> os:  ** CPU 0 EXCCAUSE 13 (load/store PIF data error)
[   90.933430] <err> os:  **  PC 0xbe01a234 VADDR 0x4
[   90.933490] <err> os:  **  PS 0x60a20
[   90.933543] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:10 CALLINC:2)
[   90.933633] <err> os:  **  A0 0xbe0172bb  SP 0Terminated
```

**To Reproduce**
~/sof-test/test-case/check-capture.sh -d 1 -l 100 -r 1

**Reproduction Rate**
TBD

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {[2776754a4cd9](https://github.com/thesofproject/linux/commit/2776754a4cd984a390a17a0c96635beb2ad408d8)}
    * SOF: {stable-v2.6/[d03ebade1f64](https://github.com/thesofproject/sof/commit/d03ebade1f64e0ae1847826fcfb4bfe64f3bd762)}
2) Name of the topology file
    * Topology: {avs-tplg/sof-hda-generic-2ch.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {TGLH_SKU0A70_HDA-cavs2_5-ipc4}

[dmesg.txt](https://github.com/thesofproject/sof/files/11785906/dmesg.txt)
[mtrace.txt](https://github.com/thesofproject/sof/files/11785909/mtrace.txt)

kv2019i, 2023-06-20:
Most probably this is https://github.com/zephyrproject-rtos/zephyr/pull/59416 . There can still be a a dmic error, but the FW should not panic.

mengdonglin, 2023-08-16:
Closing as not observed this on main branch in recent daily test.

closed 2023-08-16 by mengdonglin