thesofproject/sof #7866: [BUG][MTL] S0ix long run failure, Sof errors in dmesg logs

RDharageswari, 2023-06-27:
We see this error reported in multiple scenario:
One was long run S0ix testing.

[  352.971033] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x47000000|0x0
[  352.979537] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[  352.988991] sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0xc7000000|0x0|0x0, target: 0x67000000|0x0|0x0, ctl: 0x3
[  353.001735] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[  353.010979] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  353.020423] sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[  353.026868] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[  353.035552] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0xffffffff, ROM error: 0xffffffff
[  353.045287] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
[  353.055497] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit not enabled
[  353.063477] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[  353.072721] sof-audio-pci-intel-mtl 0000:00:1f.3: ctx_save IPC error: -110, proceeding with suspend
[  689.764439] sof-audio-pci-intel-mtl 0000:00:1f.3: GAIN (UUID: 61BCA9A8-18D0-4A18-8E7B-2639219804B7): No CPC match in the firmware file's manifest (ibs/obs: 768/768)
[  692.922594] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x47000000|0x0
[  692.931113] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[  692.940571] sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0xc7000000|0x0|0x0, target: 0x67000000|0x0|0x0, ctl: 0x3
[  692.953325] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[  692.962576] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]---------

RDharageswari, 2023-06-29:
Hi Marc,
Sure.I am trying to get the exact scenario, hence did not update it righ.Will update the entire configuration once available

mengdonglin, 2023-07-27:
@RDharageswari Can you still reproduce this issue with mtl-005.0.3 (Hot Fix 3) release?

RDharageswari, 2023-08-14:
@mengdonglin: No reports of this in the recent days..But @yongzhi1  was able to repro this issue in the PnP set up with mtl-005-hotfix3 release

macchian, 2023-08-16:
@mengdonglin , I also found the ipc time out during suspend resume test on customer MTL board.
The issue can be reproduced on fw branch mtl-005-drop-stable and mtl-005-hotfix3 as well.

@yongzhi1 , I saw that your linux PR about page fault
ASoC: SOF: Intel: MTL: catch invalid IRQ IP ptr #4344
Is it relative to this?

...
<3>[  124.809909] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x47000000|0x0
<3>[  124.818406] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
<3>[  124.830259] sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0xc7000000|0x0|0x0, target: 0xe7000000|0x0|0x0, ctl: 0x3
<3>[  124.856211] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
<3>[  124.874460] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
<3>[  124.893443] sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
<3>[  124.906717] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
<3>[  124.924700] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0xffffffff, ROM error: 0xffffffff
<3>[  124.944330] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
<3>[  124.966038] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit not enabled
<3>[  124.982946] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
<4>[  125.001601] sof-audio-pci-intel-mtl 0000:00:1f.3: ctx_save IPC error: -110, proceeding with suspend
<1>[  125.199974] BUG: unable to handle page fault for address: ffffbffbcb40001f
<1>[  125.207668] #PF: supervisor read access in kernel mode

macchian, 2023-08-16:
attach the crash logs.
[kernel.20230815.094120.64594.txt](https://github.com/thesofproject/sof/files/12357003/kernel.20230815.094120.64594.txt)

mengdonglin, 2023-08-16:
@macchian @RDharageswari This issue maybe same as #7990 and #8028 that happens with mtl-005.0.3 when CPC-based clock selection and IMR context save are both enabled.

Now for main branch, IMR context save is already enabled  and PR of CPC-based clock selection is under review https://github.com/thesofproject/sof/pull/8019

We found on main branch, with both IMR context save and CPC-based clock selection (PR) enabled, we cannot reproduce #7990 and #8028: https://github.com/thesofproject/sof/issues/7990#issuecomment-1676755248

Maybe you can try this recipe of main branch as well?

mengdonglin, 2023-08-16:
> @macchian @RDharageswari @yongzhi1 This issue seems similar to #7990 and #8028 that happens with mtl-005.0.3 when CPC-based clock selection and IMR context save are both enabled.
> 
> Now for main branch, IMR context save is already enabled and PR of CPC-based clock selection is under review #8019
> 
> We found on main branch, with both IMR context save and CPC-based clock selection (PR) enabled, we cannot reproduce #7990 and #8028. Please find details and recipe in [#7990 (comment)](https://github.com/thesofproject/sof/issues/7990#issuecomment-1676755248)
> 
> Maybe you can try this recipe of main branch as well?

macchian, 2023-08-18:
> We found on main branch, with both IMR context save and CPC-based clock selection (PR) enabled, we cannot reproduce #7990 and #8028. Please find details and recipe in [#7990 (comment)](https://github.com/thesofproject/sof/issues/7990#issuecomment-1676755248)
> 
> Maybe you can try this FW recipe of main branch as well?
@mengdonglin, thanks for your suggestions.
Yes, I provided a test mtl fw with PRs #8019 on mtl-005-hotfix3 branch.
So far the result is positive! The suspend test has run 600+ times until now.
Afterwards ODM will run more DUTs when other devices are available.

> Or you may try this kernel test PR #8007 with MTL-005.0.3 FW? This kernel PR will make FW always run at highest clock by always sending ZERO CPC values to FW?
Just a thought, Is #8007 a desired fix by always sending ZERO CPC values to FW ?

lrudyX, 2023-08-18:
@macchian Could you provide some more information how to reproduce the issue?

macchian, 2023-08-22:
> @macchian Could you provide some more information how to reproduce the issue?

@lrudyX , the suspend_stress_test tool is specific ChromeOS test tool. I am not sure if the CI relevant check-suspend-resume-with-* can isolate the issue or not. 

#suspend_stress_test -c 1500 --record_dmesg_dir=/usr/local/agingLogs --suspend_min=15 --suspend_max=20"

macchian, 2023-08-22:
> We found on main branch, with both IMR context save and CPC-based clock selection (PR) enabled, we cannot reproduce #7990 and #8028. Please find details and recipe in [#7990 [...]
@mengdonglin, from latest report, one DUT reproduce it highly reproducible rate. Attach the logs.

macchian, 2023-08-22:
> @macchian Thank you for testing the main branch! @abonislawski confirmed that main branch will always run at highest clock as it lack rimage update for CPC values as mtl-005.0.3. Can we say that except for the specific DUT, the reproduce rate of this issue is lower with main branch FW than mtl-005.0.3? And what's the approximate reproduction rate for this specific DUT?

@mengdonglin , I'm afraid that 2 DUTs were randomly failure within 20 cycles from suspend_resume. The 3rd DUT will almost certainly appear. This problem can be easily repeated from customer reports.

Do you recommend the kernel test PR #8007 worth to tr yor other PRs recommendation?

mengdonglin, 2023-08-29:
> @mengdonglin , I'm afraid that 2 DUTs were randomly failure within 20 cycles from suspend_resume. The 3rd DUT will almost certainly appear. This problem can be easily repeated from customer reports.
> 
> Do you recommend the kernel test PR #8007 worth to tr yor other PRs recommendation?
@macchian  You needn't try PR #8007 with mtl-005.0.3. Because you've already tried with main branch FW that DSP runs at the highest clock, and the issue can still be reproduced. PR #8007 just make DSP runs at highest clock with mtl-005.0.3

kv2019i, 2023-08-30:
@macchian @mengdonglin It is worth testing with https://github.com/thesofproject/sof/pull/7994 (disables IMR context save)

With the context-save, e.g. a leak in DSP resources will be compounding and lead to errors. So if 7994 has impact to repro rate, this can reveal a lot more information about the bug and help in debug.

macchian, 2023-08-31:
@

> @macchian @mengdonglin It is worth testing with #7994 (disables IMR context save)
> 
> With the context-save, e.g. a leak in DSP resources will be compounding and lead to errors. So if 7994 has impact to repro rate, this can reveal a lot more information about the bug and help in debug.

@kv2019i , I setup one device and it's very high reproduce rate no matter enabled or disabled IMR. Almost every time during 20 cycles, the DSP dump reproduce. I could share the remote access to you if you need something to check.

[  131.013601] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  131.023065] sof-audio-pci-intel-mtl 0000:00:1f.3: Firmware boot failure due to timeout
[  131.031919] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_IN_PROGRESS (3)
[  131.040875] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0xffffffff, ROM error: 0xffffffff
[  131.050595] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0xd000001c, ROM debug error: 0x2328
[  131.061093] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit not enabled
[  131.066218] pcieport 0000:00:06.1: pciehp: Slot(8): No link
[  131.069065] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------

macchian, 2023-08-31:
[dsp_dump.log](https://github.com/thesofproject/sof/files/12485713/dsp_dump.log)

kv2019i, 2023-08-31:
Thanks @macchian this now looks a bit different. The DSP fails to boot (versus a IPC timeout before) and there's a ROM debug code:

```` 
[  131.023065] sof-audio-pci-intel-mtl 0000:00:1f.3: Firmware boot failure due to timeout
[  131.031919] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_IN_PROGRESS (3)
[  131.040875] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0xffffffff, ROM error: 0xffffffff
[  131.050595] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0xd000001c, ROM debug error: 0x2328
```

mengdonglin, 2023-09-05:
@tmleman Can you help check this failure?

tmleman, 2023-09-05:
> @tmleman Can you help check this failure?

Ok.

macchian, 2023-09-13:
> > @tmleman Can you help check this failure?
> 
> Ok.

Update the mtl-006 fw test result: fails

>Can you please check if you can reproduce this issue with new mtl-006 FW release   MTL-006 2023WW36?
  Reproduced. Updating to mtl-006 fw, fw boot failure still occurs in suspend_resume 20 cycles.

>And I guess it’s not specific for C0 stepping silicon. [@Chiang, Mac](mailto:mac.chiang@intel.com) Can you help comment if you observed this FW boot failure with B-stepping or C-stepping Si?
B stepping Silicon. Currently reported failure devices are all B stepping.

[  206.492064] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  206.501314] cros-usbpd-charger cros-usbpd-charger.2.auto: PM: calling platform_pm_resume+0x0/0x43 @ 10277, parent: cros-ec-dev.1.auto
[  206.501394] cros-usbpd-charger cros-usbpd-charger.2.auto: PM: platform_pm_resume+0x0/0x43 returned 0 after 77 usecs
[  206.510751] sof-audio-pci-intel-mtl 0000:00:1f.3: Firmware boot failure due to timeout
[  206.524167] cros-usbpd-logger cros-usbpd-logger.3.auto: PM: calling platform_pm_resume+0x0/0x43 @ 10277, parent: cros-ec-dev.1.auto
[  206.535827] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_IN_PROGRESS (3)
[  206.535838] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0xffffffff, ROM error: 0xffffffff
[  206.544681] cros-usbpd-logger cros-usbpd-logger.3.auto: PM: platform_pm_resume+0x0/0x43 returned 0 after 1 usecs
[  206.557896] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0xd000001c, ROM debug error: 0x2328
[  206.566848] cros-ec-chardev cros-ec-chardev.4.auto: PM: calling platform_pm_resume+0x0/0x43 @ 10277, parent: cros-ec-dev.1.auto
[  206.576562] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit not enabled
[  206.587932] cros-ec-chardev cros-ec-chardev.4.auto: PM: platform_pm_resume+0x0/0x43 returned 0 after 0 usecs
[  206.598427] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[  206.598429] sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed to boot DSP firmware after resume -5
[  206.611261] cros-ec-debugfs cros-ec-debugfs.5.auto: PM: calling platform_pm_resume+0x0/0x43 @ 10277, parent: cros-ec-dev.1.auto
[  206.619217] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state change: 3 -> 4
[  206.630190] cros-ec-debugfs cros-ec-debugfs.5.auto: PM: platform_pm_resume+0x0/0x43 returned 0 after 2 usecs
[  206.639422] sof-audio-pci-intel-mtl 0000:00:1f.3: PM: dpm_run_callback(): pci_pm_resume+0x0/0xef returns -5
[  206.639425] sof-audio-pci-intel-mtl 0000:00:1f.3: PM: pci_pm_resume+0x0/0xef returned -5 after 2304330 usecs
[  206.639427] sof-audio-pci-intel-mtl 0000:00:1f.3: PM: failed to resume async: error -5

macchian, 2023-09-18:
@mengdonglin  @tmleman , reproduce on cav.2.7.

<7>[  250.447847] PM: resume of devices complete after 562.635 msecs
<6>[  251.109100] pcieport 0000:00:06.1: pciehp: Slot(8): No link
<4>[  252.779716] sched: RT throttling activated
<7>[  261.397621] PM: Finishing wakeup.
<6>[  261.397635] OOM killer enabled.
<6>[  261.401175] Restarting tasks ... done.
<5>[  261.408596] random: crng reseeded on system resumption
<12>[  261.409527] init: cupsd main process ended, respawning
<6>[  261.425122] Resume caused by IRQ 9, acpi
<6>[  261.430721] PM: suspend exit
<3>[  261.905049] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x47000000|0x0
<3>[  261.913530] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
<3>[  261.922976] sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0xc7000000|0x0|0x0, target: 0xe7000000|0x0|0x0, ctl: 0x3
<3>[  261.935717] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
<3>[  261.944958] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
<3>[  261.954395] sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
<3>[  261.960818] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
<3>[  261.969484] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0xffffffff, ROM error: 0xffffffff
<3>[  261.979201] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
<3>[  261.989419] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit not enabled
<3>[  261.997397] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
<4>[  262.006640] sof-audio-pci-intel-mtl 0000:00:1f.3: ctx_save IPC error: -110, proceeding with suspend
<1>[  262.925365] BUG: unable to handle page fault for address: ffff9e99ca80001f
<1>[  262.933064] #PF: supervisor read access in kernel mode
<1>[  262.938810] #PF: error_code(0x0000) - not-present page
<6>[  262.944554] PGD 100000067 P4D 100000067 PUD 10020f067 PMD 166e89067 PTE 0
<4>[  262.952149] Oops: 0000 [#1] PREEMPT SMP NOPTI
<4>[  262.957014] CPU: 8 PID: 4262 Comm: irq/196-AudioDS Tainted: G     U  W          6.1.52 #79 (HASH:050a 1)
<4>[  262.970231] Hardware name: Google Screebo4ES/Screebo4ES, BIOS Google_Screebo4ES.15612.0.0 09/12/2023
<4>[  262.980429] RIP: 0010:mtl_dsp_check_ipc_irq+0x8e/0xef [snd_sof_intel_hda_common]
<4>[  262.988706] Code: 48 8b 49 20 48 8b 89 98 00 00 00 4c 8b 59 58 48 8d 34 10 48 83 c6 20 4d 85 db 74 0e 48 89 df 41 ff d3 0f 1f 00 41 89 c6 eb 03 <44> 8b 36 66 90 41 83 fe ff 0f 95 c0 41 20 c6 44 89 f0 5b 41 5e 5d
<4>[  263.009693] RSP: 0018:ffff9e99c96afe28 EFLAGS: 00010246

tmleman, 2023-09-18:
> ROM debug status: 0xd000001c, ROM debug error: 0x2328

Status and error are from bootloader.

**_status_** 0x1C : Remove access control (this operation completed successfully),
**_error_** 0x2328: **UNSUPPORTED_BOOT_TYPE**

tmleman, 2023-09-18:
@macchian do I understand correctly that on main branch you are getting ipc timeout and rom error is from 006?

macchian, 2023-09-19:
> @macchian do I understand correctly that on main branch you are getting ipc timeout and rom error is from 006?

Actually, it starts from mtl-005. And reproduced on mtl-006 and cav2.7 as well.

tmleman, 2023-09-19:
> Actually, it starts from mtl-005. And reproduced on mtl-006 and cav2.7 as well.

cav2.7 is release for TGL from main branch?

macchian, 2023-09-19:
> > Actually, it starts from mtl-005. And reproduced on mtl-006 and cav2.7 as well.
> 
> cav2.7 is release for TGL from main branch?

From my understanding, cav2.5 and above release for ADL/RPL/MTL. @mengdonglin , could you comment it ?

kv2019i, 2023-09-19:
@tmleman I think @macchian meant SOF2.7 release on MTL hardware.

kv2019i, 2023-09-19:
@macchian @tmleman  One more test to make is to disable boot from IMR completely. This will cause the FW authentication to be repeated on each DSP boot.

This can be forced with following modprobe.d option (no kernel or FW code change needed):
```
options snd_sof sof_debug=128
```

In dmesg, "Attempting iteration %d of Core En/ROM load..." should be seen on every DSP boot when this option is set and dmesg should NOT have "IMR restore supported, booting from IMR directy" logs (if the setting is correctly applied).

This is not a fix as this will slow down the resume/boot process, but if this helps to avoid the problem, this will help to    limit the search space for debug.

macchian, 2023-09-21:
> This can be forced with following modprobe.d option (no kernel or FW code change needed):
> 
> ```
> options snd_sof sof_debug=128
> ```
Still reproduce when enabled the sof_debug=0x80. 

> Attempting iteration %d of Core En/ROM load..." should be seen on every DSP boot when this option is set 
@kv2019i , one thing need to confirm, we don't see the logs you mentioned in every DSP boot in dmesg, could you confirm what is missed?

[ 1611.353547] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x47000000|0x0
[ 1611.362027] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 1611.371472] sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0xc7000000|0x0|0x0, target: 0xe7000000|0x0|0x0, ctl: 0x3
[ 1611.384202] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 1611.393443] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 1611.402877] sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[ 1611.409300] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[ 1611.417965] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0xffffffff, ROM error: 0xffffffff
[ 1611.427688] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
[ 1611.437901] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit not enabled
[ 1611.445877] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 1611.455119] sof-audio-pci-intel-mtl 0000:00:1f.3: ctx_save IPC error: -110, proceeding with suspend
[ 1611.909000] pcieport 0000:00:06.1: pciehp: Slot(8): Card present
[ 1611.915775] BUG: unable to handle page fault for address: ffffa2928b40001f
[ 1611.923463] #PF: supervisor read access in kernel mode
[ 1611.929208] #PF: error_code(0x0000) - not-present page
[ 1611.934952] PGD 100000067 P4D 100000067 PUD 10020f067 PMD 1225f9067 PTE 0
[ 1611.942543] Oops: 0000 [#1] PREEMPT SMP NOPTI
[ 1611.947410] CPU: 8 PID: 4287 Comm: irq/196-AudioDS Tainted: G     U  W          6.1.53-06489-gd40af3977b31 #1 9eced5f266efbeb8771d81411bc77de00d07be96
[ 1611.962479] Hardware name: Google Screebo4ES/Screebo4ES, BIOS Google_Screebo4ES.1

kv2019i, 2023-09-21:
@macchian wrote:
>>  Attempting iteration %d of Core En/ROM load..." should be seen on every DSP boot when this option is set
> @kv2019i , one thing need to confirm, we don't see the logs you mentioned in every DSP boot in dmesg, could you confirm what is missed or it's expected behavior not printed?

Aa, right, you need to have dynamic debugs enabled in the build so that you see the debug prints from SOF driver. With IMR boot, this message should come once per system boot, with  sof_debug=0x80, it should come on every DSP boot (i.e. every system suspend-resume).

kv2019i, 2023-09-21:
There are two patterns visible here:
- on some cases, after system resumes, DSP has apparently booted we have a IPC timeout on SET_DX IPC (0x47000000|0x0: MOD_SET_DX)
      - afterwards there's a kernel panic in MTL DSP irq handler (we get interrupts when DSP should be in D3!?!), @bardliao @ranj063 @ujfalusi can you check the kernel logs?
      - example: https://github.com/thesofproject/sof/files/12404629/kernel.20230815.205835.87076.0.txt
- on some cases, there's DSP ROM error already at boot
      - example: https://github.com/thesofproject/sof/issues/7866#issuecomment-1717032210

ujfalusi, 2023-09-21:
@kv2019i, the kernel must not crash, so that is bad.

ujfalusi, 2023-09-21:
@kv2019i the SET_DX timeout and kernel crash: the timeout is the SET_DX to power off the DSP.
The suspend happens (while we wait for the reply?) and when the device resumes we got the suspend timeout and continue to pm runtime off, then when the resume is done, irqs enabled and we got a ghost or delayed irq, the DSP is off and we try to read a register.

Or something like this, some race messes up the sequence and the hardware is in a weird state.

macchian, 2023-09-22:
@yongzhi1 , do you think the irq receive before dsp power up? 
Should I check the reg state and see if it happens from your test PR?

+	/* Check GIE bit before proceeding */
+	if (!(snd_sof_dsp_read(sdev, HDA_DSP_HDA_BAR, SOF_HDA_INTCTL) &
+	    SOF_HDA_INT_GLOBAL_EN)) {
+		pr_warn_once("%s: spurious interrupt\n", __func__);
+		return IRQ_NONE;
+	}

ujfalusi, 2023-09-22:
> @yongzhi1 , do you think the irq receive before dsp power up? Should I check the reg state and see if it happens from your test PR?
>
> ```
> /* Check GIE bit before proceeding */
> if [...]

The DSP is off, you cannot read registers, it will cause the same page fault error.
However we could check (for testing) in the IRQ handler the pm runtime state and if it is off then we print an error and ignore the irq. That will show up in logs without the kernel crash (or we can let the kernel just crash but keep the print)

plbossart, 2023-09-22:
Can we instead try to improve the error handling of the IPC timeout during suspend? Checking the pm state while resuming doesn't seem right to me.

ujfalusi, 2023-09-22:
> Can we instead try to improve the error handling of the IPC timeout during suspend? Checking the pm state while resuming doesn't seem right to me.

I'm not sure what to improve or do differently. If the SET_DX time out for some reason we power down the DSP anyways. The timeout means that the firmware likely failed and if we did not get the exception notification then there is not much we could possibly do.
If we block the suspend then we will have a frozen DSP which is not answering.

There is something else going on here which makes the DSP to not respond and there is certainly something odd going on that the system resumes before the suspend callback finishes, or appear to do so.

plbossart, 2023-09-22:
Maybe disable the interrupts on the suspend path?

plbossart, 2023-09-22:
I've seen this before btw, if there is a workqueue that is scheduled before the suspend, it might be scheduled while the resume happens.

yongzhi1, 2023-09-22:
> @yongzhi1 , do you think the irq receive before dsp power up? Should I check the reg state and see if it happens from your test PR?
> 

This patch helps to prevent the panic in the rare case that hfintipptr becomes invalid, this hfintipptr + MTL_DSP_IRQSTS read triggers page fault.

macchian, 2023-09-25:
> The DSP is off, you cannot read registers, it will cause the same page fault error. However we could check (for testing) in the IRQ handler the pm runtime state and if it is off then we print an [...]

> This patch helps to prevent the panic in the rare case that hfintipptr becomes invalid, thus hfintipptr + MTL_DSP_IRQSTS read triggers page fault. I only saw 3 instances so far with ur log included.

@yongzhi1 ,I tested #4344 patch but not catch the invalid interrupt. The IRQ_WAKE_THREAD triggers last time before ipc timed out.

....
[  584.988471] sof-audio-pci-intel-mtl 0000:00:1f.3: PM: calling pci_pm_resume+0x0/0xef @ 6856, parent: pci0000:00
[  585.000149] sof-audio-pci-intel-mtl 0000:00:1f.3: current DSP power state: 0
...
[  596.188718] sof-audio-pci-intel-mtl 0000:00:1f.3: hda_dsp_interrupt_handler: IRQ_WAKE_THREAD
[  596.190192] random: crng reseeded on system resumption
[  596.206167] init: cupsd main process ended, respawning
[  596.212355] thermal thermal_zone9: failed to read out thermal zone (-61)
[  596.220282] Resume caused by IRQ 9, acpi
[  596.224756] PM: suspend exit
[  596.270045] init: machine-id-regen-network main process (22841) terminated with status 1
[  596.280851] init: shill-event main process (22814) terminated with status 1
[  596.695103] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x47000000|0x0
...

The same page fault behavior with https://github.com/thesofproject/linux/pull/4344#discussion_r1190565854 (@RanderWang)

Acked, the latest dsp set state is in D0 power state, not dsp being off. D0/D3 suspend/resume here, without runtime (D0ix).

ujfalusi, 2023-09-25:
@macchian, I'm not following your comment, can you clarify it?
The 4344 is the kernel PR, right?
With that applied you also get the page fault?
No "spurious interrupt" print seen in the kernel log?

macchian, 2023-09-25:
attach the full logs.
[console-ramoops-0_0925.txt](https://github.com/thesofproject/sof/files/12712357/console-ramoops-0_0925.txt)

ujfalusi, 2023-09-25:
> attachment the full logs.
> 
> > @macchian, I'm not following your comment, can you clarify it? The 4344 is the kernel PR, right? With that applied you also get the page fault? No "spurious interrupt" print seen in the kernel log?
> 
> @ujfalusi The 4344 is the kernel PR, right? yes.
> 
> With that applied you also get the page fault? yes

As I have commented here and under 4344, it is expected

> No "spurious interrupt" print seen in the kernel log? yes

Since the page fault happens before we would be printing that.

ujfalusi, 2023-09-25:
@macchian, from where this is printed:
```
current DSP power state:
```

macchian, 2023-09-25:
--- a/sound/soc/sof/intel/hda-dsp.c
+++ b/sound/soc/sof/intel/hda-dsp.c
@@ -565,6 +565,9 @@ static void hda_dsp_state_log(struct snd_sof_dev *sdev)
                        sdev->dsp_power_state.state);
                break;
        }
+       dev_err(sdev->dev, "current DSP power state: %d\n",
+                       sdev->dsp_power_state.state);
+
 }

ujfalusi, 2023-09-25:
@macchian, can you add this print and try to reproduce:
```diff
diff --git a/sound/soc/sof/pm.c b/sound/soc/sof/pm.c
index 704b21413c71..a6254c4d307d 100644
--- a/sound/soc/sof/pm.c
+++ b/sound/soc/sof/pm.c
@@ -251,6 +251,7 @@ static int sof_suspend(struct device *dev, bool runtime_suspend)
 #endif
 	/* notify DSP of upcoming power down */
 	if (pm_ops && pm_ops->ctx_save) {
+		dev_warn(sdev->dev, "calling ctx_save\n");
 		ret = pm_ops->ctx_save(sdev);
 		if (ret == -EBUSY || ret == -EAGAIN) {
 			/*
```

macchian, 2023-09-25:
> @macchian, can you add this print and try to reproduce:
>
> ```diff
> diff --git a/sound/soc/sof/pm.c b/sound/soc/sof/pm.c
> index 704b21413c71..a6254c4d307d 100644
> --- a/sound/soc/sof/pm.c
> +++ [...]

ok, see attachment. And enabled more logs.
[dmesg-ramoops-0.txt](https://github.com/thesofproject/sof/files/12714465/dmesg-ramoops-0.txt)

ujfalusi, 2023-09-25:
@macchian, what is the way to reproduce this reliably?

ujfalusi, 2023-09-25:
@macchian, can you use this diff and try to reproduce it again?
```diff
diff --git a/sound/soc/sof/pm.c b/sound/soc/sof/pm.c
index 704b21413c71..6c4e9244076c 100644
--- a/sound/soc/sof/pm.c
+++ b/sound/soc/sof/pm.c
@@ -251,6 +251,7 @@ static int sof_suspend(struct device *dev, bool runtime_suspend)
 #endif
 	/* notify DSP of upcoming power down */
 	if (pm_ops && pm_ops->ctx_save) {
+		dev_warn(dev, "[TRACKING] %s: calling ctx_save\n", __func__);
 		ret = pm_ops->ctx_save(sdev);
 		if (ret == -EBUSY || ret == -EAGAIN) {
 			/*
@@ -306,7 +307,13 @@ int snd_sof_dsp_power_down_notify(struct snd_sof_dev *sdev)
 
 int snd_sof_runtime_suspend(struct device *dev)
 {
-	return sof_suspend(dev, true);
+	int ret;
+
+	dev_warn(dev, "[TRACKING] %s: ENTER\n", __func__);
+	ret = sof_suspend(dev, true);
+	dev_warn(dev, "[TRACKING] %s: LEAVE\n", __func__);
+
+	return ret;
 }
 EXPORT_SYMBOL(snd_sof_runtime_suspend);
 
@@ -320,19 +327,37 @@ EXPORT_SYMBOL(snd_sof_runtime_idle);
 
 int snd_sof_runtime_resume(struct device *dev)
 {
-	return sof_resume(dev, true);
+	int ret;
+
+	dev_warn(dev, "[TRACKING] %s: ENTER\n", __func__);
+	ret = sof_resume(dev, true);
+	dev_warn(dev, "[TRACKING] %s: LEAVE\n", __func__);
+
+	return ret;
 }
 EXPORT_SYMBOL(snd_sof_runtime_resume);
 
 int snd_sof_resume(struct device *dev)
 {
-	return sof_resume(dev, false);
+	int ret;
+
+	dev_warn(dev, "[TRACKING] %s: ENTER\n", __func__);
+	ret = sof_resume(dev, false);
+	dev_warn(dev, "[TRACKING] %s: LEAVE\n", __func__);
+
+	return ret;
 }
 EXPORT_SYMBOL(snd_sof_resume);
 
 int snd_sof_suspend(struct device *dev)
 {
-	return sof_suspend(dev, false);
+	int ret;
+
+	dev_warn(dev, "[TRACKING] %s: ENTER\n", __func__);
+	ret = sof_suspend(dev, false);
+	dev_warn(dev, "[TRACKING] %s: LEAVE\n", __func__);
+
+	return ret;
 }
 EXPORT_SYMBOL(snd_sof_suspend);
```

I don't see how the suspend callback is called right after the system just resumed. All pm calls seams to be in pair, something else is off in the system.

ujfalusi, 2023-09-25:
It looks to me that something is blocking the irq thread to be executed and that is causing the timeout and the crash:
```
<7>[ 4343.611498] PM: Finishing wakeup.
<6>[ 4343.611504] OOM killer enabled.
<6>[ 4343.615040] Restarting tasks ... 
<4>[ 4343.615114] sof-audio-pci-intel-mtl 0000:00:1f.3: calling ctx_save
<4>[ 4343.620525] done.
<7>[ 4343.625747] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x47000000|0x0: MOD_SET_DX [data size: 8]
```
sending the SET_DX message
```
<5>[ 4343.627894] random: crng reseeded on system resumption
<7>[ 4343.634364] sof-audio-pci-intel-mtl 0000:00:1f.3: hda_dsp_interrupt_handler: IRQ_WAKE_THREAD
```
Interrupt received and we ask the irq thread to be run
```
<12>[ 4343.634812] init: cupsd main process ended, respawning
<4>[ 4343.640752] thermal thermal_zone9: failed to read out thermal zone (-61)
<6>[ 4343.648644] Resume caused by IRQ 9, acpi
<6>[ 4343.653122] PM: suspend exit
<12>[ 4343.743121] init: machine-id-regen-network main process (19793) terminated with status 1
<12>[ 4343.753466] init: shill-event main process (19787) terminated with status 1
<3>[ 4344.135591] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x47000000|0x0
<6>[ 4344.144072] sof-audio-pci-intel-mtl 0000:00:1f.3: Attempting to prevent DSP from entering D3 state to preserve context
<3>[ 4344.156032] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
<3>[ 4344.165475] sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0xc7000000|0x0|0x0, target: 0xe7000000|0x0|0x0, ctl: 0x3
<3>[ 4344.178215] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
<3>[ 4344.187454] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
<3>[ 4344.196897] sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
<3>[ 4344.203316] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
<3>[ 4344.211982] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0xffffffff, ROM error: 0xffffffff
<3>[ 4344.221706] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
<3>[ 4344.231926] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit not enabled
<3>[ 4344.239904] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
<4>[ 4344.249157] sof-audio-pci-intel-mtl 0000:00:1f.3: ctx_save IPC error: -110, proceeding with suspend
```
But it is not run in time, so we face with IOPC timeout and proceed to power off the DSP
```
<7>[ 4344.259280] sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x114c]=0x0 successful
<7>[ 4344.259288] sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x1800]=0x0 successful
<7>[ 4344.259292] sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x1140]=0x0 successful
<7>[ 4344.260433] sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x178d04]=0x2000000 successful
<7>[ 4344.261485] sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x1000]=0x0 successful
<7>[ 4344.261535] sof-audio-pci-intel-mtl 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
<7>[ 4344.261540] sof-audio-pci-intel-mtl 0000:00:1f.3: Debug PCIW: 00000010 at  00000044
<7>[ 4344.262531] sof-audio-pci-intel-mtl 0000:00:1f.3: Turning i915 HDAC power 0
<7>[ 4344.262535] sof-audio-pci-intel-mtl 0000:00:1f.3: Current DSP power state: D3
<7>[ 4344.262537] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state change: 7 -> 0
```
DSP is off now.
```
<1>[ 4345.118498] BUG: unable to handle page fault for address: ffffa8fa8b40001f
```
and the irq thread finally got scheduled to run.
```
<1>[ 4345.126200] #PF: supervisor read access in kernel mode
<1>[ 4345.131943] #PF: error_code(0x0000) - not-present page
<6>[ 4345.137686] PGD 100000067 P4D 100000067 PUD 100210067 PMD 160cad067 PTE 0
<4>[ 4345.145283] Oops: 0000 [#1] PREEMPT SMP NOPTI
<4>[ 4345.150152] CPU: 8 PID: 4078 Comm: irq/194-AudioDS Tainted: G     U  W          6.1.54 #3 41746e0722746447e72667ca3a0ba42f70ef1941
<4>[ 4345.163274] Hardware name: Google Screebo4ES/Screebo4ES, BIOS Google_Screebo4ES.15612.0.0 09/12/2023
<4>[ 4345.173480] RIP: 0010:mtl_dsp_check_ipc_irq+0x8e/0xef [snd_sof_intel_hda_common]
```
**But, there are several places in the log where this works, like this:**
```
<7>[ 4289.139438] PM: Finishing wakeup.
<6>[ 4289.139448] OOM killer enabled.
<7>[ 4289.139685] rt5682s i2c-RTL5682:00: jack_type = 0
<6>[ 4289.142968] Restarting tasks ... 
<4>[ 4289.143068] sof-audio-pci-intel-mtl 0000:00:1f.3: calling ctx_save
[... 6 lines cut ...]
<7>[ 4289.162363] sof-audio-pci-intel-mtl 0000:00:1f.3: hda_dsp_interrupt_handler: IRQ_WAKE_THREAD
```
Interrupt received and we ask the irq thread to be run
```
<7>[ 4289.162394] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x67000000|0x0: MOD_SET_DX
<12>[ 4289.162522] init: cupsd main process ended, respawning
<7>[ 4289.163086] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx done : 0x47000000|0x0: MOD_SET_DX [data size: 8]
```
The thread run and we handle the reply.
```
<4>[ 4289.168409] thermal thermal_zone9: failed to read out thermal zone (-61)
<7>[ 4289.169300] sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x114c]=0x0 successful
<7>[ 4289.176028] sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x1800]=0x0 successful
<7>[ 4289.176035] sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x1140]=0x0 successful
<6>[ 4289.176535] Resume caused by IRQ 9, acpi
[... cut ...]
<7>[ 4289.181978] sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x1000]=0x0 successful
<7>[ 4289.184209] sof-audio-pci-intel-mtl 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
<7>[ 4289.184215] sof-audio-pci-intel-mtl 0000:00:1f.3: Debug PCIW: 00000010 at  00000044
<7>[ 4289.184834] sof-audio-pci-intel-mtl 0000:00:1f.3: Turning i915 HDAC power 0
<7>[ 4289.184839] sof-audio-pci-intel-mtl 0000:00:1f.3: Current DSP power state: D3
<7>[ 4289.184841] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state change: 7 -> 0
```
DSP is off now.

macchian, 2023-09-26:
> ```diff
> -- a/sound/soc/sof/pm.c
> +++ b/sound/soc/sof/pm.c
> @@ -251,6 +251,7 @@ static int sof_suspend(struct device *dev, bool runtime_suspend)
>  #endif
>  	/* notify DSP of upcoming power down */
>  	if (pm_ops && pm_ops->ctx_save) {
> +		dev_warn(dev, "[TRACKING] %s: calling ctx_save\n", __func__);
>  		ret = pm_ops->ctx_save(sdev);
>  		if (ret == -EBUSY || ret == -EAGAIN) {
...
@ujfalusi  , @ranj063 , applied the logs in the attachment.
[dmesg.txt](https://github.com/thesofproject/sof/files/12722442/dmesg.txt)

mengdonglin, 2023-09-28:
@tmleman This FW boot failure seems related to IMR, because
- The boot failure was firstly observed with mtl-005, which enabled IMR context save;
- The boot failure cannot be reproduced with mtl-006 or sof v2.7, both of them disabled IMR context save. @macchian confirmed that with mtl-006 , non matter with or without sof_debug=0x80 (ADSP boot from IMR), not able to reproduce DSP boot failure "fw_state: SOF_FW_BOOT_IN_PROGRESS (3)".

tmleman, 2023-11-24:
While debugging I found one problem that may be causing this issue.

If FW receive SET_DX message (HOST->DSP) before receiving ACK for previously sent message/notification (DSP->HOST), IPC device is in busy state. Zephyr PM during system suspend skip suspension of busy devices. In result the IPC device is neither put to suspend nor resume during power-flow.

From the host's perspective, it manifests itself in the fact that the ROM reports FW is loaded, but we do not receive FW Ready. When you connect via gdb to core 0, FW is in idle state.

lgirdwood, 2023-11-24:
> While debugging I found one problem that may be causing this issue.
> 
> If FW receive SET_DX message (HOST->DSP) before receiving ACK for previously sent message/notification (DSP->HOST), IPC device is in busy state. Zephyr PM during system suspend skip suspension of busy devices. In result, the IPC device is neither put to suspend nor resume during power-flow.
> 
> From the host perspective, it manifests itself in the fact that the ROM reports FW is loaded, but we do not receive FW Ready. When you connect via gdb to core 0, FW is in idle state.

Good work @tmleman ! Do you have a proposal for a fix ?
@kv2019i @ujfalusi fyi

kv2019i, 2023-11-24:
Context save not enabled in stable-v2.8, moving this to v2.9 (this is gating enabling context-save in mainline).

lgirdwood, 2023-11-29:
@tmleman any update on debug - my feeling is same root cause could also cause https://github.com/thesofproject/sof/issues/8056 @kv2019i fyi.

tmleman, 2023-11-30:
@lgirdwood I haven't been able to reproduce this particular issue in my local environment. I need someone to confirm that the bug found is also the cause of this issue.

I also need to determine how FW should behave in such a situation. The easiest solution is to ignore this last ACK and force the IPC device to sleep while going into the D3 state. Another solution would be to return an error in the SET_DX response, and fix would need to be done in linux driver.

ranj063, 2023-12-04:
> If FW receive SET_DX message (HOST->DSP) before receiving ACK for previously sent message/notification (DSP->HOST), IPC device is in busy state. Zephyr PM during system suspend skip suspension of busy devices. In result, the IPC device is neither put to suspend nor resume during power-flow.

@tmleman I'm a bit skeptical about the possibility of this happening in the linux driver. Basically, if the DSP has send a message/notification to the host, we hold a spinlock until the ACK has been sent back to the DSP before initializing a new IPC.

tmleman, 2023-12-05:
@lgirdwood  I spoke with @mmaka1 about this and we agreed that this pending ACK should not be an obstacle in D3 transition. This is because the device is not actually suspended but reset and from a hardware perspective there is no trace of this ACK. I have pushed fix for this issue to review https://github.com/thesofproject/sof/pull/8573.

[9 more replies]

closed 2024-07-19 by wszypelt