thesofproject/sof #7885: [BUG][MTL][Multicore] IPC timed out for GLB_CREATE_PIPELINE

fredoh9, 2023-06-30:
**Describe the bug**

MTLP_RVP_NOCODEC_MULTICORE in daily test is broken.

```
[  491.426215] kernel: snd_sof:snd_sof_dsp_core_get: sof-audio-pci-intel-mtl 0000:00:1f.3: Core 2 powered up
[  491.426219] kernel: snd_sof:sof_ipc4_widget_setup: sof-audio-pci-intel-mtl 0000:00:1f.3: pipeline: 5 memory pages: 5
[  491.426223] kernel: snd_sof:sof_ipc4_widget_setup: sof-audio-pci-intel-mtl 0000:00:1f.3: Create widget pipeline.5 instance 0 - pipe 5 - core 2
[  491.426229] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x11000007|0x200000: GLB_CREATE_PIPELINE
[  491.931827] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x11000007|0x200000
[  491.931905] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Attempting to prevent DSP from entering D3 state to preserve context
[  491.931915] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[  491.931982] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0x91000007|0x200000|0x0, target: 0x1b060000|0x0|0x0, ctl: 0x3
[  491.932196] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[  491.932355] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  491.932459] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[  491.932536] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[  491.932635] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0x0, ROM error: 0x0
[  491.932669] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
```

**To Reproduce**
TPLG=/lib/firmware/intel/sof-ace-tplg/sof-mtl-nocodec-multicore.tplg MODEL=MTLP_RVP_NOCODEC_MULTICORE ~/sof-test/test-case/check-playback.sh -d 100 -l 1 -r 1

**Reproduction Rate**
100%

**Environment**
- Kernel Branch: topic/sof-dev
- Kernel Commit: [34b7f81747f8](https://github.com/thesofproject/linux/commit/34b7f81747f8)
- SOF Branch: main
- SOF Commit: [3e3d0cdeab7a](https://github.com/thesofproject/sof/commit/3e3d0cdeab7a)

**Screenshots or console output**

[mtl-nocodec-multicore-dmesg.txt](https://github.com/thesofproject/sof/files/11919497/mtl-nocodec-multicore-dmesg.txt)
[mtl-nocodec-multicore-mtrace.txt](https://github.com/thesofproject/sof/files/11919498/mtl-nocodec-multicore-mtrace.txt)

keqiaozhang, 2023-07-10:
A potential Zephyr fix:https://github.com/zephyrproject-rtos/zephyr/commit/431108d89e1750dbea05dce64dbb7fd329aa610e

keqiaozhang, 2023-07-10:
@abonislawski , can you please update the west.yaml to fix this issue?

closed 2023-07-12 by kv2019i