thesofproject/sof #7901: [FEATURE] use west instead of submodules.

lgirdwood, 2023-07-06:
Lets use west to manage all ingredients instead of git submodules for both Zephyr and xtos based configurations. This should initially target rimage and toml but can later be extended to topologies and any other ingredients.

marc-hb, 2023-07-12:
Some relevant plans for the Zephyr build:
- https://github.com/thesofproject/sof/commit/a9680388e55474fd8e20092d303a89d82cd4f6fe 
- https://github.com/thesofproject/sof/pull/6398#discussion_r993678758.

This ticket is for now two very different and mostly unrelated tickets in one.

While this is a massive, very useful and welcome simplification for the Zephyr build worth breaking everyone's build for a short time, changing it for the XTOS build is a lot of disruptive and non-backwards compatible build changes for no clear value yet. For Zephyr this is just moving the location of two git repos because west is already used. For XTOS it's a major re-architecture of the build system because it never used `west` at all.

Intel could do the XTOS build changes on behalf of other companies while not using XTOS itself but testing would be limited to Github Actions because no other company has so far connected their build, test and reports. So Intel would have to make the changes partially "blind", discovering what broke until possibly long after it's merged. In fact we're not even sure who exactly uses rimage+XTOS in the main branch right now.

Anyway the Zephyr and XTOS builds are 99% independent of each other so this can be done for Zephyr first and see about XTOS later.

marc-hb, 2023-10-26:
Done for rimage:
- #8178

Only tomlc99 is now a submodule. BUT it is also a west module, meaning git submodules are NOT needed by Zephyr developers.

rimage may still be used by non-zephyr users, so they would miss the tomlc99 submodule. Also, tomlc99 almost never changes so I would like to close this.

closed 2023-10-27 by marc-hb