thesofproject/sof #7914: [BUG] basic playback and capture tests failed with no bytes to copy in ADLP_SKU0B00_SDCA

fredoh9, 2023-07-10:
**Describe the bug**

ADLP_SKU0B00_SDCA_IPC4ZPH in daily test is broken.
```
2023-07-10 10:41:48 UTC [REMOTE_INFO] ===== Testing: (Round: 1/1) (PCM: Speaker [hw:0,2]) (Loop: 1/1) =====
2023-07-10 10:41:48 UTC [REMOTE_COMMAND] aplay   -Dhw:0,2 -r 48000 -c 2 -f S16_LE -d 100 /dev/zero -v -q
Hardware PCM card 0 'sof-soundwire' device 2 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 16384
  period_size  : 4096
  period_time  : 85333
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 4096
  period_event : 0
  start_threshold  : 16384
  stop_threshold   : 16384
  silence_threshold: 0
  silence_size : 0
  boundary     : 4611686018427387904
  appl_ptr     : 0
  hw_ptr       : 0
aplay: pcm_write:2127: write error: Input/output error
```
no error in dmesg, mtrace has full of no bytes to copy...
```
[   77.962581] <inf> host_comp: host_get_copy_bytes_normal: comp:0 0x40000 no bytes to copy, available samples: 384, free_samples: 0
[   77.963583] <inf> host_comp: host_get_copy_bytes_normal: comp:0 0x40000 no bytes to copy, available samples: 384, free_samples: 0
[   77.964583] <inf> host_comp: host_get_copy_bytes_normal: comp:0 0x40000 no bytes to copy, available samples: 384, free_samples: 0
[   77.965583] <inf> host_comp: host_get_copy_bytes_normal: comp:0 0x40000 no bytes to copy, available samples: 384, free_samples: 0
[   77.966581] <inf> host_comp: host_get_copy_bytes_normal: comp:0 0x40000 no bytes to copy, available samples: 384, free_samples: 0
[   77.967583] <inf> host_comp: host_get_copy_bytes_normal: comp:0 0x40000 no bytes to copy, available samples: 384, free_samples: 0
[   77.968581] <inf> host_comp: host_get_copy_bytes_normal: comp:0 0x40000 no bytes to copy, available samples: 384, free_samples: 0
[   77.969583] <inf> host_comp: host_get_copy_bytes_normal: comp:0 0x40000 no bytes to copy, available samples: 384, free_samples: 0
```

**To Reproduce**
TPLG=/lib/firmware/intel/avs-tplg/sof-adl-rt711-l0-rt1316-l12-rt714-l3.tplg MODEL=ADLP_SKU0B00_SDCA_IPC4ZPH SOF_TEST_INTERVAL=5 ~/sof-test/test-case/check-playback.sh -d 100 -l 1 -r 1

**Reproduction Rate**
Looks 100% in recent test

**Environment**

- Kernel Branch: topic/sof-dev
- Kernel Commit: [e2b8ebcd30b4](https://github.com/thesofproject/linux/commit/e2b8ebcd30b4)
- SOF Branch: main
- SOF Commit: [8937929b4ec9](https://github.com/thesofproject/sof/commit/8937929b4ec9)
- Zephyr Commit: [zephyr-v3.4.0-301-g23b3cae1b1d9](https://github.com/zephyrproject-rtos/zephyr/commit/zephyr-v3.4.0-301-g23b3cae1b1d9)

**Screenshots or console output**

[mtrace_ADLP_SKU0B00_SDCA_IPC4ZPH.txt](https://github.com/thesofproject/sof/files/12005087/mtrace_ADLP_SKU0B00_SDCA_IPC4ZPH.txt)

Intel internal daily test link:
planresultdetail/28790?model=ADLP_SKU0B00_SDCA_IPC4ZPH&testcase=check-playback-100sec

fredoh9, 2023-07-12:
For today daily test, only playback is broken, capture tests were passed.

- Kernel Branch: topic/sof-dev
- Kernel Commit: [0a80aee96685](https://github.com/thesofproject/linux/commit/0a80aee96685)
- SOF Branch: main
- SOF Commit: [70c257f9ae93](https://github.com/thesofproject/sof/commit/70c257f9ae93)
- Zephyr Commit: [zephyr-v3.4.0-496-g431108d89e17](https://github.com/zephyrproject-rtos/zephyr/commit/zephyr-v3.4.0-496-g431108d89e17)

Intel daily test: planresultdetail/28946?model=ADLP_SKU0B00_SDCA_IPC4ZPH&testcase=check-playback-100sec

lgirdwood, 2023-07-14:
@fredoh9 related to #7913 ? Both ADL and both SDW ? @kv2019i @bard any thoughts ?

mengdonglin, 2023-07-31:
@fredoh9 @keqiaozhang @lgirdwood  @bardliao @aiChaoSONG Maybe a duplicate of https://github.com/thesofproject/sof/issues/7979 caused by copier refactoring commit https://github.com/thesofproject/sof/commit/e424b874a477617d080f368ae67b46d19b7e6d4d and fixed by https://github.com/thesofproject/sof/pull/7987 by @ranj063 and @btian1

lgirdwood, 2023-07-31:
ok, pls mark as duplicate or close if fixed.

fredoh9, 2023-07-31:
@mengdonglin https://github.com/thesofproject/sof/pull/7987 in new daily test(plan=29598) is working fine after https://github.com/thesofproject/sof/pull/7987 merged

New SOF build SHA1 is,
79532b8dbc01 copier: fix no consume for multi endpoint copy

Can we wait for a few days to confirm this?

mengdonglin, 2023-08-01:
> Can we wait for a few days to confirm this?
@fredoh9  Okay.

This issue cannot be reproduce on main branch commit https://github.com/thesofproject/sof/commit/cfd786110ae7 on ww31.2 (daily report ID 29617)  The fix has been merged https://github.com/thesofproject/sof/commit/79532b8dbc0129b7755629c85c57d6bffc80a1a7