thesofproject/sof #7916: [BUG] DSP panic occurs when DSP boots up on MT8188 platform

wenliangwu, 2023-07-11:
**Describe the bug**

Upon upgrading my local SOF branch to the HEAD of main, a DSP panic occurred during boot. To identify the root cause, I utilized 'git bisect' and pinpointed that commit 3e3d0cdeab7aded526fea237290f38e471dfd30d was the first bad commit. @lyakh 
 
I am unsure why the commit is impacting the mt8188 platform. However, after several attempts, I discovered that replacing **dcache_invalidate_region** with **dcache_writeback_invalidate_region** at **_buffer_attach_** resolved the problem.
```c
/* no dirty cache lines exist for this buffer yet, no need to write back */
	dcache_invalidate_region(uncache_to_cache(list), sizeof(*list));
``` 

**To Reproduce**
Boot device and DSP panic occurs when loading DSP topology.

**Reproduction Rate**
100% 

**Expected behavior**
Boot DSP without any error.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: We are using chromium Kernel 6.1 branch.
    * SOF branch: main
    * SOF commit: 8adbb850448a7a6154cfa222e5cb44184416354c 
2) Name of the topology file
    * Topology: sof-mt8188.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: mt8188

**Screenshots or console output**
```
[  405.591231] sof-audio-of-mt8186 10b80000.adsp: snd_sof_dsp_panic: dsp_oops_offset 5253120 differs from panic offset 233492486
[  405.602729] sof-audio-of-mt8186 10b80000.adsp: ------------[ DSP dump start ]------------
[  405.611069] sof-audio-of-mt8186 10b80000.adsp: DSP panic!
[  405.616592] sof-audio-of-mt8186 10b80000.adsp: fw_state: SOF_FW_BOOT_COMPLETE (7)
[  405.624200] sof-audio-of-mt8186 10b80000.adsp: adsp dump : pc 0x4e118abe, data 0x4e118abe, dbg_inst 0x5002000,
[  405.634340] sof-audio-of-mt8186 10b80000.adsp: ls0stat 0x20, status 0x80004, faultinfo 0x0
[  405.642657] sof-audio-of-mt8186 10b80000.adsp: reason: runtime exception (0x6)
[  405.649940] sof-audio-of-mt8186 10b80000.adsp: trace point: 0x0dead006
[  405.656567] sof-audio-of-mt8186 10b80000.adsp: panic at :0
[  405.662129] sof-audio-of-mt8186 10b80000.adsp: error: DSP Firmware Oops
[  405.668826] sof-audio-of-mt8186 10b80000.adsp: error: Exception Cause: LoadProhibitedCause, A load referenced a page mapped with an attribute that does not permit loads
[  405.683961] sof-audio-of-mt8186 10b80000.adsp: EXCCAUSE 0x0000001c EXCVADDR 0xffffff15 PS       0x00060125 SAR     0x00000000
[  405.695377] sof-audio-of-mt8186 10b80000.adsp: EPC1     0x4e1064ef EPC2     0x4e1169f9 EPC3     0x4e11bf97 EPC4    0x00000000
[  405.706780] sof-audio-of-mt8186 10b80000.adsp: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[  405.718164] sof-audio-of-mt8186 10b80000.adsp: EPS2     0x00060f20 EPS3     0x00060920 EPS4     0x00000000 EPS5    0x00000000
[  405.729527] sof-audio-of-mt8186 10b80000.adsp: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x1c000000
[  405.740929] sof-audio-of-mt8186 10b80000.adsp: stack dump from 0x60115970
[  405.747855] sof-audio-of-mt8186 10b80000.adsp: 0x60115970: 00000000 00000000 00000000 00000000
[  405.756517] sof-audio-of-mt8186 10b80000.adsp: 0x60115980: 00000000 00000000 00000000 00000000
[  405.765191] sof-audio-of-mt8186 10b80000.adsp: 0x60115990: 4e1064ef 00000000 00000000 00000000
[  405.773888] sof-audio-of-mt8186 10b80000.adsp: 0x601159a0: 4e180570 00ffffff 748b8626 00000002
[  405.791054] sof-audio-of-mt8186 10b80000.adsp: 0x601159b0: 8e106c33 60115cc0 00000000 ce1064ef
[  405.799725] sof-audio-of-mt8186 10b80000.adsp: 0x601159c0: 4e1064ef 00060530 00000001 00000002
[  405.808356] sof-audio-of-mt8186 10b80000.adsp: 0x601159d0: 4e191840 00ffffff 00000000 00000004
[  405.817149] sof-audio-of-mt8186 10b80000.adsp: 0x601159e0: 00000000 ffffff00 00000000 00000000
[  405.825825] sof-audio-of-mt8186 10b80000.adsp: ------------[ DSP dump end ]------------
[  406.092520] sof-audio-of-mt8186 10b80000.adsp: ipc tx timed out for 0x30130000 (msg/reply size: 12/0)
[  406.101762] sof-audio-of-mt8186 10b80000.adsp: IPC timeout
[  406.107453] sof-audio-of-mt8186 10b80000.adsp: error: tplg component load failed -110
[  406.115302] sof-audio-of-mt8186 10b80000.adsp: error: failed to load DSP topology -22
[  406.123137] sof-audio-of-mt8186 10b80000.adsp: ASoC: error at snd_soc_component_probe on 10b80000.adsp: -22
```

lgirdwood, 2023-07-11:
@lyakh @marcinszkudlinski fyi.

wenliangwu, 2023-07-12:
@lgirdwood  OK. I will send a test PR later.

wenliangwu, 2023-07-12:
I submitted a pull request at https://github.com/thesofproject/sof/pull/7925. 
Additionally, could you please provide me with the cache configuration used for the Intel platform? I would like to confirm if it is possible to configure the same setting on MediaTek platform using the DSP MPU setting, and if this configuration update can also resolve the problem.

lgirdwood, 2023-07-12:
> I submitted a pull request at #7925. Additionally, could you please provide me with the cache configuration used for the Intel platform? I would like to confirm if it is possible to configure the [...]

See the core-isa.h for Intel platforms. They will be in the v2.2 branch and now Zephyr.

wenliangwu, 2023-07-13:
core-isa.h is DSP CORE configuration. I may have misunderstood your previous statement, and my phrasing may have been incorrect, too. If the difference lies in the hardware cache configuration, I guess it cannot be resolved through SW reconfiguration.
Earlier, I was referring to the type of memory being used, which is why it can be configured in the MPU. (https://github.com/thesofproject/sof/blob/main/src/platform/mt8188/platform.c#L138)

closed 2023-07-31 by lgirdwood