thesofproject/sof #7926: [BUG] LunarLake: number of cores must be changed to 5

plbossart, 2023-07-12:
cloning issue https://github.com/thesofproject/linux/issues/4472

The hardware supports 5 cores but the firmware can only support 4 so far. This is clearly not quite right. Adding as a P1 issue to make sure this is tracked and resolved for both firmware and kernel.

lgirdwood, 2023-07-12:
@mwasko pls re-assign as needed, we can then validate with kernel when ready. Thanks.

abonislawski, 2023-07-18:
@jxstelter is this zephyr core count issue still valid?

marcinszkudlinski, 2023-07-19:
looks like newest SOF (21ed10abf8a2673f13dea3fd39a063c561083054)  has 5 cores enabled
[image]

lgirdwood, 2023-07-26:
I think we can close now unless we need a kernel PR ? @plbossart .

plbossart, 2023-07-26:
kernel PR https://github.com/thesofproject/linux/pull/4487 is merged already so we are good to go.

closed 2023-07-26 by plbossart