thesofproject/sof #7931: [BUG][Multicore] topology loading failed with source module-copier.21.2 not found

fredoh9, 2023-07-12:
**Describe the bug**

MTLP_RVP_NOCODEC_MULTICORE platform failed to load multicore topology.

```
[    5.092484] snd_sof:sof_route_load: sof-audio-pci-intel-mtl 0000:00:1f.3: sink host-copier.0.capture control none source dai-copier.SSP.NoCodec-0.capture
[    5.092488] snd_sof:sof_route_load: sof-audio-pci-intel-mtl 0000:00:1f.3: sink host-copier.2.capture control none source dai-copier.SSP.NoCodec-2.capture
[    5.092491] snd_sof:sof_route_load: sof-audio-pci-intel-mtl 0000:00:1f.3: sink gain.1.1 control none source host-copier.0.playback
[    5.092494] snd_sof:sof_route_load: sof-audio-pci-intel-mtl 0000:00:1f.3: sink gain.5.1 control none source host-copier.2.playback
[    5.092496] snd_sof:sof_route_load: sof-audio-pci-intel-mtl 0000:00:1f.3: sink gain.19.1 control none source module-copier.21.2
[    5.092498] sof-audio-pci-intel-mtl 0000:00:1f.3: error: source module-copier.21.2 not found
[    5.092533] sof-nocodec sof-nocodec: ASoC: topology: add_route failed: -22
[    5.092556] sof-nocodec sof-nocodec: ASoC: topology: could not load header: -22
[    5.092726] sof-audio-pci-intel-mtl 0000:00:1f.3: error: tplg component load failed -22
[    5.092758] sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed to load DSP topology -22
[    5.092784] sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[    5.092839] sof-nocodec sof-nocodec: ASoC: failed to instantiate card -22
[    5.093102] sof-nocodec: probe of sof-nocodec failed with error -22
```

**To Reproduce**
Just loading the latest sof-mtl-nocodec-multicore.tplg

**Reproduction Rate**
100%

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**

- Kernel Branch: topic/sof-dev
- Kernel Commit: [0a80aee96685](https://github.com/thesofproject/linux/commit/0a80aee96685)
- SOF Branch: main
- SOF Commit: [70c257f9ae93](https://github.com/thesofproject/sof/commit/70c257f9ae93)

Intel internal daily test link: planresultdetail/28947

fredoh9, 2023-07-12:
When I replaced with the topology from yesterday(1fadc2118754), topology loading is fine.
@yongzhi1 probably side effect from https://github.com/thesofproject/sof/pull/7847?

yongzhi1, 2023-07-13:
Hi, @fredoh9 , sorry I tested on mtl-005-drop-stable only and all DMIC dev works fine.
[dmesg-dmic-SFX2.txt](https://github.com/thesofproject/sof/files/12043431/dmesg-dmic-SFX2.txt)

wszypelt, 2023-07-14:
@fredoh9 is it a topology problem or FW? if FW then please upload IPC Payloads

PR #7960 (2023-07-20): topology2: fix conditional inclusion of DMIC snippets for nocodec topology
DMIC now is controlled by two variables, PASSHTHROUGH and NUM_DMICS,
we should consider the values for both variables to decide the inclusion of DMIC
pipelines, routes, widgets, etc.

Fixes: #7931

Without this PR, there are DMIC components declared in the topology even with NUM_DMICS=0, and our tplgtool.py will fail to parse the topology because of a widget apprears in SectionRoute, but not in SectionWidget. Or there is an incomplete and isolated route in the topology graph.
commit fc1ddf95 (2023-07-20): topology2: fix conditional inclusion of DMIC snippets for nocodec

DMIC now is controlled by two variables, PASSHTHROUGH
and NUM_DMICS, we should consider the values for both
variables to decide the inclusion of DMIC pipelines,
routes, widgets, etc.

Fixes: #7931