thesofproject/sof #7968: [BUG] [MTL] Sum of KCPS consumption is much higher when audio is used for the 1st time after boot

yongzhi1, 2023-07-20:
**Describe the bug**
After boot, the very first time when audio is used, KCPS number is always much higher than the 2nd time.

**To Reproduce**
1. Power on DUT
2. Run any audio use case

```
aplay -Dhw:0,31 -c2 -r48000 -fS16_LE /tmp/sine.wav -vv -d5
```
And observe KCPS numbers from mtrace dump:
```
[00:00:00.197,913] <inf> pipe: add_pipeline_cps_consumption: Sum of KCPS consumption: 837642, core: 0
```
**Reproduction Rate**
100%

**Expected behavior**
A clear and concise description of what you expected to happen.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {https://chromium.googlesource.com/chromiumos/third_party/kernel/+/refs/heads/chromeos-6.1}
    * SOF: {mtl-005-drop-stable}
2) Name of the topology file
    * Topology: {sof-mtl-max98357a-rt5682-ssp2-ssp0.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {MTL/REX}

**Screenshots or console output**
[dmesg-mtrace-playback-after-boot.txt](https://github.com/thesofproject/sof/files/12101085/dmesg-mtrace-playback-after-boot.txt)

lgirdwood, 2023-07-26:
plus @btian1 and @andrula-song

yongzhi1, 2023-08-08:
Thanks @ujfalusi for the fix https://github.com/thesofproject/linux/pull/4508, the issue should be moved to https://github.com/thesofproject/linux instead.

closed 2023-08-08 by yongzhi1