thesofproject/sof #7979: [BUG] Speaker playback IO error

aiChaoSONG, 2023-07-25:
**Describe the bug**
Playback a wave to speaker with aplay, see io error in terminal immediately. no obviously error in kernel log, and in mtrace, xrun happened.

```
[    1.999690] <inf> host_comp: host_get_copy_bytes_normal: comp:0 0x40000 no bytes to copy, available samples: 384, free_samples: 0
[    2.000706] <inf> host_comp: host_get_copy_bytes_normal: comp:0 0x40000 no bytes to copy, available samples: 384, free_samples: 0

```

**To Reproduce**
simple do aplay to speaker pipeline: 
aplay -Dhw:0,2 -f dat -c 2 -i -vvv /dev/zero

**Reproduction Rate**
100%

The first bad commit is: e424b874a477617d080f368ae67b46d19b7e6d4d

mengdonglin, 2023-07-26:
@ranj063 Can you help check this regression?

ranj063, 2023-07-27:
> @ranj063 Can you help check this regression?

@mengdonglin I think I know the issue. We're missing the src buffer update in the case of multie endpoint playback. I've given @btian1 the change to try out and check if it fixes the problem

lgirdwood, 2023-07-31:
> it fixed this issue, please check with below PR, @aiChaoSONG @bardliao @ranj063 #7987

@aiChaoSONG Now merged #7987.

aiChaoSONG, 2023-07-31:
Verified fix #7987, close.

closed 2023-07-31 by aiChaoSONG