thesofproject/sof #7980: [BUG][MTL-005.3] IPC timed out when doing start/stop speaker/jack pipeline while dmic is running

aiChaoSONG, 2023-07-25:
**Describe the bug**
An error of ipc timed out is observed when start/stop speaker/jack pipeline while dmic is running.

**To Reproduce**
1. do arecord through dmic pipeline:  arecord -Dhw:0,4 -f s32_le -c2 -i -r 48000 -vvv /dev/null
2. do aplay through speakder pipeline: aplay -Dhw:0,2 -f dat -c 2 -d5 -i -vvv /dev/zero
3. kill aplay with CTRL+C, and start it again
4. loop step 3.

**Reproduction Rate**
not high, about 5%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {sof-dev 70d2c68e48d0}
    * SOF: {MTL-005.3 release}
2) Name of the topology file
    * Topology: {sof-mtl-rt711-l0-rt1316-l23-rt714-l1.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {MTL SDW}

[dmesg_ipc_timedout.txt](https://github.com/thesofproject/sof/files/12158208/dmesg_ipc_timedout.txt)

syedk008, 2023-07-26:
I do not observe the issue with the sof-fw commit id:1a96df1e63e68754c58ce777bd8c5a62f6d06f58.

mengdonglin, 2023-07-26:
We didn't reproduce this issue on SOF main branch commit https://github.com/thesofproject/sof/commit/1a96df1e63e68754c58ce777bd8c5a62f6d06f58 in 0.5 hour *stress* test.

But we cannot check if this issue can be reproduced by the tip of SOF main branch, as another issue https://github.com/thesofproject/sof/issues/7979 was introduced by commit https://github.com/thesofproject/sof/commit/e424b874a477617d080f368ae67b46d19b7e6d4d on top of https://github.com/thesofproject/sof/commit/1a96df1e63e68754c58ce777bd8c5a62f6d06f58. 

[image]

keqiaozhang, 2023-10-09:
This issue cannot be reproduced on mtl-006. No reproductions in CI tests. Closing this bug.

closed 2023-10-09 by keqiaozhang